espressif/esptool #135: code uploaded succeeds but ESP8266 doesn't run (NodeMCU) for ESP8266_RTOS_SDK v1.4.0

sarwadenj, 2016-10-14:
Hi,
I'm trying to run blinking LED program using ESP8266_RTOS_SDK.
program link-->https://github.com/sarwadenj/ESP8266_RTOS/tree/master/test_2
also i written another program and set WiFi in station mode--->https://github.com/sarwadenj/ESP8266_RTOS/blob/master/wifi/user/user_main.c

I compiled my program using 
./gen_misc.sh
and set 
STEP 1: use boot_v1.2+ by default
STEP 2: generate bin: user1.bin
STEP 3: spi speed: 40 MHz
STEP 4: spi mode: QIO
STEP 5: spi size: 4096KB
spi ota map: 1024KB + 1024KB

and uploaded code going into Bin_path-->
sudo esptool.py --port /dev/ttyUSB0 write_flash --flash_mode qio 0x2000 user1.4096.new.6.bin

it uploaded successfully but code didn't run. So what should be the problem. is I'm missing something.

Is there any solution for this?
Thanks in advance.

projectgus, 2016-10-17:
Please try these suggestions in the troubleshooting guide:
https://github.com/themadinventor/esptool#write_flash-succeeds-but-esp8266-doesnt-run

projectgus, 2016-11-04:
Did you make any progress with this?

projectgus, 2016-11-07:
I'm going to close this as it seems like it's not an esptool.py issue, but feel free to reopen if you have some information.

closed 2016-11-07 by projectgus