espressif/esptool #136: Reset To Bootloader Issues on Windows

negativekelvin, 2016-10-18:
On windows, it seems like there are timing issues with the reset to bootloader functions using the DTR and RTS circuit for esp32 because setDTR and setRTS are sent separately.  Possible fix:

```python
            # issue reset-to-bootloader:
            # RTS = either CH_PD or nRESET (both active low = chip in reset)
            # DTR = GPIO0 (active low = boot to flasher)
            self._port._dtr_state = False
            self._port._rts_state = True
            self._port._reconfigure_port()
            time.sleep(0.05)
            self._port._dtr_state = True
            self._port._rts_state = False
            self._port._reconfigure_port()
            time.sleep(0.05)
            self._port.setDTR(False)
```

I haven't scoped this or tested it because my v1 board has the transistor bug.

projectgus, 2016-10-18:
Thanks @negativekelvin. I don't know if you're following the forum thread, but I'm guessing you are. 
http://esp32.com/viewtopic.php?f=2&t=334&p=1503#p1497

I considered this solution as well (although I didn't test it either.) The problem is, I'm not comfortable relying on the private interface of pyserial. At minimum, it's necessary to check it hasn't changed over any previous supported pyserial versions. But even if it works with all those, if the private interface changes in the future then it could break esptool without warning.

A hacky but somewhat safer way would be to check all these properties exist with hasattr() before calling them, and fail back to the public API instead. But this is still super hacky.

The safest, most maintainable, fix that I can think of is probably to call the win32 GetCommState/SetCommState API functions directly. This is a stable public API, and the only private part of pyserial it needs to touch is the port handle. I still don't like that idea, but it seems least likely to suddenly break.

projectgus, 2016-10-18:
Or to submit a patch to pyserial, some variant of the existing `apply_settings()` that applies all settings as an "atomic" operation.

negativekelvin, 2016-10-18:
Yes, that makes sense.  I don't know if those properties and methods are very likely to change, so yes I like the check and fall back strategy until pyserial can support it.

negativekelvin, 2016-10-18:
However it looks like `_reconfigure_port()` only updates the RTS/DTR lines on windows, not other platforms.

bvernoux, 2016-10-18:
See my comment here about that problem on ESP32 Windows7 with esptool.py from esp-idf
https://github.com/themadinventor/esptool/issues/130#issuecomment-254651499

bvernoux, 2016-10-20:
In fact after analysis of rudi patch it is pure luck (thanks to watchdog) 
For more details on Rudi patch see Saleae Logic  1.2.10 capture:
* hydrabus.com/esp32_esptool_download/esptool_download_MyPatch_40MHz_80MSamples.logicdata
[image]

I have simplified it like that
esptool.py modifications (MyPatch):
from
```
            self._port.setRTS(True)
            time.sleep(0.05)
            self._port.setDTR(True)
            self._port.setRTS(False)
            time.sleep(0.05)
            self._port.setDTR(False)
```
to
```
            self._port.setDTR(False) # EN=0 and IO0=0
            time.sleep(0.05)
            self._port.setRTS(False) # EN=1 and IO0=1(glitch on IO0 during 1ms)
            self._port.setDTR(True)  # EN=1 and IO0=0
```

The main problem is the glitch on IO0 during 1.1ms which does not allow to enter download mode as IO0 is sampled by ESP32 during this glitch but we are lucky that we have a reset by watchdog after 0.385s (with message RTCWDT_RTC_RESET) which then boot in download mode because EN=1 and IO0=0

For more details see Saleae Logic  1.2.10 capture of MyPatch:
* hydrabus.com/esp32_esptool_download/esptool_download_RudiPatch_40MHz_80MSamples.logicdata
[image]

negativekelvin, 2016-10-21:
@bvernoux are you able to capture the waveforms for the method using reconfigure_port in the first post on windows?

projectgus, 2016-10-21:
I just tried a different fix for this issue (1f5d9d2) which should be equivalent to calling _reconfigure_port, and I thought would definitely correct the issue.

Unfortunately the timing seems to be the same (trace attached), even when both DTR & RTS are set in the same Windows API call. Which I think means the CP2102 Windows driver is doing this, and there's no way to change that pulse in software.

[image]

negativekelvin, 2016-10-21:
That is disappointing.  Time to ask silicon labs?

negativekelvin, 2016-10-21:
> we are lucky that we have a reset by watchdog after 0.385s

What is the cause of this?

projectgus, 2016-10-21:
>> we are lucky that we have a reset by watchdog after 0.385s
> What is the cause of this?

It's a silicon bug, that will be fixed at some point.

A solution that seems to work reliably is to add a ~2.2uF capacitor to the EN pin (between EN and GND), so it rises slower (there is a ~12K pullup on this pin.) This is not super desirable, as it requires a hardware change, but it works.

bvernoux, 2016-10-28:
I confirm adding a 2.2uF capacitor on EN (connected between EN and GND) fix the problem
[image]

krzychb, 2016-11-05:
@projectgus and all - thank you guys!
No more fiddling with EN / Boot buttons.
Life is much easier with this simple hack :smile: 
Krzysztof

[image]

ladyada, 2016-11-13:
hiya - got bit by this bug, and found a few things helped. capacitor helps but easier to change the second time.sleep(0.05) to time.sleep(0.5) e.g. https://github.com/adafruit/arduino-esp32/blob/master/tools/esptool.py#L279
that along with a formal 'reset' procedure to swat the ESP32 back into 'user mode'
https://github.com/adafruit/arduino-esp32/blob/master/tools/esptool.py#L296
has made developing code work great on win7 + CP210x... plz re-consider the wontfix? there's a lot of those DevKitC's in the world now :) i can test on mac/linux too

bvernoux, 2016-11-13:
For information it is already fixed in actual branch even without the capacitor => thanks to an ESP32 silicon bug with reset by watchdog after 0.385s
I do not see why this modification will improve anything.
Do you have a sigrok capture to show the difference with this patch ?

projectgus, 2016-11-14:
Hi @ladyada ,

> For information it is already fixed in actual branch even without the capacitor

I don't believe this is fixed in any branch at the moment, without adding either a capacitor or the `time.sleep(0.5)` modification that @ladyada mentions.

>plz re-consider the wontfix? there's a lot of those DevKitC's in the world now :)

The problem I see is because the 0.5s delay happen to rely on a silicon bug, which is already fixed in the forthcoming chip revision. When the updated chips come out, the bug will bite everyone again (if the capacitor on EN hasn't been changed in the board design) and it will also take longer any time esptool.py fails to connect, due to the extra delays.

There are perhaps some other ways to work around this, either a command line option like `--variant` or `--board` to let people specify they have the gen1 silicon, or a long-standing suggestion (#27) to read back the "header" message that the chip prints in order to detect boot mode failures earlier.

ladyada, 2016-11-14:
yah ok - i didn't realize it was a silicon issue on the esp32 (thought it was cp210x) in which case... i agree to keep as wontfix.
i will update the devkitc product page to tell people to add a capacitor if they're having upload probs on windows

projectgus, 2016-11-14:
> yah ok - i didn't realize it was a silicon issue on the esp32 (thought it was cp210x) in which case... i agree y'all should keep as wontfix.

Just to be sure we're on the same page, the silicon issue on esp32 is what allows the extra `time.sleep(0.5)` delay to work as a workaround (the chip resets twice due to an unexpected WDT), rather than the cause of this bug. The cause of this bug is an interaction between the reset circuit timing and Windows driver behaviour (probably CP210x-specific but I haven't confirmed this.)

>i will update the devkitc product page to tell people to add a capacitor if they're having upload probs on windows

That would be awesome, thanks very much. :)

ladyada, 2016-11-14:
ahh ok yes reading thru more carefully i understand the double-bug interaction! too funny - ok thanks!

aircable, 2016-11-25:
the new version 2.0-dev does not work with the ESP8266, but does on the ESP32.
the version 1.3dev works on ESP8266 but not on the ESP32.
This bug should be easy to fix, eh?

$ python esptool32.py -p /dev/tty.SLAB_USBtoUART63 read_mac
esptool.py v2.0-dev
Connecting...
Detecting chip type... ESP8266
Uploading stub...
Running stub...
Stub running...
(24, 254, 52, 155, 167, 200)
Traceback (most recent call last):
  File "esptool32.py", line 2061, in <module>
    main()
  File "esptool32.py", line 1903, in main
    operation_func(esp, args)
  File "esptool32.py", line 1600, in read_mac
    print_mac("WiFi MAC", mac[0])
  File "esptool32.py", line 1595, in print_mac
    print '%s: %s' % (label, ':'.join(map(lambda x: '%02x' % x, mac)))
TypeError: argument 2 to map() must support iteration

$ python esptool.py -p /dev/tty.SLAB_USBtoUART63 read_mac
esptool.py v1.3-dev
Connecting...
MAC: 18:fe:34:9b:a7:c8

projectgus, 2016-11-25:
@aircable thanks for reporting this, but can you [open a new issue](https://github.com/espressif/esptool/issues/new) for this please? It is not related to this reset to bootloader bug.

aircable, 2016-11-25:
thanks I didn't find the link....
will do

rojer, 2016-12-23:
just found my way to this issue. not before spending some time banging my head against the wall, though :)

the workaround seems fine, i just want to confirm: devkit hardware will be revised to increase the capacitor value, right? hopefully, in time for new chip revision which fixes the glitch.

projectgus, 2016-12-27:
Yes, newer Espressif DevKits will fix this issue via increased capacitance on the EN pin.

projectgus, 2017-01-10:
esptool v2.0 (current master, and the version in esp-idf) has a new `--after esp32r0` option which will work around this issue for ESP32 original revision chips.

The option can also be selected under "make menuconfig" in esp-idf.

MartyMacGyver, 2017-01-23:
I've been investigating this and I think there's a more permanent solution that'd be effective and not too burdensome. Interestingly, the `esp32r0` workaround doesn't seem to help at all for this yet it sounds like a similar issue if not the same one.

A long discussion follows with technical details I've gleaned, but the bottom line is that holding the chip _in reset_ longer does the trick. The code change to esptool.py is a one-liner:
    ...
    self._port.setDTR(False)  # IO0=HIGH
    self._port.setRTS(True)   # EN=LOW, chip in reset
    time.sleep(2.0)  # Sleep longer
    self._port.setDTR(True)   # IO0=LOW
    self._port.setRTS(False)  # EN=HIGH, chip out of reset
    ...

This has the advantage of working on both current boards (and 2 seconds isn't long to wait relative to everything else). If it were a default it'd work smoothly with the Arduino-ESP32 as well - not sure how one is supposed to get `--before esp32r0` to work there - is it a config option there (the reset-on-disconnect thing is of related interest but for other discussions).

Note: for the hard-reset bit, while I see the `--before esp32r0` option, I don't see it in `--after`. Not sure if there is any problem with board reset (as long as DTS is in the right state) so maybe that was a typo?

----
The Core Board v2 and ESP-WROVER-KIT act differently upon serial disconnect, and downloads negotiation works slightly differently for each. 

Previous discussions about this include:
https://github.com/espressif/esp-idf/issues/59
https://github.com/espressif/arduino-esp32/issues/127#issuecomment-274368250

Tested on Windows with the latest serial drivers via RealTerm:
Si Labs CP210x driver v6.7.4.261 (non-enumerating) for the Core Board v2
FTDI driver v2.12.24.0 (enumeration disabled) for the ESP-WROVER-KIT

Note: interestingly, clear/set can be positive logic (clear = 0/low/false, set = 1/high/true), or the inverse (active low). In esptool.py the active low mode is used - here, I used active high notation.

For the ESP-WROVER-KIT, to get to download mode, it's pretty simple:
  - Start with RTS and DTR set
  - Do the following in quick succession (no long pause between the steps):
      - clear DTR (holds reset low)
      - set DTR
      - clear RTS (enters download mode)
      - set RTS (can immediately do this)
      - toggle DTR clear+set to reset in run mode

The Core Board v2 is different though - it needs about 2 seconds delay between clearing DTR and setting it:
  - Start with RTS and DTR set
  - Do the following in quick succession (except as noted):
      - clear DTR (holds reset low)
      - wait 2 seconds
      - set DTR
      - clear RTS (enters download mode)
      - set RTS (can immediately do this)
      - toggle DTR clear+set to reset in run mode

The boot up sequence differs slightly too - on the Core Board v2 that long pause causes a flash error.

_ESP-WROVER-KIT_
    
    ets Jun  8 2016 00:22:57
    
    rst:0x1 (POWERON_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
    ets Jun  8 2016 00:22:57
    
    rst:0x10 (RTCWDT_RTC_RESET),boot:0x6 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))
    waiting for download

_Core Board v2_
    
    ets Jun  8 2016 00:22:57
    
    rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
    flash read err, 1000
    Falling back to built-in command interpreter.
    OK
    >ets Jun  8 2016 00:22:57
    
    rst:0x10 (RTCWDT_RTC_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
    waiting for download
    
Also peculiar: if DTR and RTS are both cleared, the EN switch on the ESP-WROVER-KIT won't work at all (this is not the case on the Core Board v2).

(As an aside, on both devices, if RTS is set, setting DTR (from cleared) resets the board. If DTR is clear, clearing RTS (from set) also resets the board.)

The only obvious difference between the two RTS/DTR circuits in the corresponding schematics is that the Core Board v2 specs 12kohm resistors between the CP2102 bridge and the bases of the S8050 transistors, while the ESP-WROVER-KIT specs 1kohm resistors between the FT2232HL bridge and the bases of the S8050 transistors. (The BOOT and EN switch circuits appear to be identical on both.) 

Note: I have no shunts bridging the CTS/RTS headers on the WROVER board. It doesn't appear it would have a positive effect.

MartyMacGyver, 2017-01-23:
One other thing... I'm using non-enumerating drivers here. I strongly recommend people stay far away from the enumerating drivers (or disable enumeration for the ports using it), particularly in Windows, if they don't have a reason to be using it (typically emulating a mouse).

If the mouse filter gets spurious data (as is quite likely even from a factory fresh device during reset/boot-up) you could get a blue screen. [Longer discussion here.](https://www.falatic.com/index.php/172/usb-serial-devices-and-the-unexpected-side-effects-of-serial-enumeration-bsod-with-solutions)

The default driver Windows installs for the CP2102 is enumerating - the one I used to test all this is the same one but non-enumerating. That alone could compound everything above as there is that extra layer of introspection for enumeration purposes (and they do act differently... it's annoying. My fix above should work in either case, though the risk of BSOD would remain with enumeration).

grisevg, 2017-01-26:
This is not windows specific. Same issue on macOS with cp210x drivers: https://github.com/espressif/esp-idf/issues/305

projectgus, 2017-03-02:
Thanks to @MartyMacGyver and everyone else who helped diagnose this, we seem to have a reliable fix now which doesn't require any special command line arguments.

This fix should propagate to the ESP-IDF & Arduino esptool.py versions, soon.

Also, new development board hardware should no longer exhibit the underlying bug in the first place (provided 100nF or more of capacitance is present on the EN pin, with a 10K or higher pullup on that pin).

Please open an issue if this fix doesn't appear to fix auto-reset on your board (either reopen this issue, or if you have more specifics to provide then please open a new one instead.) Thanks.

closed 2017-03-02 by projectgus

danirebollo, 2020-01-08:
Hi, 
I've been able to fix this with 1uF capacitor from IO0 to GND. I have another on EN but the one that solves this bug on some silicon revisions are the 1uF capacitor on IO0.

cdegroot, 2020-03-13:
FWIW, I have the NodeMCU ESP32S boards (https://www.amazon.ca/KeeYees-Development-Bluetooth-Microcontroller-ESP-WROOM-32/dp/B07QCP2451) that needed button pressing to program. A 1µF cap didn't change anything, a 2.2µF did the trick. ESPtool.py v2.8 under Linux.

cybermaus, 2022-01-10:
Cross-link to similar problem for CP2102 chips and its solution:
https://github.com/espressif/esptool/issues/706#issuecomment-996122044

gabrielfonte, 2022-07-29:
> ```python
>             time.sleep(0.05)
> ```

I have found the same situation with Windows 10 and CP2102, this one solved the issue.

stefanschuermans, 2023-01-14:
Hi all,

Thanks for all the analysis and descriptions so far. I ran into the issue of auto-resetting an "ESP32S" module (based on ESP-WROOM-32 and CP2102) today on Linux (Debian 11.6) and looked into it a bit. I noticed that the CP2102 apparently shows some behavior regarding activating RTS and DTR lines that deviates from most other USB/serial converters I have used so far. It looks to me as if an activation command (ioctl) for one of RTS or DTR somehow resets the other line to the idle state. Some of the commands are apparently even ignored (that means do not change the pin states at all).

I do not know if it is a solution that anybody else would like to use, but I manged to make it work for my case. The following patch to esptool.py (which is quite dirty because it's messing with pyserial internals) made the auto-reset work:
```
diff --git a/esptool.py b/esptool.py
index bea3555..44d3a21 100755
--- a/esptool.py
+++ b/esptool.py
@@ -459,7 +459,14 @@ class ESPLoader(object):
         #
         # DTR & RTS are active low signals,
         # ie True = pin @ 0V, False = pin @ VCC.
-        if mode != 'no_reset':
+        if mode == 'cp210x_linux_reset':
+            import fcntl
+            fcntl.ioctl(self._port.fd, serial.serialposix.TIOCMSET, serial.serialposix.TIOCM_RTS_str)
+            time.sleep(0.1)
+            fcntl.ioctl(self._port.fd, serial.serialposix.TIOCMSET, serial.serialposix.TIOCM_DTR_str)
+            time.sleep(0.1)
+            fcntl.ioctl(self._port.fd, serial.serialposix.TIOCMSET, serial.serialposix.TIOCM_zero_str)
+        elif mode != 'no_reset':
             self._setDTR(False)  # IO0=HIGH
             self._setRTS(True)   # EN=LOW, chip in reset
             time.sleep(0.1)
@@ -2819,7 +2826,7 @@ def main(custom_commandline=None):
     parser.add_argument(
         '--before',
         help='What to do before connecting to the chip',
-        choices=['default_reset', 'no_reset', 'no_reset_no_sync'],
+        choices=['default_reset', 'no_reset', 'no_reset_no_sync', 'cp210x_linux_reset'],
         default=os.environ.get('ESPTOOL_BEFORE', 'default_reset'))
 
     parser.add_argument(
```
Flashing needs to be done with `make flash CONFIG_ESPTOOLPY_BEFORE=cp210x_linux_reset` in order to activate the special reset mode.

Some technical details as far as I understand them so far:

The pyserial module uses the TIOCMBIS and TIOCMBIC ioctl calls to modify one of the DTR and RTS bits at a time. Using those commands, I could not convince the CP2102 to output signals with a timing that worked. Using the TIOCMSET ioctl call, it is possible to set RTS and DTR to a new state at the same time. The CP2102 still did not output the signals exactly as indicated in the command. (I do not know why. I don't even know if it should.). However, I manged to find a sequence of commands that puts RTS to low (activate reset), and a bit later RTS to high and DTR to low (take back reset and indicate "flash mode" via GPIO0) within the same microsecond. Apparently, this is fast enough so that the ESP32 sees the GPIO0 at low (flash mode) when coming out of reset and auto-flashing works for me.

radimkarnis, 2023-01-14:
Hi @stefanschuermans,
a similar solution has been implemented just recently in https://github.com/espressif/esptool/commit/353cefc85d7011d8d26d38861e8c736160df2ce6. Does this help in your case? You can try pulling the latest esptool or install it with `pip install esptool==4.5.dev1`.

stefanschuermans, 2023-01-14:
Hi @radimkarnis. Thanks for this pointer. I have tried this version and it works for me as well. :-) That's great, so no I do not need a custom solution.

MrParisRoys, 2023-02-15:
wie lad ich diese version herunter und wie kann die installiern bitte hilfe