espressif/esptool #139: v1.2 release?

pfalcon, 2016-10-29:
Is there any progress/ETA on v1.2 release? I understand there's desire to fit https://github.com/themadinventor/esptool/pull/127 into 1.2, but that's kind of changes which is hard to do for anyone but a maintainer, and even for a maintainer it's a bit a chore to do it all-in-once. So, may I suggest to cut release 1.2 soon, and apply https://github.com/themadinventor/esptool/pull/127 immediately after it, install syntax checkers to not let python2-only syntax in, and then unbreak remaining cases, while any PR would be checked right away. Thanks in advance.

projectgus, 2016-11-01:
I think this is a good suggestion, I'll cut a v1.2 release today or tomorrow.

Angus

closed 2016-11-04 by projectgus

pfalcon, 2016-11-04:
Great, thanks!

commit 9dfcb350 (2016-11-04): Version v1.2

Closes #139