espressif/esptool #140: Broken shebang in 1.2

crasu, 2016-11-05:
pip install esptool==1.1
...
(nodemcu)rose:~/projects/nodemcu/docker-build$ esptool.py 
usage: esptool [-h] [--port PORT] [--baud BAUD]

pip install esptool==1.2
...
(nodemcu)rose:~/projects/nodemcu/docker-build$ esptool.py 
bash: /home/christian/.virtualenvs/nodemcu/bin/esptool.py: /home/gus/dev/esp/esptool-py/test_env/bin/python: bad interpreter: No such file or directory

projectgus, 2016-11-05:
Thanks for reporting this. Looks like I got bit by this bug building the wheel:
https://bitbucket.org/pypa/wheel/issues/135/bad-interpreter-lines-can-sneak-into

I'm away from my computer today but I'll fix this asap. Building the package on Travis might be a reliable solution.

pfalcon, 2016-11-05:
I can reproduce this with ubuntu 14.04. The problem isn't with git repo, but with Python wheel published on PyPI. It has 2 esptool.py files inside: one at the root of .whl, having good shebang, another in esptool-1.2.data/scripts/ with that @projectgus' adhoc shebang.

Didn't look into 1.1. It's distro on PyPI in particular has a normal source package besides .whl.

projectgus, 2016-11-06:
Please try again, I've deleted the wheel package for pypi and uploaded a plain source package.

The bug seems to be triggered when you have a shebang line, a virtualenv activated, and try to build a wheel:
http://stackoverflow.com/questions/33792696/how-to-prevent-python-wheel-from-expanding-shebang/33808977

An odd thing is v1.1's wheel file contents include the second esptool.py with a wrong shebang line generated into it also, but noone seems to have a problem there. Possibly pip has been installing the source package instead.

projectgus, 2016-11-06:
Even weirder, "pip install" with 1.2 now downloads the source package tarball, builds a wheel from it, and then installs the wheel. Even when this process happens inside a virtualenv, that wheel ends up OK:
```
$ unzip -l /home/gus/.cache/pip/wheels/f1/70/54/a5b96d296178d2496518cf3be381e583d1d6bdd1981bfd5c19/esptool-1.2-cp27-none-any.whl
Archive:  /home/gus/.cache/pip/wheels/f1/70/54/a5b96d296178d2496518cf3be381e583d1d6bdd1981bfd5c19/esptool-1.2-cp27-none-any.whl
  Length      Date    Time    Name
---------  ---------- -----   ----
    53947  2016-11-04 22:15   esptool.py
    53934  2016-11-06 12:03   esptool-1.2.data/scripts/esptool.py
      906  2016-11-06 12:03   esptool-1.2.dist-info/DESCRIPTION.rst
     1018  2016-11-06 12:03   esptool-1.2.dist-info/metadata.json
        8  2016-11-06 12:03   esptool-1.2.dist-info/top_level.txt
       96  2016-11-06 12:03   esptool-1.2.dist-info/WHEEL
     1772  2016-11-06 12:03   esptool-1.2.dist-info/METADATA
      643  2016-11-06 12:03   esptool-1.2.dist-info/RECORD
---------                     -------
   112324                     8 files
```

At leas it's an easy enough workaround to always post source packages and never post wheels from now on, though...

pfalcon, 2016-11-06:
> At least there's an easy enough workaround to always post source packages and never post wheels from now on...

Thanks for that, Angus. Wheels where pretty distant in my backlog of things to look at, and now move to the end of it at all...

crasu, 2016-11-06:
Works for me now - thank you

pfalcon, 2016-11-06:
@projectgus : I hope you don't want to say that you did espressif-way: released the new package with different content under the same version number?

commit 793efa71 (2016-11-06): Version v1.2.1

Closes #140 (packaging bug)
closed 2016-11-06 by projectgus