espressif/esptool #143: Not seeing the version

marcelstoer, 2016-11-10:
I was trying to figure out how to make `esptool.py` print its version. Neither the README nor `esptool.py -h` tells me.

I don't know Python but still stuck my nose into the code. Based on what I found at [esptool.py#L1028](https://github.com/espressif/esptool/blob/master/esptool.py#L1028) I was expecting to be able to run `esptool.py version` and get what I need. Obviously not..

```
/Applications/esptool (master) > esptool.py version
usage: esptool [-h] [--port PORT] [--baud BAUD]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,flash_id,read_flash,erase_flash}
               ...
esptool: error: argument operation: invalid choice: 'version' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'flash_id', 'read_flash', 'erase_flash')
```

When I checked all occurrences of `__version__` I found [esptool.py#L1037](https://github.com/espressif/esptool/blob/master/esptool.py#L1037) and wondered why I'm only seeing 

```
/Applications/esptool (master) > esptool.py -h
usage: esptool [-h] [--port PORT] [--baud BAUD]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,flash_id,read_flash,erase_flash}
               ...

ESP8266 ROM Bootloader Utility

positional arguments:
```
rather than "ESP8266 ROM Bootloader Utility" being prepended by the version.

pfalcon, 2016-11-10:
Perhaps you have too old esptool.py.

~~~
$ esptool.py version
esptool.py v1.2-dev
1.2-dev
~~~

marcelstoer, 2016-11-10:
I thought about this but I've always used the latest from the `master` branch.

```
/Applications/esptool (master) > git branch --all
  dev
* master
  remotes/origin/HEAD -> origin/master
...
```

`vim esptool.py` opens a file that says `__version__ = "1.3-dev"`on line 34 which is identical to what I see here on GitHub.

EXCEPT that I forgot to install the latest. So, `esptool.py xxx` actually ran the system-wide but old script even though there is an updated script in the current directory. All good now.

closed 2016-11-10 by marcelstoer