espressif/esptool #144: File generated by elf2image is invalid under specific circumstances

shodan8192, 2016-11-10:
Hello,

I'm integrating ESP-IDF into PlatformIO, and found problem with image generated by esptool.

In default, script in PlatformIO generate library files named in such pattern :

`libIDFComponents_freertos.a`

then, to linking stage, library is passed by path and fullname :
`.pioenvs\nano32\libIDFComponents_freertos.a`

gcc manual say, that it's ok to do that :
> The only difference between using an -l option and specifying a file name is that -l surrounds library with ‘lib’ and ‘.a’ and searches several directories.

but flash image generated from elf, that was built that way is invalid - application stuck at random point, dependent of program content.

If I change the script, so library file name become :
`libfreertos.a`
and to linker is passed : 
`.pioenvs\nano32\libfreertos.a`
then it's all right - program working properly, regardless of it's content.

In both cases elf file has the same size, but invalid image is slighty larger than valid.
Please see attached zip file, which contain two directories with firmware.bin and firmware.elf respectively for both cases.

Best regards

projectgus, 2016-11-10:
Hi Shodan,

I think something else is happening in the build process. If you look at the output of "xtensa-esp32-elf-objdump -h" for both ELF files, you'll see that the section sizes are quite different.

The esp-idf default linker script (under `components/esp32/ld/esp32.common.ld`) puts certain libraries into IRAM. The relevant section is here:

```
 .iram0.text :
  {
    /* Code marked as runnning out of IRAM */
    _iram_text_start = ABSOLUTE(.);
    *(.iram1 .iram1.*)
    *libfreertos.a:(.literal .text .literal.* .text.*)
    *libesp32.a:panic.o(.literal .text .literal.* .text.*)
    *libphy.a:(.literal .text .literal.* .text.*)
    *librtc.a:(.literal .text .literal.* .text.*)
    *libpp.a:(.literal .text .literal.* .text.*)
    *libhal.a:(.literal .text .literal.* .text.*)
    _iram_text_end = ABSOLUTE(.);
  } > iram0_0_seg
```

So libfreertos.a will be linked into IRAM (and therefore can be used when flash caching is disabled), whereas libIDFComponents_freertos.a will not be unless you've modified the linker script to match the new name.

The firmware is crashing because at some point during startup it's accessing FreeRTOS code while flash is unmapped.

(The .bin files are slightly different lengths because irom/drom sections on esp32 must be padded to 64KB boundaries, so the padding will be different for different section sizes.)

projectgus, 2016-11-10:
I've closed the issue as I'm fairly sure this will solve it, but feel free to reopen if you're not sure or if there's something else not right here.

closed 2016-11-10 by projectgus

shodan8192, 2016-11-10:
Thank you - this is a explanation which I needed - I saw those mappings before, but I completely forgot about them.