espressif/esptool #145: CH340 OSX flash error with speed greater 115200

sheinz, 2016-11-21:
I'm not sure if this is the issue of esptool.py or serial driver.

If flash with speed greater than 115200 (230400, 460800) esptool fails with error:
```
esptool.py -p /dev/tty.wchusbserial1410 --baud 230400 write_flash -fs 32m -fm qio -ff 40m \
                0x0 ../bootloader/firmware_prebuilt/rboot.bin 0x1000 ../bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/tests.bin
esptool.py v1.3-dev
Connecting...
Running Cesanta flasher stub...

A fatal error occurred: Invalid head of packet ('\xbc')
make: *** [flash] Error 2
```
The same behaviour on every board with CH340 usb-serial chip. And only on OSX Capitan 10.11.6
The same boards works fine with Linux.

On the other hand I can connect to a board with a terminal program on speeds 230400 and 460800 just fine and communication works also fine. So, the driver seems to support those speeds, on the first glance.

projectgus, 2016-11-24:
Hi @sheinz ,

That's interesting, if another serial program works then it does sound like it could be a pyserial or an esptool bug.

I don't have an OS X box on hand to test with. Is it possible to get some traces of the serial interaction? On Linux I'd use strace for this, I believe OS X has DTrace/dtruss but I don't know anything else about those tools.

Or alternatively a logic analyzer capture on the serial pins could be useful, just to confirm the baud rate change and look at what traffic is received there.

Angus

sheinz, 2016-11-25:
Hi Angus,

I captured the communication with logic analyzer.
The final data from the device is 
[image]

Speed: 230400
The data captured with saleae logic analyzer:
[esp8266_230400.tar.gz](https://github.com/espressif/esptool/files/613926/esp8266_230400.tar.gz)

projectgus, 2016-11-25:
Hi @sheinz,

That's interesting. Here's what I see in the capture:

* The initial negotiation all happens at 115942bps (according to the Salae.) This is nominally 115200bps according to esptool.py (that's the configured baud rate for initial negotiation.)
* esptool.py sends sync packets (0x55) a couple of times, the esp8266 locks onto the "115200bps" (actually 115942bps) signal with it's auto-baud hardware and talks back at approximately 11592bps also.
* esptool uploads the stub code at 115942bps. The stub includes a "parameters" section with the baud rate to switch to, which is 230400bps.
* The baud mismatch between the driver requesting 115200bps and getting 115942bps (+0.6%) implies either a minor driver problem (wrong clock divider), or a slightly out-of-spec crystal on the CH340 board.
* At 1.453s in the capture, the stub starts running and sends a SLIP frame with its  'OHAI' greeting at exactly 230400bps (this is the 0xc0, 0x4f, 0x48, 0x41, 0x49, 0xc0) above.
* However, the CH341 is listening at nominal "230400bps" which we can predict is going to be really 231782bps. And I suspect this causes it to see a corrupted frame header of 0xBC instead of 0xC0. And it all falls apart.

So...

The reason the stub uses a fixed baud rate of this kind rather than auto-bauding is that for higher baud rates this is normally more accurate, the ESP8266 auto-bauding is often off by a small amount and this causes errors at higher baud rates (one megabit and up.)

So, some things to note:

* You can confirm this is the case by modifying the line in esptool.py:565 here:
```
esp.run_stub(json.loads(_CESANTA_FLASHER_STUB), [baud_rate], read_output=False)
```
to
```
esp.run_stub(json.loads(_CESANTA_FLASHER_STUB), [int(baud_rate * 1.006)], read_output=False)
```
... and see if it suddenly works

* 0.6% bit rate mismatch is actually not that bad. I would expect a hardware UART receiver to be able to deal with this. There's no standard for TTL serial, but AFAIK the [generally accepted limit](http://electronics.stackexchange.com/questions/7610/what-is-the-maximum-acceptable-baud-rate-error-for-rs-232-serial-communication) is a few percent mismatch. This implies either the hardware is really intolerant of bit rate mismatch, the signal is bad in some other way, or I've misdiagnosed the underlying cause...

* If you transmit data from the CH340G from a different serial program at nominal 230400bps, what baud rate does the Salae see?

* If you use different baud rates with esptool.py that might be easier to divide from the CH340G master clock (maybe 250000, 300000, 500000), does it suddenly work?

* If you use a different OS (maybe run a Linux VM with USB passthrough), is the baud rate still off by the same amount?

Angus

projectgus, 2016-11-25:
Another thought: maybe the OS X driver isn't changing the baud rate in time, so it's not a clock mismatch but just that it's still listening at 115200bps at that point? The gap between the last 115200bps interaction and the 230400bps OHAI is only 16.8ms.

If this is the bug, something you can try is checkout the `feature/esp32_v20_refactor` branch (flashing works on ESP8266) and ESP32). This branch does baud rate setting in a different way, and the 230400bps interaction is initiated by the host not the ESP.

sheinz, 2016-11-26:
Hi Angus,

Wow! Thanks for such thorough analysis!
>You can confirm this theory by modifying the line in esptool.py:565 here:
esp.run_stub(json.loads(_CESANTA_FLASHER_STUB), [baud_rate], read_output=False)
to
esp.run_stub(json.loads(_CESANTA_FLASHER_STUB), [int(baud_rate * 1.006)], read_output=False)

That didn't help.

> If you use different baud rates with esptool.py that might be easier to divide from the CH340G master clock (maybe 250000, 300000, 500000), does it suddenly work?

Nope. Doesn't work.

> If you transmit data from the CH340G from a different serial program at nominal 230400bps, what baud rate does the Salae see?

I will check that. Haven't tried it yet.

> If you use a different OS (maybe run a Linux VM with USB passthrough), is the baud rate still off by the same amount?

I will also check that later.

> If this is the bug, something you can try is checkout the feature/esp32_v20_refactor branch (flashing works on ESP8266 and ESP32). This branch does baud rate setting in a different way, and the 230400bps interaction is initiated by the host not the ESP.

I tried this branch and it works with 230400bps but not higher.
```
$ esptool.py -p /dev/tty.wchusbserial1420 --baud 460800 write_flash -fs 4MB -fm qio -ff 40m
   0x0 ../../bootloader/firmware_prebuilt/rboot.bin 0x1000 ../../bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/simple.bin
esptool.py v2.0-dev
Connecting...
Detecting chip type... ESP8266
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Attaching SPI flash...
Configuring flash size...

A fatal error occurred: Timed out waiting for packet header
```

I lean towards your last idea regarding slow OSX driver. As the same hardware works fine in other OSs and baud rate mismatch is not that great.

marcelstoer, 2016-12-21:
I observed similar issues with that chip on OS X, see #102.

projectgus, 2016-12-21:
#102 looks like a different issue, in that case the driver is rejecting the baud rate and refusing to set it (generating the OSError exception). The bug in that case is that esptool doesn't report the driver failure to the user in a clear way.

In this issue, the driver is accepting the baud rate but then something is going wonky at the hardware level shortly after that.

diimdeep, 2017-04-13:
I have board with CH340G (Esp8266 NodeMCU clone from Lolin vendor)
Only 115200 works.

```
MacBook-Pro:src_esp8266_lolin_nodemcu_v3 wolf$ esptool.py --port /dev/cu.wchusbserial1410 --baud 230400 write_flash --flash_size=detect -fm dio 0 .pioenvs/nodemcuv2/firmware.bin
esptool.py v1.3
Connecting....
Auto-detected Flash size: 32m
Running Cesanta flasher stub...

A fatal error occurred: Invalid head of packet ('\xbc')
MacBook-Pro:src_esp8266_lolin_nodemcu_v3 wolf$ esptool.py --port /dev/cu.wchusbserial1410 --baud 460800 write_flash --flash_size=detect -fm dio 0 .pioenvs/nodemcuv2/firmware.bin
esptool.py v1.3
Connecting....
Auto-detected Flash size: 32m
Running Cesanta flasher stub...

A fatal error occurred: Invalid head of packet ('\x04')
MacBook-Pro:src_esp8266_lolin_nodemcu_v3 wolf$ esptool.py --port /dev/cu.wchusbserial1410 --baud 921600 write_flash --flash_size=detect -fm dio 0 .pioenvs/nodemcuv2/firmware.bin
esptool.py v1.3
Connecting....
Auto-detected Flash size: 32m
Running Cesanta flasher stub...

A fatal error occurred: Invalid head of packet ('\xcd')
MacBook-Pro:src_esp8266_lolin_nodemcu_v3 wolf$ esptool.py --port /dev/cu.wchusbserial1410 --baud 1500000 write_flash --flash_size=detect -fm dio 0 .pioenvs/nodemcuv2/firmware.bin
esptool.py v1.3
Connecting....
Auto-detected Flash size: 32m
Running Cesanta flasher stub...

A fatal error occurred: Invalid head of packet ('\xfd')
MacBook-Pro:src_esp8266_lolin_nodemcu_v3 wolf$ esptool.py --port /dev/cu.wchusbserial1410 --baud 115200 write_flash --flash_size=detect -fm dio 0 .pioenvs/nodemcuv2/firmware.bin
esptool.py v1.3
Connecting....
Auto-detected Flash size: 32m
Running Cesanta flasher stub...
Flash params set to 0x0240
Wrote 360448 bytes at 0x0 in 31.3 seconds (92.2 kbit/s)...
```

projectgus, 2018-04-12:
Closing this issue as I think it's probably fixed now, and if not fixed then very old.

The problem using macOS with CH340G should be fixed in the latest release & master branch, as of 2ec0bc0d58.

There was also some discussion about this problem in the PR #247, and another possible (unconfirmed) fix here: https://github.com/espressif/esptool/tree/experiment/ch340_timeout_bug - although I think the  commit linked above was enough to resolve the problem.

*If you have current esptool.py and are still getting failures then please open a new issue with the details rather than commenting on this one*.

closed 2018-04-12 by projectgus