espressif/esptool #147: Ability to set the timeout on esptool.py while waiting for ESP32 to enter flash mode.

nkolban, 2016-11-25:
I have an issue with my ESP32 module (DevKitC) that causes it not to automatically switch to flash mode when I run esptool.py.  This has caused me to manually switch the ESP32 into flash mode.  What I find is that I run esptool.py and I have some number of seconds (10?) to manually switch the ESP32 into its flash mode.  What I'd like is an option on esptool.py (if possible) to specify the timeout period before esptool.py gives up trying to flash the device.

projectgus, 2016-11-25:
Do you have the blue (V1) DevKitC and/or are you running Windows?

For V2 DevKitC on Windows ([known issue here](https://github.com/themadinventor/esptool/issues/136), you can hold down the "BOOT" button while you run esptool.py. I'm not sure if this also works on the V1 DevKitC.

If this isn't helpful then we can look at adding an additional command line option for timeout.

nkolban, 2016-11-25:
I am using DevKitC V2 (Black) and couldn't get the reboot recipes to work ... neither changing script nor adding a 2.2uF capacitor.

projectgus, 2016-11-25:
That's unusual. So, Windows is your host OS? If you hold down "BOOT" when running esptool.py, does it work first time?

nkolban, 2016-11-25:
Sorry ... I should have answered more of the original question ...

In my environment, I am running a Windows 10 server ... however ... on top of that, I am running latest Ubuntu using latest VirtualBox.  The DevKitC V2 is connected via USB to the Windows 10 machine which is then passed through to Ubuntu on virtual box.  To workaround the problem (which I had forever), I have a raspberry PI driving the EN and GPIO pins running a Node.js script that causes a flash boot of the ESP32 when I send the PI a REST request through a web page.

projectgus, 2016-11-25:
Wow, that's a complex fix!

* Are you passing through the DevKitC as a raw USB device (under USB in VirtualBox SettingS) or a serial port (under Serial Ports)?
*  If you hold down "BOOT" when running esptool.py (without Raspberry Pi in the loop), does it work first time?

nkolban, 2016-11-26:
Wow ... I'm inept today at answering support questions properly ... sorry my friend.

Physically holding down the BOOT button found on the DevKitC V2 board and, while held, running "make flash" causes an immediate working flash without having to toggle any physical pins.

projectgus, 2016-11-26:
Thanks, and no need to apologise.. but could you answer the other question I asked you in my most recent post, please?

nkolban, 2016-11-26:
Dear gosh ... I did it again didn't I!!  <doh> ... For the connection, in my VirtualBox settings, I am passing through the USB->UART via a USB device and NOT through Serial ports.  To be honest, I didn't know serial port connectivity was an option.

radimkarnis, 2020-12-08:
I’m closing this issue because it has been inactive for a few years and it seems like it's not relevant anymore. Please reopen if further discussion is needed.

closed 2020-12-08 by radimkarnis

forkspawn, 2022-05-27:
esp-idf/components/esptool_py/esptool/esptool.py

Change the value DEFAULT_CONNECT_ATTEMPTS