espressif/esptool #149: Flash SST26VF032B does not work

Testi89, 2016-11-30:
When I try to flash a SST26VF032B from microchip/SST, esptool gets a timeout and flashing is not possible.
I used a SST26VF032B as a substitute for a 4Mb chip on an older ESP12.
Reading the chipID from ESP8266 is working, so my setup seems to be correct.

I can find the SST26VF032B in the flash table, so I am wondering why is it not possible to flash this chip.

projectgus, 2016-12-09:
Unfortunately, the datasheet for the SST26VF032 suggests it's probably not compatible with all of the SPI flash command set used by ESP8266.

* From "Device Operation" section: "To provide backward compatibility to traditional SPI
Serial Flash devices, the device’s initial state after a power-on reset is SPI bus protocol supporting only
Read, High Speed Read, and JEDEC-ID Read instructions. A command instruction configures the
device to Serial Quad I/O bus protocol"
* Does the "esptool.py flash_id" command work? This uses the JEDEC-ID read instruction, so it should work according to the datasheet.
* The documentation for the "Page Program" command (02h) seems to indicate that on this chip this is a Quad I/O programming operation using 4 data lines. ESP8266 writes to flash using this command as a standard SPI operation. Most Quad I/O SPI Flash chips support Page Program (02h) as a standard SPI operation, and have a separate command Quad Page Program (32h) for quad I/O programming, but it seems that this chip does not.

The only solution may be to find a different chip that supports more of the common SPI flash commands.

lukeweston, 2017-06-07:
Is there any documentation out there in the community with "known good" (and known bad) specific Flash silicon tested with the ESP8266 (and ESP32)? There do seem to be a couple of quirky ones out there.

hm2dev, 2017-06-11:
Surprisingly I ran into the same problem as Testi89 at about the same time, as I was unable to obtain the W25Q32 from my suppliers.
I'm now testing an ISSI IS25WP032 as replacement for the in the Espressif reference schematics. 
This chip is also mentioned here https://esp32.com/viewtopic.php?f=2&t=826 as having been spotted on an ESP32 module.
Another option seems to be the GigaDevice GD25Q32, as mentioned here http://www.lucadentella.it/en/2016/12/22/esp32-4-flash-bootloader-e-freertos/

I' not sure which flash_chips.h Testi89 is referring to, the only one I can find is from coreboot.
https://code.coreboot.org/p/flashrom/source/tree/HEAD/tags/0.9.8/flashchips.h
If it is the right one, that might be a source for more compatible types.

lukeweston, 2017-06-11:
I recently got a batch of Winbond 25Q32FVSIG and they don't work. (On ESP8266 not the '32.)
Maybe I just got a bad batch of silicon. (They are cheap AliExpress ones.)

I didn't have any other chips in stock, so I popped the can off an ESP12 and removed the BergMicro 25Q32ASSIG and swapped it onto one of my boards - and this works fine.
But without a datasheet for the latter it's hard to say if there are differences.

closed 2018-04-12 by projectgus