espressif/esptool #151: No longer working with Arduino IDE

TijuanaKez, 2016-12-18:
I want to flash NodeMCU 1.0 via Arduino IDE and is recommend to use esptool.py on macOS as C version fails. 
But now esptool.py also fails.

usage: esptool [-h] [--port PORT] [--baud BAUD]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash,verify_flash,erase_flash,version}
               ...
esptool: error: argument operation: invalid choice: 'nodemcu' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'chip_id', 'flash_id', 'read_flash', 'verify_flash', 'erase_flash', 'version')
esptool: error: argument operation: invalid choice: 'nodemcu' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'chip_id', 'flash_id', 'read_flash', 'verify_flash', 'erase_flash', 'version')

projectgus, 2016-12-19:
It looks like the command that Arduino IDE is using to invoke esptool.py is wrong. If you turn on "Verbose Upload" in Arduino Preferences then it may give more details to debug this.

How did you modify Arduino IDE in order to use esptool.py?

projectgus, 2017-01-30:
I'm going to close this, but feel free to reopen if you have some more information about the command that the Arduino IDE is using.

closed 2017-01-30 by projectgus