espressif/esptool #154: Question: auto detection for flash mode feasible?

marcelstoer, 2016-12-21:
Your [docs](https://github.com/espressif/esptool#flash-mode---flash_mode--fm) state

> Some ESP8266 modules, including the ESP-12E modules on some (not all) NodeMCU boards, are dual I/O and the firmware will only boot when flashed with `--flash_mode dio`.

Yet, the [NodeMCU firmware docs](http://nodemcu.readthedocs.io/en/latest/en/flash/#esptoolpy) claim 

> `qio` for 512 kByte modules and `dio` for >=4 MByte modules (`qio` might work as well, YMMV).

So, if that _were_ true, it slightly contradicts your docs, then it would be possible for esptool.py to set the flash mode automatically once the flash size has been detected. Correct?

projectgus, 2016-12-21:
I don't know how the NodeMCU docs came to that conclusion, it doesn't make a lot of sense to me.

One thing that is true is that any flash configuration which works in Quad I/O mode (QIO) should also work in Dual I/O mode (DIO), although not necessarily the other way around. Read performance will be sub-optimal running a QIO-supported flash in DIO mode (reads use two I/O lines instead of four, so the data transfer rate is halved). We could maybe change the default mode in esptool.py v2.0 to the safer-but-slower 'dio'.

It is also theoretically possible for stub code (or a software bootloader) to auto-detect flash mode in most cases, via a combination of factors:
* Some flash IDs can be recognised as requiring a particular read mode.
* Read some data in QIO mode and compare it to the same data read in DIO mode. Most modules that are wired in DIO-only mode will give junk reads for half the bits in QIO mode. This is not foolproof, it will give a false-positive if the SPI flash is wired up in quad mode but the additional data pins are also connected to something else (like reused for external peripherals). But it could be useful in a lot of cases.

Would require someone to do the work to support such a detection mode, though.

marcelstoer, 2016-12-22:
Thanks Angus.

closed 2016-12-22 by marcelstoer