espressif/esptool #155: timeout error on master (then command completes)

sarnold, 2016-12-21:
At least as far as query commands, I just got it so I don't have anything to flash it with yet...  It looks clean with 1.2.1 but master always returns a timeout error before retrying and giving the result:

```$ esptool.py --port /dev/ttyUSB1  chip_id
esptool.py v1.3-dev
Connecting...
Couldn't connect. [<class '__main__.FatalError'>; Timed out waiting for packet header]. Retrying...
Chip ID: 0x00c6ddf1
```

sarnold, 2016-12-21:
Like I said, 1.2.1 looks fine:

```$ esptool.py --port /dev/ttyUSB1 flash_id
esptool.py v1.2.1
Connecting...
Manufacturer: ef
Device: 4016
```

projectgus, 2016-12-21:
Thanks for reminding me about this, printing these interim errors was added in #127 and I meant to do something about it as it's a bit confusing. The previous versions are also seeing a timeout and then retrying, but they don't print anything.

sarnold, 2016-12-24:
NonFatalError?  ;)

commit 22bba06f (2017-01-06): Connect: Don't print interim errors, print final error if failing

Also prints a '.' each time it retries.

Closes #155
closed 2017-01-06 by projectgus