espressif/esptool #160: Verification fails on written/verified flash

thespooler, 2017-01-03:
After write_flash with --verify saying OK,
verify_flash with same file fails.
I tried erase_flash, write_flash again, still verifies OK in the write pass, but a second verify_flash fails again.

Am I using it wrong?
Fresh builds of esptool and micropython masters if it maters. 

```
Spoolers-MacBook:Downloads spooler$ python3 /Volumes/case-sensitive/esptool/esptool.py --port /dev/cu.SLAB_USBtoUART --baud 115200 write_flash 0 /Volumes/case-sensitive/micropython/esp8266/build/firmware-combined.bin --verify
esptool.py v1.3-dev
Connecting...
Couldn't connect. [<class '__main__.FatalError'>; Timed out waiting for packet header]. Retrying...
Couldn't connect. [<class '__main__.FatalError'>; Timed out waiting for packet header]. Retrying...
Couldn't connect. [<class '__main__.FatalError'>; Timed out waiting for packet header]. Retrying...
Couldn't connect. [<class '__main__.FatalError'>; Timed out waiting for packet header]. Retrying...
Couldn't connect. [<class '__main__.FatalError'>; Timed out waiting for packet header]. Retrying...
Auto-detected Flash size: 32m
Running Cesanta flasher stub...
Flash params set to 0x0040
Wrote 573440 bytes at 0x0 in 49.6 seconds (92.4 kbit/s)...
Leaving...
Verifying just-written flash...
Verifying 0x8b77c (571260) bytes @ 0x00000000 in flash against /Volumes/case-sensitive/micropython/esp8266/build/firmware-combined.bin...
-- verify OK (digest matched)
Spoolers-MacBook:Downloads spooler$ python3 /Volumes/case-sensitive/esptool/esptool.py --port /dev/cu.SLAB_USBtoUART verify_flash 0 /Volumes/case-sensitive/micropython/esp8266/build/firmware-combined.bin 
esptool.py v1.3-dev
Connecting...
Couldn't connect. [<class '__main__.FatalError'>; Timed out waiting for packet header]. Retrying...
Couldn't connect. [<class '__main__.FatalError'>; Timed out waiting for packet header]. Retrying...
Couldn't connect. [<class '__main__.FatalError'>; Timed out waiting for packet header]. Retrying...
Couldn't connect. [<class '__main__.FatalError'>; Timed out waiting for packet header]. Retrying...
Couldn't connect. [<class '__main__.FatalError'>; Timed out waiting for packet header]. Retrying...
Running Cesanta flasher stub...
Verifying 0x8b77c (571260) bytes @ 0x00000000 in flash against /Volumes/case-sensitive/micropython/esp8266/build/firmware-combined.bin...
-- verify FAILED (digest mismatch)

A fatal error occurred: Verify failed.
```

projectgus, 2017-01-03:
The flash size & mode are recorded in the 8 byte header at offset 0. When writing to flash at this offset, esptool will change the flash size and mode to match the passed in parameters. The initial verify knows this, and can account for it.

The second verify command doesn't know this, and because these bytes are different verification fails.

This is not ideal, but I'm not sure what the best way around it is.

pfalcon, 2017-01-03:
@projectgus : Maybe after all analyze address passed to verify_flash, and skip bytes at addresses 2, 3.

I'd also recommend to remove over-careless note from write_flash's --verify:

~~~
  --verify              Verify just-written data (only necessary if very
                        cautious, data is already CRCed
~~~

I'd suggest "(only necessary if very cautious, data is already CRCed" (note - missing closing parens) to be removed - the presupposition that "data is CRCed" will save from broken flash or random error in it is misguided, and there *are* cases of mis-flashes. (One particular I had yesterday was done without  --verify, so I can't say how much it's helpful (== implemented well to catch the errors), so I'd actually recommend to make --verify the default, and add --no-verify). Thanks.

thespooler, 2017-01-03:
To default to --verify seems sane.
As for skipping the header, the code seems to indicate that the first two bytes should be skipped, that is, offset 0 and 1.
So this code https://github.com/espressif/esptool/blob/master/esptool.py#L1022-L1023:
```
if address == 0 and (image[0] == b'\xe9' or image[0] == 0xE9) and flash_params is not None:
            image = image[0:2] + flash_params + image[4:]
```
Could just be replaced with a simpler skip.
I'll whip a PR soon.

thespooler, 2017-01-03:
PR ready #161

thespooler, 2017-01-03:
#162 should be good. 

@projectgus, I just skipped the first four bytes, not sure if it would be a good idea to add more checks since I have no clue as to what bytes 0 and 1 are. I guessed magic numbers and skipped all 4. If I'm wrong on that and they should be hashed, I'll need to rework to copy the whole image array, zero the flash_params bytes and rehash that, same with image.

Also, as @pfalcon mentioned, I made --verify the default, with a new switch for --no-verify.

pfalcon, 2017-01-03:
> I have no clue as to what bytes 0 and 1 are. I guessed magic numbers and skipped all 4.

They are static signature and number of "segments" to load into RAM. Skipping all 4 bytes would qualify as a quick solution. But fairly speaking, as they come from user image (not changed by esptool) and there mismatch is fatal, then checking them would be right way to achieve 100% verification.

I suggest waiting for @projectgus feedback though before making further changes.

thespooler, 2017-01-03:
So only the flash_params are not present in the image file?

pfalcon, 2017-01-03:
Or more specifically, esptool may adjust them based on command line args (or their defaults).

projectgus, 2017-01-04:
Thanks @pfalcon, @thespooler for the suggestion & PR.

After thinking about it some more, I'd prefer to add the --flash_size and -flash_mode parameters to the verify_flash command, and expect the correct bytes when verifying at offset 0x0. Essentially have it work the same way as when calling write_flash. Most of the code for dealing with this is already in esptool.py and it may even simplify some of the logic. @thespooler , are you interested in sending a PR of this kind?

BTW, the overall write-verification approach is getting better in esptool v2.0. The v1.x stub code calculates MD5 of the data sent over serial when writing flash, so any serial corruption is always detected but not flash corruption. The v2.0 code calculates MD5 of the data read back from flash after writing, so it will detect both kinds of corruption.

thespooler, 2017-01-04:
I will take a look at it in the next few days, just don't expect anything before friday.
But from a user standpoint, it seems more complex. 
More options I'm required to provide correctly to get a pass/fail. 
But maybe correctness is more important than simplicity in this case, dunno really.

pfalcon, 2017-01-04:
> I'd prefer to add the --flash_size and -flash_mode parameters to the verify_flash command. Essentially have it work the same way as when calling write_flash.

Please mind that users are not expected to pass those params to write_flash. Well, maybe you expect them, but I personally don't, nor do that myself.

projectgus, 2017-01-04:
> Please mind that users are not expected to pass those params to write_flash

I guess that's a sign we have sensible defaults, at least now that the detection behaviour is in. :)

This still gives consistent behaviour if they are added to verify_flash - provided the same defaults apply to both commands, the same values will be expected when verifying.

closed 2017-01-06 by projectgus

projectgus, 2017-01-06:
When you get a chance, please try the latest master branch. As long as the same --flash_size and --flash_mode arguments (if any) are passed to verify_flash as were used for write_flash, the verification result will be the same.

thespooler, 2017-01-06:
Looks good. 
Would you like me to submit a separate PR for the --verify as default and --no-verify?

Also, it's quite good you took this, as my NodeMCU just went out in magic black smoke, I'll have to wait for a new one to test anything.

commit bbdeb668 (2017-01-06): verify_flash: Make behaviour same as write_flash --verify

Specifically, when flashing offset 0 --flash_mode and --flash_size are
applied via verify_flash, same as with write_flash.

Closes #160