espressif/esptool #164: esptool for esp32, make app problem

michal037, 2017-01-05:
Hello, 
   i try to compile 'myapp' project for esp32. I have prepared esp-idf and xtensa for esp32, all working.
When i go into the directory 'myapp' and write last command 'make app', esptool print this error:

`LD app-template.elf
esptool.py v2.0-dev
Traceback (most recent call last):
  File "/home/michal/esp32/esp-idf/components/esptool_py/esptool/esptool.py", line 2177, in <module>
    main()
  File "/home/michal/esp32/esp-idf/components/esptool_py/esptool/esptool.py", line 1993, in main
    operation_func(args)
  File "/home/michal/esp32/esp-idf/components/esptool_py/esptool/esptool.py", line 1624, in elf2image
    e = ELFFile(args.input)
  File "/home/michal/esp32/esp-idf/components/esptool_py/esptool/esptool.py", line 1274, in __init__
    with open(self.name, 'rb') as f:
IOError: [Errno 2] No such file or directory: '/home/michal/esp32/myapp/build/app-template.elf'
/home/michal/esp32/esp-idf/components/esptool_py/Makefile.projbuild:49: commands for object '/home/michal/esp32/myapp/build/app-template.bin' failed
make: *** [/home/michal/esp32/myapp/build/app-template.bin] Error 1
michal@debian:~/esp32/myapp$ 
`

All operations, I do under Debian Jessie 8.3 64bits. What can I do with this, to solve this problem ?

projectgus, 2017-01-06:
Thanks for reporting this. It seems like there's two bugs here, esptool shouldn't be printing such an enormous stack trace for "IOError: [Errno 2] No such file or directory: '/home/michal/esp32/myapp/build/app-template.elf"

Then there's what looks like it might be an esp-idf bug, the ELF file doesn't exist at the time elf2image is being called.

Can you please run 'make V=1' and attach the output as a log file to a comment here (or post it on gist.github.com or another service)?

michal037, 2017-01-06:
`michal@debian:~/Pulpit$ cd ..
michal@debian:~$ cd esp32
michal@debian:~/esp32$ ls
esp-idf  myapp  xtensa-esp32-elf
michal@debian:~/esp32$ cd myapp
michal@debian:~/esp32/myapp$ make V=1
including /home/michal/esp32/esp-idf/components/bootloader/Makefile.projbuild...
including /home/michal/esp32/esp-idf/components/bootloader_support/Makefile.projbuild...
including /home/michal/esp32/esp-idf/components/esp32/Makefile.projbuild...
including /home/michal/esp32/esp-idf/components/esptool_py/Makefile.projbuild...
including /home/michal/esp32/esp-idf/components/mbedtls/Makefile.projbuild...
including /home/michal/esp32/esp-idf/components/nghttp/Makefile.projbuild...
including /home/michal/esp32/esp-idf/components/partition_table/Makefile.projbuild...
make -C /home/michal/esp32/esp-idf/components/bootloader/src V=1 BUILD_DIR_BASE=/home/michal/esp32/myapp/build/bootloader TEST_COMPONENTS= /home/michal/esp32/myapp/build/bootloader/bootloader.bin
make[1]: Wejście do katalogu '/home/michal/esp32/esp-idf/components/bootloader/src'
including /home/michal/esp32/esp-idf/components/esptool_py/Makefile.projbuild...
including /home/michal/esp32/esp-idf/components/bootloader/Makefile.projbuild...
including /home/michal/esp32/esp-idf/components/bootloader_support/Makefile.projbuild...
make -C /home/michal/esp32/myapp/build/bootloader/bootloader_support -f /home/michal/esp32/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/home/michal/esp32/esp-idf/components/bootloader_support/component.mk COMPONENT_NAME=bootloader_support build
make[2]: Wejście do katalogu '/home/michal/esp32/myapp/build/bootloader/bootloader_support'
xtensa-esp32-elf-gcc -std=gnu99 -Os -ggdb -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -std=gnu99 -Os -ggdb -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -I/home/michal/esp32/myapp/build/bootloader/../include -D BOOTLOADER_BUILD=1 -I /home/michal/esp32/esp-idf/components/esp32/include -DESP_PLATFORM -MMD -MP -DESP_PLATFORM -MMD -MP   -DNDEBUG  -DNDEBUG -I /home/michal/esp32/esp-idf/components/bootloader_support/include -I /home/michal/esp32/esp-idf/components/bootloader_support/include_priv -I /home/michal/esp32/esp-idf/components/bootloader_support/include_priv -I /home/michal/esp32/esp-idf/components/log/include -I /home/michal/esp32/esp-idf/components/spi_flash/include -I /home/michal/esp32/esp-idf/components/micro-ecc/micro-ecc -I /home/michal/esp32/esp-idf/components/bootloader/src/main/include -I /home/michal/esp32/myapp/build/bootloader/include  -Isrc -c /home/michal/esp32/esp-idf/components/bootloader_support/src/bootloader_flash.c -o src/bootloader_flash.o
xtensa-esp32-elf-gcc -std=gnu99 -Os -ggdb -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -std=gnu99 -Os -ggdb -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -I/home/michal/esp32/myapp/build/bootloader/../include -D BOOTLOADER_BUILD=1 -I /home/michal/esp32/esp-idf/components/esp32/include -DESP_PLATFORM -MMD -MP -DESP_PLATFORM -MMD -MP   -DNDEBUG  -DNDEBUG -I /home/michal/esp32/esp-idf/components/bootloader_support/include -I /home/michal/esp32/esp-idf/components/bootloader_support/include_priv -I /home/michal/esp32/esp-idf/components/bootloader_support/include_priv -I /home/michal/esp32/esp-idf/components/log/include -I /home/michal/esp32/esp-idf/components/spi_flash/include -I /home/michal/esp32/esp-idf/components/micro-ecc/micro-ecc -I /home/michal/esp32/esp-idf/components/bootloader/src/main/include -I /home/michal/esp32/myapp/build/bootloader/include  -Isrc -c /home/michal/esp32/esp-idf/components/bootloader_support/src/efuse.c -o src/efuse.o
xtensa-esp32-elf-gcc -std=gnu99 -Os -ggdb -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -std=gnu99 -Os -ggdb -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -I/home/michal/esp32/myapp/build/bootloader [... cut ...]
/home/michal/esp32/esp-idf/components/esptool_py/Makefile.projbuild:49: commands for objects '/home/michal/esp32/myapp/build/bootloader/bootloader.bin' failed
make[1]: *** [/home/michal/esp32/myapp/build/bootloader/bootloader.bin] Error 1
make[1]: Leaving directory '/home/michal/esp32/esp-idf/components/bootloader/src'
/home/michal/esp32/esp-idf/components/bootloader/Makefile.projbuild:31: commands for objects '/home/michal/esp32/myapp/build/bootloader/bootloader.bin' failed
make: *** [/home/michal/esp32/myapp/build/bootloader/bootloader.bin] Error 2
michal@debian:~/esp32/myapp$ 
`

projectgus, 2017-01-06:
Please edit your coment and attach this data as a file, or post it on a pastebin service or gist.github.com and edit the comment to contain a link.

michal037, 2017-01-06:
Sorry for this, https://github.com/michal037/esptool-log/blob/master/esptool.log

projectgus, 2017-01-06:
Thanks. I didn't think through my request properly, you can do a 'make clean' and then 'make V=1' and update the log with that output? Thanks.

michal037, 2017-01-06:
output: https://github.com/michal037/esptool-log/blob/master/esptool_log2.log

projectgus, 2017-01-06:
Something very odd is happening here:

This line should link a bootloader ELF, output to `/home/michal/esp32/myapp/build/bootloader/bootloader.elf`

```
xtensa-esp32-elf-gcc -nostdlib -L/home/michal/esp32/myapp/build/bootloader/esptool_py -L/home/michal/esp32/myapp/build/bootloader/bootloader -L/home/michal/esp32/myapp/build/bootloader/bootloader_support -L/home/michal/esp32/myapp/build/bootloader/log -L/home/michal/esp32/myapp/build/bootloader/spi_flash -L/home/michal/esp32/myapp/build/bootloader/micro-ecc -L/home/michal/esp32/myapp/build/bootloader/main -u call_user_start_cpu0  -Wl,--gc-sections -Wl,-static -Wl,--start-group  -lbootloader_support -llog -lspi_flash -lmicro-ecc -L /home/michal/esp32/esp-idf/components/bootloader/src/main -lmain -T esp32.bootloader.ld -T /home/michal/esp32/esp-idf/components/esp32/ld/esp32.rom.ld -lgcc -Wl,--end-group -Wl,-EL -o /home/michal/esp32/myapp/build/bootloader/bootloader.elf -Wl,-Map=/home/michal/esp32/myapp/build/bootloader/bootloader.map
```

It seems to succeed, and then the very next line refers to the same ELF file path:

```
python /home/michal/esp32/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 elf2image --flash_mode "dio" --flash_freq "40m" --flash_size "16MB"  -o /home/michal/esp32/myapp/build/bootloader/bootloader.bin /home/michal/esp32/myapp/build/bootloader/bootloader.elf
```

Which causes an error because the file doesn't exist.

* Does the build/bootloader directory exist?
* Can you find a bootloader.elf anywhere in your build directory?
* What toolchain are you using and how did you compile/install it?
* Is there anything in /home/michal/esp32/myapp/build/bootloader/bootloader.map

michal037, 2017-01-06:
1) bootloader directory exist: only a few directores named: 
`michal@debian:~/esp32/myapp/build/bootloader$ ls`
`bootloader_support  log  main  micro-ecc  spi_flash`

2) He's nowhere, nothing has been compiled.

3) I used last toolchain, installing step by step from:
`https://github.com/espressif/esp-idf/blob/master/docs/linux-setup.rst`
I decided to not compile xtensa-esp32-elf, I downloaded ready for use binaries.

4) bootloader.map don't exist.

----
I perform all the steps in Oracle Virtual Box. Where I install clearly debian from file named: `debian-8.6.0-amd64-CD-1.iso`. 
Step by step using `Set up of Toolchain for Linux` guide, I trying to prepare this toolchain.

I decided to pack this files and upload it on dropbox for You.
`https://www.dropbox.com/s/8m6nuxfzcvjf02r/esp32.tar.gz?dl=0`
I using this files, with this commands in `~/.bashrc` file:
`export PATH=$PATH:$HOME/esp32/xtensa-esp32-elf/bin`
`export IDF_PATH=~/esp32/esp-idf`

Nothing more than that.

michal037, 2017-01-07:
I was able to do everything correctly, very strange error. Again I downloaded all over again + I do system upgrade. Now it works properly.

closed 2017-01-07 by michal037

projectgus, 2017-01-08:
Very unusual, but I'm glad you got it working OK.