espressif/esptool #165: Cannot erase or flash beyond 256k boundry on 1MB(8m) ESP-01

jpenner64, 2017-01-06:
I’d like to think this is NOT a unique problem, however after searching everywhere I am at a loss. This seems close to a corrupted SPIEraseArea issue, but using latest esptool.py 1.2.1 and would like at least a second opinion.

Symptom: When attempting to erase or flash using esptool.py, the function aborts with the following error; “A fatal error occurred: Timed out waiting for packet header”. On an erase, there is no byte count, however on a flash the byte count stops at 262144; (256k boundary)

Hardware: On Windows 10, I am trying to flash an 8m (1MB) “AI-Cloud Inside” ESP-01 (black) and using various USB>TTY devices; like CP2102 USB>TTL, or OpenSmart ESP-01 adapter.

Background: When I first received these modules they were working fine and I believe the issue started when I attempted to reflash. I have reflashed many other older ESP-01s (blue/black) but these two are problematic for me now.

Examples:
•	C:\Users\jpenner>esptool.py -p COM9 -b 115200 write_flash 0x000000 C:\Python27\ESPbins\nodemcu_latest.bin
o	esptool.py v1.2.1
o	Connecting...
o	Auto-detected Flash size: 8m
o	Running Cesanta flasher stub...
o	Flash params set to 0x0020
o	Writing 405504 @ 0x0... 262144 (64 %)
o	A fatal error occurred: Timed out waiting for packet header
•	C:\Users\jpenner>esptool.py -p COM9 -b 115200 write_flash -fs 8m 0x000000 C:\Python27\ESPbins\blank_1MB.bin
o	esptool.py v1.2.1
o	Connecting...
o	Running Cesanta flasher stub...
o	Writing 1048576 @ 0x0... 262144 (25 %)
o	A fatal error occurred: Timed out waiting for packet header 
Observations:
•	Installing any bin greater than 256k will error, and consistently stop at 262144
•	Different version and bins files from Ai-Thinker, Espressif, etc all show the same symptom
•	Using esptool.py with the write_flash option “-fs 8m” to specifically state 1MB does not change results as related to a similar issue.
•	Other flash tools result in the same failure, timing out after a flash erase. Presumably hitting at the same 256k boundary marker as esptool.py
o	Ai Thinker flash tool	esp8266_flasher.exe			erase before flash
o	Espressif flash tool	ESPFlashDownloadTool_v3.4.4.exe 	erase before flash
o	NodeMCU flash tool	ESP8266Flasher.exe (x32 or x64) 	erase before flash
o	Python flash tool	esptool.py				erase or flash
Assumptions: 
•	SPIEraseArea corruption issue caused by my initial erase before reflash. If so, how can I go about recovering?
•	Possible hardware damaged to extended storage area has occurred
•	Or….

jpenner64, 2017-01-06:
Here is more specifically what I have:

Only silkscreen on PCB is the “Ai-Cloud inside”. On the module ICs the etchings are both:
•	Controler IC	Espressif ESP8266EX, 322015, P3PU84		(COM7 & COM9 for reference)
•	Flash ROM IC	BergMicro 25O80ASSIG					(COM7 & COM9 for reference)
TTLs currently are
•	Silicon Labs CP210x USB to UART Bridge (COM7)
•	USB-SERIAL CH340 (COM9)
Flash IDs as follows:
•	C:\Users\jpenner>esptool.py -p COM7 flash_id
o	esptool.py v1.2.1
o	Connecting...
o	Manufacturer: e0
o	Device: 4014
•	C:\Users\jpenner>esptool.py -p COM9 flash_id
o	esptool.py v1.2.1
o	Connecting...
o	Manufacturer: e0
o	Device: 4014
Size interpretation:
•	Although 25O80ASSIG is not listed in a flashchips.h reference I am taking the 80 as part of IC etching and the 4014 as signs this is the 8m I know it to be. 
•	Newer modules purchased as 8m, and prior versions owned were not below 4m

projectgus, 2017-01-08:
Thanks for these details. How are you powering your ESP-01?

jpenner64, 2017-01-09:
With your question, I thought I should also run through some examples. All of these results netted nothing new/positive, but it was a good try and run through:
•	Normally, powered from USB 3.0 port on Windows 10
•	Initially, powered LaptopUSB>TTL>ESP-01. Later, introduced a USB2 MultiPort extension;  LaptopUSB>Multi port(USB2)>TTL>ESP-01. Both results the same; negative flashes up to 262144
Introducing a V/A detection device (keweisi) the connection is as follows, and results below:
•	Multiport - LaptopUSB3>Multi port(USB2)>keweisi>TTL>ESP-01:
o	ESP-01 #1 – TTL CH340 	4.77v/0.04-0.05A(Connected/Flashing)
o	ESP-01 #2 – TTL CP210x	4.78v/0.04-0.05A(Initial connection/Flashing), 4.84v/0.02A(TTL-Idle)
•	Direct - LaptopUSB3>keweisi>TTL>ESP-01
1.	ESP-01 #1 – TTL CH340 	4.97v/0.04-0.05A(Connected/Flashing)
2.	ESP-01 #2 – TTL CP210x	4.97v/0.04-0.05A(Initial connection/Flashing), 0.02A(TTL-Idle)
•	No V/A monitoring - LaptopUSB3>TTL>ESP-01
1.	ESP-01 #1 – TTL CH340 	Power consumption unknown
2.	ESP-01 #2 – TTL CP210x	Power consumption unknown
•	All cases results are the same. Flash will process up to 262144 and stop; “A fatal error occurred: Timed out waiting for packet header”
Using an older laptop; (does not have Python on, but) tested Ai-Thinker flash. This process will not give me a byte count, and attempts an erase before flash.
•	Erase/Flash and Power recorded as follows:
1.	ESP-01 #1 – TTL CH340 	Invalid head of packet, 4.88v/0.04-0.05A(Connected/Flashing)
2.	ESP-01 #2 – TTL CP210x	Invalid head of packet, 4.97v/0.04-0.05A(Initial connection/Flashing), 0.02A(TTL-Idle)

projectgus, 2017-01-09:
It doesn't seem to be that the flash has accidentally had write protection turned on, as [Berg B25Q80](http://www.chinan-sx.com/uploadfile/cfile/2010119111110143.pdf) doesn't have a write protection option which would start at the 256KB boundary.

The 3.3V regulators on TTL serial adapters may not provide enough current to reliably power the ESP8266. When the flash chip performs a block erase, it and/or the ESP8266 may be glitching the 3.3V power line. Over long periods, the DC current is relative low but the consumption is spiky - it can be hard to see these spikes using "slow" current monitoring devices.

There's a [blog post here](http://esp8266flashissuesandfixed.blogspot.com.au/2016/03/esp8266-01-ft232-usbttl-module-1.html) that talks about changing the ESP_FLASH_BLOCK value in order to successfully flash an ESP-01 with this flash chip. This change has been suggested by people multiple times, but so far on investigation it's turned out to be working around a power problem (writing in smaller chunks smooths out the power draw), so even if flashing succeeds this way it's likely that further operations will fail (WiFi causes even larger spikes of current consumption.)

Try powering the ESP-01 from a different 3.3V power source. There are some tips in the README here:
https://github.com/espressif/esptool#insufficient-power

jpenner64, 2017-01-09:
Thanks for the suggestions here....

Mods to the ESP_RAM_BLOCK and ESP_FLASH_BLOCK can be helpful and I have made necessary changes to allow flashing of other boards successfully. Regardless, playing around with the numbers further is something to try here. Also, powering via a dedicated source to support amp draw spikes is something else to try. I know have had problems with operating the WiFi successfully when powered through other modules; but never on just flashing.

My suspicions is now at the point I believe I somehow damaged these 2 modules, but cannot recall how. Regardless, I will go through these suggestions and bring back findings. Fingers crossed.

jpenner64, 2017-01-21:
Follow-up: After a break, I ran through a number of scenarios, which I will not detail here. However, I did double check settings in pytool.py and to ensure ability to flash other identical boards with same ICs have no issue; (Other flash attempts successful), Operationally, I do use ESP-01 with external power for the same reason mentioned.

So I then turned back to these 2 specific boards and last suggestions.: Experimented some more with changing values of ESP_RAM_BLOCK and ESP_FLASH_BLOCK and powered ESP externally to accommodate potential spikes. Same results; fails to continue flashing beyond 256k.

Summary: This chain is centered on issues with two (2) specific boards which could have been damaged. It is not taken as an issue with the pytool.py or in general any ESP-01 8m module. I will continue a little more with testing these 2 modules, but just might write them off as damaged and move on.

Thanks for your help. Much appreciated

closed 2017-01-21 by jpenner64