espressif/esptool #170: Clear

paveleremin, 2017-01-11:
Is it possible to clear my ESP8266 1Mb to default settings?
I mean, leave only boot section.
Then I plan to upload new firmware - EspEasy.

r44083, 2017-01-29:
I subscribe to the issue. It would be very convenient, because regular erasing erases too much.

projectgus, 2017-01-29:
Yes, esptool.py v2.0 (the current master branch) has this feature. You can use erase_region to erase a region with a specified start point and size:

So, to leave the first sector (0x1000 bytes) in place and erase up to the 1Mb boundary:

```
esptool.py erase_region 0x1000 0xff000
```

Note that `esptool.py write_flash` will automatically erase any sectors before writing to them, so you may not need to erase this entire region. However if there is any data that you need erased but are not about to overwrite with new data via `write_flash`, you can use the `erase_region` command to erase it.

closed 2017-01-29 by projectgus