espressif/esptool #171: Make All error "No module named serial"  for ESP32 Wrover

ksmith9109, 2017-01-17:
I am using Windows 7 64 and having problems with the esptool install.
Please provide a detailed install method.  I had no luck with the manual methods in the readme.
Thanks
Ksmith

Traceback (most recent call last):
  File "C:/msys32/sdk/esp-idf/components/esptool_py/esptool/esptool.py", line 24, in <module>
    import serial
ImportError: No module named serial
make[1]: *** [/sdk/esp-idf/components/esptool_py/Makefile.projbuild:49: /sdk/esp-idf/myexamples-01092016/02_blink/build/bootloader/bootloader.bin] Error 1
make: *** [/sdk/esp-idf/components/bootloader/Makefile.projbuild:31: /sdk/esp-idf/myexamples-01092016/02_blink/build/bootloader/bootloader.bin] Error 2

projectgus, 2017-01-30:
Did you follow the ESP-IDF Windows Setup Guide from the docs? http://esp-idf.readthedocs.io/en/latest/windows-setup.html Which option from there are you using, the prebuilt & downloaded MSYS2 environment or one where MSYS is installed manually? How are you launching the shell which you're using to run `make`?

The version of Python that comes in the prebuilt environment has pyserial installed already. However if you have another copy of Python installed system-wide then this may be the version which is being run. If you're using a different shell, rather than the msys_cmd shell link mentioned in the Windows docs, then this can lead to a different Python being run (depending on your Windows path.)

An easy fix is to install pyserial manually into your system Python - you can try running `python -m pip install pyserial`.

You may run into other problems in esp-idf if your system Python is Python 3 not Python 2.x. Running "make menuconfig" -> "SDK Tool Configuration" and setting "Python 2 interpreter" to `python2` may help.

ksmith9109, 2017-01-30:
Dear Angus,

 

To focus on 

“The version of Python that comes in the prebuilt environment has pyserial installed already. However if you have another copy of Python installed system-wide then this may be the version which is being run.”

 

I have installed, renamed the msys32 root directory, and re-installed the software components several times.

 

My current system seems to work but I have noticed the Windows essentials script assumes that python is installed but fails when it tries to run a python script.

 

Also, the Windows essentials script may not work to refresh an existing install.  

 

General PATH problems  mixing the Windows path and the path  environment within msys32 can cause many problems.

 

I am proposing a standalone confirmation  script that enumerates both environments and warns of any general environment mis-configurations.

 

Thanks for your hard work.  I will try and work on the problem myself in future days.

 

Regards

Kent Smith

projectgus, 2017-01-30:
Hi Kent,

Do I understand correctly that you're not using the downloaded pre-configured environment, but following the "Alternative Step 1" with `windows_install_prerequisites.sh`?

There is a bug in this script regarding Python installation, so this is probably the problem you are seeing. Running `python -m pip install pyserial` should solve the problem.

There's a fix pending for this underlying problem with the setup script. (espressif/esp-idf#264).

ksmith9109, 2017-01-30:
Yes, that is the general nature of the problem I was having.   Eventually figured a work around involving getting the windows and msys32 environments to match properly.

A great tool that has helped me a lot is [http://www.esp32.com/viewtopic.php?f=13&t=1050](url)

It allows me to navigate using familiar windows explorer tool but drop into and execute bash commands in various folders.    A well designed aid...

closed 2017-01-30 by ksmith9109

szundi, 2018-03-22:
python -m pip install pyserial
it was the key for me, thank you.

phoolchandra, 2018-04-07:
python -m pip install pyserial
worked for me also.

ahmedawais, 2018-05-31:
it didnt work for me

projectgus, 2018-06-01:
> it didnt work for me

You will need to provide more information so someone can help you. Please supply some more details about what OS/environment you're using, what the full output from each command you tried was, what else you've already tried, etc.

Thanks,

Angus

ahmedawais, 2018-06-01:
issue resolved i was running this command python -m pip install pyserial   in msys2
i again ran this into cmd and now its working fine

ethapay, 2020-05-07:
I am using ubuntu 18.04. It worked for me after I did as per instructions in https://engineerworkshop.com/2019/03/02/esp32-compiler-error-in-arduino-ide-heltec-esp32-tools-esptool-esptool-py-no-module-named-serial-tools-list_ports/