espressif/esptool #175: A fatal error occurred: MD5 of file does not match data in flash!

darcode1, 2017-02-02:
Okay so I have checked tones of online resources and did come across problems similar to mine, tried all those things, still couldnt get it done.

I tried to update my ESP8266 firmware and gives out the following result. 

Please have a look

```
Junaids-MacBook-Pro:esptool-master junaid$ sudo ./esptool.py --port /dev/tty.SLAB_USBtoUART write_flash --flash_mode dio --flash_size 4MB 0x0 ESP8266_NONOS_SDK_V1-3.bin
esptool.py v2.0-beta1
Connecting....
Detecting chip type... ESP8266
Uploading stub...
Running stub...
Stub running...
Attaching SPI flash...
Configuring flash size...
Compressed 3445360 bytes to 3185200...
Wrote 3445360 bytes (3185200 compressed) at 0x00000000 in 281.5 seconds (effective 97.9 kbit/s)...
File  md5: 5e76b987a034b5e079378be9f2cacf70
Flash md5: 93e9b0feabaf850434d4129b3f86d248
MD5 of 0xFF is 5ac6f4629a853cc827aa131bf5832d7b

A fatal error occurred: MD5 of file does not match data in flash!
```

could it be a faulty firmware file ? I am using the one from here: http://bbs.espressif.com/viewtopic.php?f=46&t=2198

Further information:

- Flash_id
```
Junaids-MacBook-Pro:esptool-master junaid$ sudo ./esptool.py --port /dev/tty.SLAB_USBtoUART flash_id
esptool.py v2.0-beta1
Connecting....
Detecting chip type... ESP8266
Uploading stub...
Running stub...
Stub running...
Manufacturer: e0
Device: 4014
Hard resetting...
```
This means, the ESP8266 has GD25Q80B and according to its datasheet [here](http://www.concord-at.com/download/datasheets/GD25Q80B_Rev1.5.pdf) it has 8 M-bit of Serial Flash.

and Flash_status

```
Junaids-MacBook-Pro:esptool-master junaid$ sudo ./esptool.py --port /dev/tty.SLAB_USBtoUART read_flash_status
esptool.py v2.0-beta1
Connecting....
Detecting chip type... ESP8266
Uploading stub...
Running stub...
Stub running...
Status value: 0x0000
Hard resetting...
```

Any help would be highly appreciated. I am lost here.

projectgus, 2017-02-02:
3445360 bytes is just over 3 megabytes. So it can't fit into a 1 megabyte (8 megabit) flash chip. This causes the MD5 error when it comes to verify what it wrote.

If you leave off the `--flash-size 4MB` argument then `esptool.py` will autodetect the flash size and give an error when it sees the firmware won't fit.

To make this less confusing, it's probably best if `esptool.py` runs auto-detection even when a flash-size argument is provided, and prints a warning if the flashing will exceed the detected size.

darcode1, 2017-02-02:
Okay that does make sense. but the problem is my ESP8266 came with the same firmware loaded that i am trying to load on it now. here have a look (old picture).

[image]

I was just playing around with the ESP8266, downgraded the firmware and, worked like a charm but when I tried to upload it to the version above, this happened. Is there any other work around ?

projectgus, 2017-02-02:
I don't understand where you got the file "ESP8266_NONOS_SDK_V1-3.bin" from? Is this a file you compiled from the SDK?

Inside the zip file "ESP8266_NONOS_SDK_V1.5.4_16_05_20.zip", the `bin/at/512+512` directory contains a prebuilt firmware image with an AT firmware binary. The `bin/at/README.md` file has lists of filenames and offsets - under the first heading "Flash size 8Mbit" is the list you need. If you pass those offsets and file names to `esptool.py write_flash`, it should flash the firmware correctly.

projectgus, 2017-02-02:
(BTW, there is difference between Ai-Thinker's firmware and the AT firmware produced by Espressif. Even though the Ai-Thinker firmware was compiled with SDK 1.5.4, I think Ai-Thinker modify it a bit. I don't know what the specific differences are, though.)

darcode1, 2017-02-03:
YES, thank you for your prompt reply. really appreciate it. After sending you that message I was able to figure this out. Spent 6 hours figuring that out. New to ESP8266, so getting used to it.

Thanks

radimkarnis, 2020-12-08:
I’m closing this issue because it has been inactive for a few years and it seems like it's not relevant anymore. Please reopen if further discussion is needed.

closed 2020-12-08 by radimkarnis