espressif/esptool #177: esp8266 frustrations

billiho, 2017-02-11:
Being new to this world of Arduino and the esp8266 chip I found the arduino was awesome and easy to program. Not so much for the esp8266. Nothing but a giant pain the the youknowwhat. I tried uploading the wifiscan example and the chip stopped responding to at commands. Great, now what. I spent too much time on the web looking at all the ways to flash, all the addresses to flash, all the suggestions for wiring, etc, etc. ARRRRGHH!!! Next, I got a second chip. It worked so I tried uploading the blink and all was well. I tried the webclient and it didn't work, I tried a couple others including the wifiscan and now the second chip is in lala land and if I watch com6 I can see it says .

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

All I want to do is get the stupid thing working again. The best I have been able to do is do some flashing with either the exprssif tool v3.4.4 or the esp8266_flasher program.  The problem is the 3.4.4 tool wants a master key location and I can't find any info on that, and I can't find what file goes in the system params section. I have also tried the esp8266_flasher tool with limited success. The chip returns this info from 3.4.4 flasher tool:

flash vendor:
E0h : N/A
flash devID:
4014h
QUAD;8Mbit
crystal:
26 Mhz 

so I set it to 8Mbit and try to set up the file locations but it's not working. Here's my setup and the info 
[image]

I have no idea if this is right as I have not found the magic document that says this is what to do. I HAVE found a bunch of different documents including the info that came with firmware:

AT_v0.50_on_esp_iot_sdk_v1.4.0_150918

nonboot/eagle.flash.bin 0x00000                   //do I put this stuff in or the stuff below this section?
nonboot/eagle.irom0text.bin 0x40000
blank.bin 0x3E000
blank.bin 0x7E000

1024KB Flash
Content Address Size
Boot 0x00000 – 0x00FFF 4KB
App 1 (user1.bin) 0x01000 – 0x7BFFF 492KB
User params 0x7 C000 – 0x7FFFF 16KB
Do not use 0x8 0000 – 0x80FFF 4KB
App 2 (user2.bin) 0x81000 – 0xF BFFF 492KB
System params 0xFC000 – 0xF FFFF 16KB  WTF is this? what file do I use?

I have at this point looked at too many web pages, too many docs, and my brain is dissolving into goo. There must be someone out there who can say "you dummy, put these files in this order and dump it onto the chip"

file1 name and location
file 2 name and location 
etc.

Blink uploads and the blue light flashes for a few milliseconds, not the 1 or 2 seconds it's supposed to do in the example. The second chip is about the same.

Any help would be great. I have hard wired the circuit in and I have enough power and the gpio and reset ground as they should when flashing. I have iused a usb to serial converter and tried thru the arduino.

billiho, 2017-02-11:
Here's an update with trying to flash according to the 3.4.4 flasher program help file that comes up when you hover over the address box section. I have also grabbed what the output looks like.
[image]. I tried this on several different versions with no luck.

billiho, 2017-02-12:
It worked on one so far! Yeah!

I was able to do the nonboot stuff. I tried a ton of times and sometimes it would get up to 99% then say invalid packet. Finally I took the transmit from the rs232 and reran it on my board which didn't make much difference, but I thought it may picking up some stray noise. Then I tried and tried. with little success. GRRRR!!!!

I finally decided"screw it, I'm gonna increase the speed and see if that works." My thinking was the usb interface was able to transmit faster so why not see if the chip could receive as fast. 

Slowly I increased the transmit rate on the 3.4.4 flasher and I had some success as it got faster, I even got the user1.bin file to upload 100% once, and that was the one I was having the biggest issue with. 

I was unable to do all the files at once so I decided I would do them one at a time. It took many tries but slowly one would fully upload and then I went onto the next. I had a really hard time with user 1.bin file and probably tried that 20 times. Finally it stuck as did all the files. I rebooted and tried an at command and it replied. Yahoo!

AT+GMR

AT version:0.51.0.0(Nov 27 2015 13:37:21)
SDK version:1.5.0
compile time:Nov 27 2015 13:57:56
OK
Here's the file I got and the location

AT_v0.50_on_esp_iot_sdk_v1.4.0_150918

http://bbs.espressif.com/viewtopic.php?f=46&t=1442

Here's a screenshot of what worked. My esp is a 1mb version.
[image]

Now onto the next.

projectgus, 2017-02-13:
Thanks for posting the updated information. I'm glad you got things working.

To clarify, the GUI-based Espressif Flash Tool is a different tool to the command line esptool.py which is hosted in this repo. Support for the GUI flash tool (and the ESP8266 SDK) is provided via http://bbs.espressif.com/ .

If you're using the ESP8266 with Arduino, you may also be interested to look into ESP8266 Arduino support:
https://github.com/esp8266/arduino
(This allows you to program the ESP8266 directly with an Arduino sketch written to execute on it, no need for AT commands or similar.)

closed 2017-02-13 by projectgus