espressif/esptool #178: Support make_image subcommand for esp32

young-mu, 2017-02-14:
Now **make_image** subcommand is just for esp8266.

projectgus, 2017-02-27:
This is correct. elf2image works on both architectures, but make_image only works on ESP8266.

make_image is fairly limited command, and even on ESP8266 the images it generates are not always very useful. I'd recommend generating the ELF file you need (via some combination of linker arguments and objcopy commands) and then using elf2image to convert this to a binary.

Is there a particular use case you have in mind that requires make_image?

young-mu, 2017-02-28:
OK, this is my scenario. I want to make a binary file to an image and map it to an specified address (like 0x3FFE0000). So in application, I can access these data of binary file by address 0x3FFE0000.

projectgus, 2017-04-24:
Sorry I didn't write back to you before, I thought I had.

If you're using ESP-IDF then one other way to make this work is to use the "embedding" feature in the IDF build system:
http://esp-idf.readthedocs.io/en/latest/build-system.html#embedding-binary-data

This will allow you to embed the file into the ELF, it'll be mapped to flash cache in .rodata, and you can access it directly as a large read-only buffer in read-only memory.

If you're using a different build system then you can still use the same approach as esp-idf does, using objcopy to embed the file.

I'll keep this feature request open in order to expand make_image, although because the ESP32 has a more complex flash cache it may be harder to implement a useful command.

young-mu, 2017-04-24:
Ok, thank you. Now I use *spi_flash_mmap* API to achieve my goal.

radimkarnis, 2020-12-10:
I’m closing this issue because it has been inactive for a few years and it seems like it's not relevant anymore. Please reopen if further discussion is needed.

closed 2020-12-10 by radimkarnis