espressif/esptool #180: Invalid packet error

grcasanova, 2017-02-25:
I am trying to program an ESP8266-01 WiFi module using an Arduino Uno as a USB to TTL programmer. I have removed the ATmega and grounded the RESET pin (quite useless without the uC though). Then, using esptool.py I tried to upload a Bare minimum sketch with the Arduino IDE and later using the following:

`python esptool.py -p /dev/cu.usbport write_flash 0x000000 "nodemcu.bin"
`
However, both ways give me the same output:

`A fatal error occurred: Failed to connect to ESP8266: Invalid head of packet ('\x08')
`
I have tried sending AT commands to the ESP01 like "AT" or "AT+GMR" and everything works fine, but I cannot upload any sketch because of that error.

I also followed esp8266 official forum advice to edit the Python script according to the following, but without success.
```
Line 43 ESP RAM BLOCK = 0x1800 //change this to 180
Line 44 ESP RAM BLOCK = 0x400 //change this to 40
```
The level conversion from 5v TTL to 3.3v is done through simple resistors (1k and 2k).

Wiring:
CHPD to Vcc (3.3v) to enable the device.
GPIO0 to ground through a push button, to put it into programming mode.
RST pull-up to Vcc through a 10k resistor and a button to ground.

projectgus, 2017-02-27:
Arduino's USB/Serial firmware is fussy about the DTR & RTS control lines, which are also used by default to signal automatic reset on many ESP8266 development boards.

To use an Arduino Uno or similar as the interface, run esptool.py as follows: 
```
python esptool.py --before no_reset -p /dev/cu.usbport write_flash 0x000000 "nodemcu.bin"
```

A note about this is is [mentioned in the README](https://github.com/espressif/esptool#reset-before-operation).

grcasanova, 2017-02-28:
Ok thanks, the sketches are now uploaded fine but unfortunately they don't run. I tried a bare minimum sketch with just serial ping and another blinking the built-in led but they don't seem to work. Regarding the wiring, the GPIO0 is now on Vcc.

projectgus, 2017-03-01:
It's hard to tell without more details, but I suggest working through the list of troubleshooting possibilities here: https://github.com/espressif/esptool#write_flash-succeeds-but-program-doesnt-run

grcasanova, 2017-03-02:
Here is a full description http://arduino.stackexchange.com/questions/35150/esp01-unsuccessful-code-execution

projectgus, 2017-03-02:
The stackexchange post doesn't address any of the items listed at the troubleshooting link above. To begin with:
* How is the ESP-01 powered?
* Have you tried passing the -fm dio argument to flash in dual I/O mode instead of QIO?

projectgus, 2017-03-02:
Also, what is the full command line you're passing to esptool.py? The command line in the first post is flashing NodeMCU, but it seems this is Arduino IDE output.

projectgus, 2017-04-24:
@grcasanova did you make any progress with this issue?

grcasanova, 2017-04-25:
Hi, yes it seems my breadboard resistance was too high and the ESP-01 wasn't booting properly. I had to skip breadboard prototyping and connect the Arduino directly to the ESP and things seemed to work.

projectgus, 2017-04-25:
I see, glad you resolved it.

I'm going to close this issue, but please reopen if you think there's an esptool issue here.

closed 2017-04-25 by projectgus