espressif/esptool #181: Install esptool.exe on Windows

hickford, 2017-02-28:
Hi. Installing esptool on Windows `pip install esptool` installs a Python script `C:\Python36\Scripts\esptool.py`. It would be more convenient if it installed an .exe file `esptool.exe`

Other applications do this such as `pip install microrepl` https://github.com/ntoll/microrepl

CharlesScoville, 2017-03-05:
Eh... exactly why would an .exe be more convenient than a .py _if you already have Python installed_? 

It's not like it's a graphics or cryptography primitive that needs to be optimized for speed, nor is there a compatibility issue since this runs on (the platform independent) Python interpreter. 

Esptool is a command-line utility. Performance is bound by the embedded devices themselves and cross compatibility ensured by Python.  It doesn't make sense to burden the developers with an extra step for no gain.

. . .

On the other hand, if you're asking for the _total package_ to be a simple/single .exe, then you should know it's already been done... and then some. Check out nodemcu-pyflasher. You can get said .exe here.

https://github.com/marcelstoer/nodemcu-pyflasher/releases

hickford, 2017-03-05:
Hi. Pardon me, I haven't explained well. The boon is the `.exe` wrapper will always execute with the right version of Python (consider if you had two copies of Python and esptool.py in your path). More at https://setuptools.readthedocs.io/en/latest/setuptools.html#automatic-script-creation

The change is as simple as one line to setup.py, setuptools creates the `.exe` wrapper at install time https://packaging.python.org/distributing/#scripts . Let me prepare a pull request.

There might be other benefits on other platforms, I don't know

projectgus, 2017-03-05:
> The boon is the .exe wrapper will always execute with the right version of Python (consider if you had two copies of Python and esptool.py in your path).

FWIW, I think installing via the existing `scripts` directive will do this as well. If you install via `setup.py install` or `pip`, then the actual `esptool.py` file is a thin wrapper that calls the correct verison of Python and sources the script. At least that's the case on POSIX, I'd expect the same on Windows but maybe not.

(Of course this isn't the case if you run esptool.py as-is in the git repository or another standalone place. Which is one of the reason why the docs encourage a pip install.)

In any case, thanks for sending the PR. I have some more specific thoughts there, will comment there instead.

projectgus, 2017-03-05:
> I'd expect the same on Windows but maybe not.

Ah, I remember now. Windows won't execute a `.py` file properly at all, will it? You have to invoke `python` each time. Hmm.

(Left more specific comments on the PR.)

hickford, 2017-03-06:
Right on my computer `esptool.py` opens a text editor. The workaround is to run `python -m esptool`

closed 2017-04-04 by projectgus

flywire, 2019-01-04:
I thoroughly support this request, other projects do it. It is much more convenient to just run a .exe in windows rather than trying to figure out which python to install and how it is configured. The file could be generated as part of a new release.

flywire, 2019-07-07:
Available at:  https://dl.espressif.com/dl/esptool-2.3.1-windows.zip - not sure how versions maintained.

OMGdaDPS, 2019-12-10:
doesn't run on Windows 10. Just opens then closes.

projectgus, 2019-12-10:
@OMGdaDPS It's a command line program, did you run it from a Command Prompt window?

OMGdaDPS, 2019-12-13:
Oh no, sorry. I seen it was a .exe file, so i thought that it would open some sort of GUI window. Ha. So its pretty much the same as it is on Linux? Same flag and parameters ? @projectgus

flywire, 2019-12-13:
@OMGdaDPS Can you confirm the source of the .exe?

OMGdaDPS, 2019-12-13:
I got it right from the link you posted up there. Is that what you meant?

flywire, 2019-12-13:
Just note purpose of that post was requesting  .exe be generated for distributions here because on other sites ...
> Versions are outdated and probably not maintained

Not disputing benefits of linux ... just with 90% of desktop browsing in Windows it's hardly relevant ;)

OMGdaDPS, 2019-12-13:
ok ;). Let me know when this .exe starts working.

flywire, 2019-12-13:
I'm not saying it doesn't work, just it's not necessarily maintained. Instructions to generate .exe: https://github.com/espressif/esptool/issues/181#issuecomment-451337626

bobatsar, 2023-09-18:
Just for anybody else looking for binaries. They are release now at https://github.com/espressif/esptool/releases