espressif/esptool #183: A fatal error occurred: Timed out waiting for packet header when using 4MB   bin file

SmartArduino, 2017-03-05:
esptool.py  --port COM77 -b 921600  write
_flash --flash_freq 40m --flash_mode dio --flash_size 4MB 0x0000  t.bin
esptool.py v2.0-beta2
Connecting....
Detecting chip type... ESP8266
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Attaching SPI flash...
Configuring flash size...
Flash params set to 0x0240
Compressed 4178048 bytes to 231980...
Writing at 0x00038000... (100 %)
A fatal error occurred: Timed out waiting for packet header

projectgus, 2017-03-05:
Thanks for reporting this. It seems like the timeout is not long enough for very large compressible files, where the ESP does a long decompression operation on the remote side. I'm able to reproduce by generating an all-zeroes file of 2MB or larger.

In the case of the file you are flashing, I'm guessing most of the data is at the beginning and then the very end, with a gap in the middle of a repeating byte or pattern? 

Will work on a fix for this, but I expect you can work around the issue for now by adding `-u` (for uncompressed flashing) to the write_flash arguments (ie `write_flash -u --flash_freq , etc`). This will be slower, but it will flash successfully.

SmartArduino, 2017-03-06:
Thanks for u advice.
Mof the data is at the beginning and then the very end, with a gap in the middle of a repeating byte  in the file bin.
The bin may be download from the attached file:

[t.zip](https://github.com/espressif/esptool/files/820033/t.zip)

SmartArduino, 2017-03-06:
I change 
the self._port.timeout = 100

That is ok for 4M bin.

projectgus, 2017-03-06:
Thanks. Glad this fixed the problem. In most cases esptool.py wants a shorter timeout than this (so you don't have to wait 25 seconds to find out something has gone wrong), so the long term fix will probably be detecting how much uncompressed data was sent in each block, and adjusting the timeout to match (more uncompressed data = longer timeout.)

commit e24aa91f (2017-06-14): Increase timeout when flashing heavily compressed images

Fixes #183

Also cleans up timeout "magic numbers" to use names.

"Overall file ratio" is a poor heuristic for the amount of uncompressed data in
a given single block, but it seems to work well enough - the original timeout
was fairly generous as well.
closed 2017-06-14 by projectgus