espressif/esptool #186: flash speed/mode bytes in file not used

SmallRoomLabs, 2017-03-06:
The flash parameter bytes in the binary are always overridden by the default value of the tool, but the readme seems to indicate that the tool would use the data in the file unless overridden by command line parameters. `When flashing a bootable image to an ESP8266 at offset 0x0, the image header bytes are updated automatically using these arguments.`

```
$ hexdump -C cpm8266_01.bin | head -1
00000000  e9 03 03 0f f8 23 10 40  00 00 10 40 20 30 00 00  |.....#.@...@ 0..|
```

The old stable tool:
```
$ esptool.py write_flash 0x0 cpm8266_01.bin 

esptool.py v1.2
<snip>
Flash params set to 0x0040
Writing 4194304 @ 0x0...
```

Same result with the newer version:
```
$ esptool.py write_flash 0x0 cpm8266_01.bin 

esptool.py v2.0-beta1
<snip>
Auto-detected Flash size: 4MB
Flash params set to 0x0040
Compressed 4194304 bytes to 325684...
```

projectgus, 2017-04-24:
Thanks for pointing this out. The README doesn't properly describe the current behaviour, which is that the command line arguments (whether defaults or not) are used to update the .bin file when it is flashed.

A better behaviour might be to do would be what the README describes, ie don't touch the header unless some arguments are supplied. Except for maybe flash size, where the default behaviour is "detect".

SmallRoomLabs, 2017-04-24:
Changing the current behavior in the 1.x might not be a good idea.  But as long as 2 is in beta I think it's fair game to make the change.  

Command line arguments (if supplied) should override the settings in the file. But having default values override the file settings is not so good.   

So please make the change now before 2 goes into GA and it's too late to make a behavioral change....

projectgus, 2017-04-25:
Only bugfixes are going into 1.x, so that's done.

> So please make the change now before 2 goes into GA and it's too late to make a behavioral change....

This was on my original list for things to change in v2, but in the end it seemed like it might mess up people's existing esptool workflows a bit. I'll think it through again.

commit ef70691b (2017-06-14): write_flash & verify_flash: Add 'keep' default for --flash_size & --flash_mode

Except for size, esptool.py will no longer change image file settings by default.

Closes #186
closed 2017-06-14 by projectgus