espressif/esptool #187: Fatal error when upload 0 byte files with esptool 1.2

SmallRoomLabs, 2017-03-06:
I get a fatal error when trying to upload an empty file using the v1.2 tool. Using the 2.0 tool it works just fine, probably because the 0-byte file gets "compressed" to 8 bytes before upload ;-)

```
$ ls -l *.bin
-rw-rw-r-- 1 matseng matseng 15280 Mac   6 11:13 image.elf-0x00000.bin
-rw-rw-r-- 1 matseng matseng     0 Mac   6 12:09 image.elf-0x10000.bin
```

```
$ esptool.py write_flash 0x0 image.elf-0x00000.bin 0x10000 image.elf-0x10000.bin 
esptool.py v1.2
Connecting...
Auto-detected Flash size: 32m
Running Cesanta flasher stub...
Flash params set to 0x0040
Writing 16384 @ 0x0... 16384 (100 %)
Wrote 16384 bytes at 0x0 in 1.4 seconds (91.8 kbit/s)...
Writing 0 @ 0x10000... 
A fatal error occurred: Expected digest, got: 00000000
```

And yes, I actually sometimes have a 0 byte 0x10000 file. I'm using the NOSDK by cnlohr and I don't get any data in that segment unless I actually force something to end up there like with:
```
volatile uint32_t KEEPMEHERE __attribute__((section(".irom.text"))) = 0xDEADBEEF;
```

projectgus, 2017-04-24:
Sorry for the delay in getting back to you.

Thanks for reporting this. I haven't started a maintenance branch for v1.x yet, but when I do I'll commit a fix for this. I'm not sure if there will be another v1.x release, it will depend if any other bugs show up.

Angus

projectgus, 2017-06-14:
This is fixed in the new stable v2.0 release. If there is a bugfix v1.x release then I'll also revisit it there, but for now this is not planned (will only be if a major bug appears.)

closed 2017-06-14 by projectgus