espressif/esptool #188: IROM_ALIGN AssertionError when using fexceptions

benariamir, 2017-03-07:
Hi guys,
I'm new to this so please forgive me if that's a non-issue.

We're making a new IoT device and we do want exceptions handling as we don't want the risk of a bug completely crashing our device, or at least if an exception is caught report it back to the server.

I'm running into an issue when rasing the fexceptions flag.
(I'm using platform.io to build which uses esptool.py v2.0-beta1)

I made a small "hello world" app and all was good.
firmware.elf size (bytes) w/o exceptions -> 2323360 bytes
firmware.elf size (bytes) w/ exceptions -> 2323452 bytes

Then I included WiFi.h and everything went to hell.
Building it w/o exceptions -> no problem (2375280 bytes)
Building it with exception, build failed and I got the following:

xtensa-esp32-elf-g++ -o .pioenvs/esp32dev/firmware.elf -nostdlib -Wl,-static -u call_user_start_cpu0 -Wl,--undefined=uxTopUsedPriority -Wl,--gc-sections -Wl,-EL -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -Wl,-T"esp32_out.ld" .pioenvs/esp32dev/src/main.o -L/Users/amirbenari/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib -L/Users/amirbenari/.platformio/packages/framework-arduinoespressif32/tools/sdk/ld -L.pioenvs/esp32dev -Wl,--start-group .pioenvs/esp32dev/libFrameworkArduinoVariant.a .pioenvs/esp32dev/libFrameworkArduino.a -lgcc -lstdc++ -lapp_update -lbootloader_support -lbt -lbtdm_app -lc -lc_nano -lcoap -lcoexist -lcore -lcxx -ldriver -lesp32 -lethernet -lexpat -lfatfs -lfreertos -lhal -ljson -llog -llwip -lm -lmbedtls -lmdns -lmicro-ecc -lnet80211 -lnewlib -lnghttp -lnvs_flash -lopenssl -lphy -lpp -lrtc -lsdmmc -lsmartconfig -lspi_flash -ltcpip_adapter -lulp -lvfs -lwpa -lwpa2 -lwpa_supplicant -lwps -lxtensa-debug-module .pioenvs/esp32dev/lib/libWiFi.a -Wl,--end-group
xtensa-esp32-elf-size -B -d .pioenvs/esp32dev/firmware.elf
"/Users/amirbenari/.atom/packages/platformio-ide/penv/bin/python" "/Users/amirbenari/.platformio/packages/tool-esptoolpy/esptool.py" --chip esp32 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o .pioenvs/esp32dev/firmware.bin .pioenvs/esp32dev/firmware.elf
text	   data	    bss	    dec	    hex	filename
355601	  41992	  20976	 418569	  66309	.pioenvs/esp32dev/firmware.elf
esptool.py v2.0-beta1
Traceback (most recent call last):
File "/Users/amirbenari/.platformio/packages/tool-esptoolpy/esptool.py", line 2257, in <module>
main()
File "/Users/amirbenari/.platformio/packages/tool-esptoolpy/esptool.py", line 2072, in main
operation_func(args)
File "/Users/amirbenari/.platformio/packages/tool-esptoolpy/esptool.py", line 1708, in elf2image
image.save(args.output)
File "/Users/amirbenari/.platformio/packages/tool-esptoolpy/esptool.py", line 1282, in save
assert (f.tell() + 8) % IROM_ALIGN == segment.addr % IROM_ALIGN
AssertionError
*** [.pioenvs/esp32dev/firmware.bin] Error 1

I think that has something to do with the firmware.elf size as the save function have something to do with "IROM/DROM segment flash mappings need to align on 64kB boundaries."

Any ideas?

projectgus, 2017-03-07:
Are you able to attach the ELF file to this issue? If it's sensitive then perhaps you can email it to me - angus at espressif.

Did you make any modifications to the build process apart from compiling with -fexceptions?

benariamir, 2017-03-12:
For some reason I'm failing to recreate this with a simple "hello world" program.
What I'm currently managing is pin pointing it to a certain line in code of the wifi scan example (https://github.com/platformio/platformio-examples/blob/develop/espressif/esp32-arduino-wifiscan/src/WiFiScan.ino).

Line 36 is causing this. Once commented problem goes away.
You can find the source files and elf [here](https://drive.google.com/drive/folders/0Bx2SqJsgu8LgRnpUZnBPOXFGV2s?usp=sharing).

If this is somehow related to the WiFi library please feel free to close this and point me to the project it comes from.

projectgus, 2017-03-30:
Hi,

Sorry for the slow reply, I think this is the same cause as this issue:
https://github.com/espressif/esp-idf/issues/459

As modifications will be required on the IDF side, I'm going to close this issue in favour of the linked one. Please feel free to add more details/comments there.

closed 2017-03-30 by projectgus