espressif/esptool #190: i can not receive RTS  info

ouki-wang, 2017-03-29:
I used a standard USB CDC device（stm32）
when i close the serial，i receive the RTS info
21 22 02 00 00 00 00 00

but when I use esptool.exe，I can not receive 21 22 02 00 00 00 00 00 to control RTS pin

ouki-wang, 2017-03-29:
can you enable RTS control？

ouki-wang, 2017-03-29:
will esptool.exe source code be opened?

projectgus, 2017-03-30:
What esptool.exe are you referring to?

ouki-wang, 2017-03-30:
@projectgus run get.exe. I can get a esptool.exe

ouki-wang, 2017-03-30:
I have found a project source https://github.com/igrr/esptool-ck
Is is an official release version?

projectgus, 2017-04-24:
I'm sorry but I don't understand the problem you are having at all, I need more details.

* Are you using Arduino for ESP32 to install esptool.exe?
* What device is connected to the STM32 CDC firmware?
* Where does the CDC firmware source code from?
* What host OS are you on?
* Can you post the output from a failed flash with esptool?

projectgus, 2017-06-13:
Closing due  to a lack of information to follow up, but feel free to reopen if you have more information.

closed 2017-06-13 by projectgus