espressif/esptool #191: esptool.py can't support dio flash mode

YanMinge, 2017-03-29:
I have print the flash_mode arg from esptool.py, I'm sure the incoming parameters are dio, but when finish the image flash, it return qio mode issue.

projectgus, 2017-03-29:
Can you supply the full command line and output from esptool.py, please?

projectgus, 2017-03-29:
Also the serial output from when it returns "qio mode issue" at boot would be useful if you have it. Thank you.

YanMinge, 2017-03-30:
when reset，the issue is:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3fff0008,len:8
load:0xffffffff,len:-1
```
but I sure the config is dio mode.
[image]

YanMinge, 2017-03-30:
The full command for esptools is: 
`$(Q)$(ESPTOOL) --chip esp32 --port $(PORT) --baud $(BAUD) write_flash -z --flash_mode $(FLASH_MODE) --flash_freq $(FLASH_FREQ) --flash_size $(FLASH_SIZE) 0 $^`
and the flash_mode is setting to dio mode. but if the esp32 module support qio mode, It will successfully burn and run. seem the esptool only support qio mode.

projectgus, 2017-03-30:
It would have been helpful if in the beginning you had mentioned you were using Micropython on ESP32.

This seems like it comes from the Micropython ESP32 support behaviour. The flash mode is set on the bootloader binary image, and is set when the bootloader binary image is generated. esptool.py can also override this when flashing the bootloader, but only if it is flashed at offset 0x1000 (bootloader offset on ESP32.) Micropython takes all the .bin images and creates one large image which is flashed it at offset 0x0, so esptool doesn't detect that it can/should override the flash mode on the fly.

If you run `make clean` to remove bootloader.bin and then re-run `make deploy` then you should get the correct flash mode.

If you think this behaviour should be fixed, please open a bug on the Micropython ESP32 repository: https://github.com/micropython/micropython-esp32/issues and link to this issue. We can possibly detect this situation in esptool.py (flashing at 0x0 with a bootloader image embedded in the binary) but I'd rather not if this can be fixed some other way.

YanMinge, 2017-03-30:
Your answer is professional and effective, Thanks for you help.

closed 2017-03-30 by YanMinge