espressif/esptool #192: ESP32/VMWare - Device not ready.

CodeFreak604, 2017-04-01:
I have a Win10 hosted VMWare Workstation (11.1.2) that works perfectly when programming Arduino ESP-8266, that fails when trying to program an ESP32.  To get setup I followed the Arduino ESP32 instructions here: https://github.com/espressif/arduino-esp32/blob/master/doc/windows.md  I'm using Python 3.6.0 and Arduino 1.8.15.  This WMWare virtual machine and USB port works fine when programming ESP-8266.  When I try to upload to an ESP-32 it says "This device is not ready."  Any suggestions?

```
Sketch uses 115459 bytes (11%) of program storage space. Maximum is 1044464 bytes.
Global variables use 9488 bytes (3%) of dynamic memory, leaving 285424 bytes for local variables. Maximum is 294912 bytes.
esptool.py v2.0-dev
Traceback (most recent call last):
  File "esptool.py", line 2177, in <module>
  File "esptool.py", line 1954, in main
  File "esptool.py", line 153, in __init__
  File "site-packages\serial\__init__.py", line 88, in serial_for_url
  File "site-packages\serial\serialwin32.py", line 62, in open
serial.serialutil.SerialException: could not open port 'COM5': WindowsError(21, 'The device is not ready.')
Failed to execute script esptool
the selected serial port Failed to execute script esptool
 does not exist or your board is not connected
```

moxy-bc, 2017-04-02:
Not sure if this is related to this, or not...
https://github.com/espressif/esptool/issues/136#issuecomment-274413187

projectgus, 2017-04-04:
Hi @CodeFreak604,

I don't think this is related to the issue @moxy-bc linked, that failure happens at a later stage than this.

A couple of questions/suggestions:
* Can you verify COM5 exists and is shown in Windows Device Manager? Are there are warnings shown for COM5 in Device Manager?
* Is the serial port shared from the host to the guest at the serial port level, or as a USB device? If shared as a serial port, we have had reports of problems with VM serial port emulation before - try sharing the entire USB device from the host to the guest.

projectgus, 2017-04-24:
@CodeFreak604 , did you make any more progress with this issue?

projectgus, 2017-06-13:
Closing due to a lack of information to follow up, but feel free to reopen if you have more information.

closed 2017-06-13 by projectgus