espressif/esptool #195: esptool cannot work for cp210x on Ubuntu 16.04

young-mu, 2017-04-19:
Does anyone encounter this issue?

The USB-to-UART chip is cp2104, on Ubuntu 16.04 (kernel is 4.4.x), esptool.py cannot work correctly. But on MacOS and Windows, it works well. 

So I guess that perhaps it's cp210x module's issue on Linux.

projectgus, 2017-04-21:
Can you supply some more details, please?

* What output do you get when it fails?
* What driver are you using, the built-in Linux driver or have you installed a driver from Cypress?
* What chip (ESP8266 & ESP32), and what development board (if any)?

young-mu, 2017-04-24:
The chip is ESP32, the dev board is developed by 3rd party [link](https://item.taobao.com/item.htm?spm=a230r.1.14.75.ircNef&id=538209801368&ns=1&abbucket=18#detail), in which the USB-to-UART chip is CP2104.

The driver is built-in Linux driver cp210x. 

```shell
$ lsmod | grep cp210x
cp210x                 24576  0
usbserial              53248  1 cp210x
```

Here are the outputs when I try to get mac address by using esptool.py.

```shell
$ sudo esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 read_mac
Connecting...........................................

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
```

Here is my test environment:

- Ubuntu 16.04.1 LTS
- 4.4.0-72-generic

projectgus, 2017-04-24:
Do you get the same failure for both the Espressif ESP32-DevKitC and for the Widora Air board? Or only for the Widora Air board?

Is Ubuntu running in a VM or on real hardware?

young-mu, 2017-04-24:
Yes, the failure is the same for two boards. And I have tested it for both VM and real hardware.

projectgus, 2017-05-03:
I'm not able to reproduce this with a clean Ubuntu 16.04 VM and an Espressif Core Board V2 (CP2102).

[image]

After testing this I noticed that you have quite an old version of esptool.py (it doesn't print the version number.) If you try with the latest version then it will hopefully work.

young-mu, 2017-05-04:
OK, I have tried again on my PC using **the latest version esptool**. It works well.
Thank you very much. I'll close this ticket.

closed 2017-05-04 by young-mu