espressif/esptool #197: esptool is giving error with elf2image option with esp_iot_sdk_v1.2.0.

SivaNallamalli, 2017-04-24:
I am compiling blinky example from https://github.com/esp8266/source-code-examples/. 
using  sdk esp_iot_sdk_v1.2.0. Only change added was adding a empty void user_rf_pre_init(){}
in main source code to resolve  compilation issue.

I am getting fallowing error with make flash using blinky example

esptool.py v2.0-beta2
WARNING: Unexpected ELF section header length a9d9 is not mod-28
Traceback (most recent call last):
  File "/bin/esptool.py", line 2301, in <module>
    _main()
  File "/bin/esptool.py", line 2294, in _main
    main()
  File "/bin/esptool.py", line 2096, in main
    operation_func(args)
  File "/bin/esptool.py", line 1713, in elf2image
    e = ELFFile(args.input)
  File "/bin/esptool.py", line 1326, in __init__
    self._read_elf_file(f)
  File "/bin/esptool.py", line 1349, in _read_elf_file
    self._read_sections(f, shoff, shstrndx)
  File "/bin/esptool.py", line 1366, in _read_sections
    all_sections = [read_section_header(offs) for offs in section_header_offsets]
  File "/bin/esptool.py", line 1364, in read_section_header
    name_offs,sec_type,_flags,lma,sec_offs,size = struct.unpack_from("<LLLLLL", section_header[offs:])
struct.error: unpack_from requires a buffer of at least 24 bytes

projectgus, 2017-04-25:
Hi @SivaNallamalli ,

Thanks for reporting this. Could you attach the ELF file to a comment on this issue, please?

Angus

SivaNallamalli, 2017-04-25:
I created a zip file, that contains all the source, Makefiles & binaries that are generated after compilation. I am unable to attach my app.out binary that was generted.

All the binaries that are generated are available  in build folder.
[basic_example.zip](https://github.com/espressif/esptool/files/954095/basic_example.zip)

Shai1436, 2017-05-02:
Hi, @projectgus  I am facing the same "struct.error" while making binary files for my ELF file.

> esptool.py elf2image build/kaa_demo

[Link for ELF file](https://drive.google.com/open?id=0B9xEKi__fYguMzZOQS1Ba2lpUTA)

projectgus, 2017-05-03:
Thanks @SivaNallamalli , @Shai1436 . These ELF files were very helpful (their layout is different to the ELF files I'd previously seen.)

This is issue should be fixed in the esptool.py master branch now. I was able to run elf2image on both your supplied ELF files, and the output looks correct. Please reopen if that's not the case.

closed 2017-05-03 by projectgus

Shai1436, 2017-05-03:
Thanks @projectgus  for the fix, but the problem exists in my case. It's still giving the same error.

     esptool.py v2.0-beta2

    WARNING: Unexpected ELF section header length f4e0 is not mod-28

    Traceback (most recent call last):

    File "/usr/local/bin/esptool.py", line 9, in <module>

    load_entry_point('esptool==2.0-beta2',` 'console_scripts', 'esptool.py')()

    File "build/bdist.linux-x86_64/egg/esptool.py", line 2294, in _main
    
    File "build/bdist.linux-x86_64/egg/esptool.py", line 2096, in main
    
    File "build/bdist.linux-x86_64/egg/esptool.py", line 1713, in elf2image
    
    File "build/bdist.linux-x86_64/egg/esptool.py", line 1326, in __init__
    if __name__ == '__main__':
    File "build/bdist.linux-x86_64/egg/esptool.py", line 1349, in _read_elf_file
    
    File "build/bdist.linux-x86_64/egg/esptool.py", line 1366, in _read_sections
    
    File "build/bdist.linux-x86_64/egg/esptool.py", line 1364, in read_section_header
    
    struct.error: unpack_from requires a buffer of at least 24 bytes
I have downloaded the files from the master branch. I have also tried deleting the previous corresponding python packages and rebooting the device.

projectgus, 2017-06-15:
Hi @Shai1436 sorry, because the issue was still closed I missed your reply.

The esptool.py version with the fix was 2.0-beta3, so I think you still had an older version without the fix.

I tested with esptool.py v2.0 final and your ELF file is processed correctly. If you're still getting errors with v2.0 final, please reopen this issue.

commit 2cb00b77 (2017-05-03): elf2image: Fix error when section header is not at end of ELF file

Closes #197

Includes test case (first ELF file posted in linked issue.)