espressif/esptool #198: A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

croisez, 2017-04-26:
Hello ESP32 lovers,
I am facing a problem with a WEMS ESP32 dev board which I bough on Aliexpress very recently:
https://fr.aliexpress.com/item/ESP32-Development-Board-WiFi-Bluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP-32-ESP-32S-Board/32801337284.html

I am able to flash it using Eclipse/esp-idf/xtensa-esp32-elf triplet. The /dev/tty.SLAB_USBtoUART device configured by "make menuconfig" makes the flashing process a success. Further on, my user application (one of the examples of esp-idf) starts and does what expected.

But, with Arduino IDE, this is not the same story. I have installed the toolchain as written at ESP32-Arduino github page, then selected "ESP32 dev module" as the board in Arduino IDE, and also selected /dev/tty.SLAB_USBtoUART as the flashing device.
Like I did for the Eclipse case, I reset dev board with RST and BOOT buttons. I verified to get the "DOWNLOAD" invite on the terminal, then closed the terminal pushed on flash button of Arduino IDE. All I get in Arduino logs is the following:

<PRE>
Sketch uses 118,890 bytes (11%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 9,492 bytes (3%) of dynamic memory, leaving 285,420 bytes for local variables. Maximum is 294,912 bytes.
/Users/lmc/Dropbox/arduino/hardware/espressif/esp32/tools/esptool --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_freq 80m --flash_mode dio --flash_size 4MB 0x1000 /Users/lmc/Dropbox/arduino/hardware/espressif/esp32/tools/sdk/bin/bootloader.bin 0x8000 /var/folders/zx/4rs36y2j0315wny99n1j7fj80000gn/T/arduino_build_538507/Blink.ino.partitions.bin 0xe000 /Users/lmc/Dropbox/arduino/hardware/espressif/esp32/tools/partitions/boot_app0.bin 0x10000 /var/folders/zx/4rs36y2j0315wny99n1j7fj80000gn/T/arduino_build_538507/Blink.ino.bin 

esptool.py v2.0-beta
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____
 
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
</PRE>

The programming does not detect the dev board in download mode, and the flashing does not start.
As a workaround, I hooked up an external FTDI adapter cable directly to the ESP32, and then Arduino IDE is able to start the flashing. But after, the Arduino sketch does not seems to run. My simple Blink sketch is not running (it toggles a GPIO and prints some text on Serial console).

* Issue 1: I saw a post about the problem of the FreeRTOS scheduler not started on Core1, or not enabled Core1, which could resolve the Arduino sketch not running (but I don't know how to configure it within ESP32-Arduino libraries/board from inside Arduino IDE. Could it be a efuse which should be reset to enable Core1 to start?

* Issue 2: Concerning my flashing issue, I don't understand why my dev module can be successfully programmed by using the Makefile from within Eclipse/esp-idf, while it ends-up with a timeout from within ArduinoIDE/ESP32-Arduino toolchain.

Anyone could help me about my two issues?
Thank you so much.

projectgus, 2017-04-28:
Please don't post the same issue in multiple places, we're a small team here and it creates unnecessary confusion and overlap. In future, please open a new issue for each distinct question/problem and choose the repository where you think the question is most appropriate. It's always possible to move to a new repository if it turns out you've asked in the wrong place. :)

> Concerning my flashing issue, I don't understand why my dev module can be successfully programmed by using the Makefile from within Eclipse/esp-idf, while it ends-up with a timeout from within ArduinoIDE/ESP32-Arduino toolchain.

This is indeed odd. Both Arduino and ESP-IDF use esptool.py, but they may use slightly different versions. I have a question & a couple of suggestions:

* Do you need to manually press any buttons when flashing from Eclipse? Looking at the dev board you have, I would guess it should not require any manual button pressing (the Espressif Core Board doesn't, and it looks similar to a Core Board). But I don't know anything about this manufacturer, it's always possible the design is different.

* If you hold down the BOOT button (no other button pressing needed) and then start the flash from Arduino, does flashing start correctly? If it starts OK then you can release the BOOT button once it's started.
* If that doesn't work, if you hold down BOOT Button, start the flash in Arduino, and then wait for a fraction of a second before pressing and releasing the RST button a single time (while continuing to hold BOOT), does this work?

croisez, 2017-05-01:
I apologize for any inconvenient. I will pay attention in the future.
Concerning your questions:
- yes I must push on GPIO0+RST buttons for ESP32 to enter Download mode, although there is two mosfets on the boards, which if I understand right, are driven by RTS/DTS signals from the RS232 SLA chip. But ok, this is another problem. Perhaps the driver installed on my OSX is filtering those signals, or perhaps these signals are not handled by the board I choose (I am choosing the first entry "ESP32 dev module" because I don't found my board reference in the menu).
- from inside Eclipse, yes, the flashing starts correctly when I hold down the BOOT button, but only when I do a RST as well. I never tried with BOOT button only. You mean that the RST should be driven by software? BTW I remarked that my board never reboots after a successfully flash, although it is written on screen that esptool is rebooting the board.
- As already said, the combination of BOOT and RST pushes allows me to flash the board with success, but not from within Arduino IDE, but from Eclipse, yes

croisez, 2017-05-02:
OK. Arduino is actually able to program my board when I manually push on BOOT button before flashing process begins. No need to push on RST button. There is still a problem with the Arduino sketch which does not start upon reset of the board.  But this is not related to esptool. So I can close this issue. Thanks for support

closed 2017-05-02 by croisez

architmuchhal12, 2017-05-23:
I am facing the same issue with the WEMOS LOLIM 32 v1.0.0. It doesn't has a BOOT button though, only the RESET button. What should I do?
(https://forum.wemos.cc/topic/334/lolin32-v1-0-0-is-release)

projectgus, 2017-05-24:
Hi @architmuchhal12,

Looking at the schematic for the LOLIN32, it is designed to reset automatically into flashing mode without any manual button presses. What version of esptool.py are you using (it should be printed out when you run it), and what OS are you running on?

Another thing you could try it using a jumper wire to bridge IO0 to GND before flashing and removing it to run the firmware normally, although this is obviously not an ideal long-term solution.

You may need to raise this issue with WeMos. It could be hardware-related.

Angus

architmuchhal12, 2017-05-24:
Hey Angus,

By the looks of it, I think I'm using _esptool.py v2.0._ I am not sure if this is the version to be used. 
Also, I am running the Windows 7 OS.
You can check the screenshot attached.
[image]

**Error:**
```
`Sketch uses 108830 bytes (10%) of program storage space. Maximum is 1044464 bytes.
Global variables use 9616 bytes (3%) of dynamic memory, leaving 285296 bytes for local variables. Maximum is 294912 bytes.
esptool.py v2.0-beta3
Connecting....
Uploading stub...

A fatal error occurred: Response doesn't match request
A fatal error occurred: Response doesn't match request
`

```

projectgus, 2017-05-25:
This seems to me like it's probably a hardware problem, esptool.py is able to talk to the ESP32 but as it starts the flashing process something goes wrong and communications are lost/corrupted.

Either the particular board you have, or the USB cable, may be at fault.

PRosenb, 2017-06-10:
What worked for me all times (on Mac) is the following:
- unplug ESP32
- start to hold the BOOT button
- plug ESP in
- execute: make flash monitor
- release the BOOT button when it tries to connect

I hope it helps.

tehniq3, 2017-07-16:
I have same problems... i try in Arduino IDE and PlatformIO-Atom, same errro message: A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

tjm-, 2017-08-01:
Hi
     I'm having a similar problem.   I bought a ESP-32 ESP-32S Development Board WiFi Bluetooth Ultra-Low Power Consumption Dual Cores ESP32 Board, installed the IDF on Ubuntu in a Virtual Box, built the hello_world example and it flashed fine with the monitor showing the message and restarts.  I then tried another app but got the message
 esptool.py v2.1-beta1
Connecting.....
Chip is ESP32D0WDQ6 (revision 0)
Uploading stub...

A fatal error occurred: Timed out waiting for packet header

I tried the hello_world again and got the same message.   I've since tried re-installing the IDF on my Windows 10 host and re-building hello_world but got the same message.  I've tried the various button holdings as suggested here and elsewhere but to no avail and I also tried a different version of esptool (v2.0.1) with the same results.

Any help would be appreciated

Thanks
Tom

architmuchhal12, 2017-08-01:
I pretty much believe that it is a hardware issue. I had the ESP32 from WEMOS and it did have issues. Later on, I ordered one from Sparkfun and since then, I haven't;t faced this issue even once.
I would recommend to change your ESP32 and then try.

tjm-, 2017-08-01:
Success, of a sort!  I connected another usb->uart to tx0 and rx0 and tried flashing; after manually resetting and pressing boot (after the connect message) everything worked as expected - stub uploaded, run, etc all the way to hash verified hard reset (which didn't work).  Manual reset and app running.  

I'm not sure what isn't actually working with the on-board usb->uart but this seems to work for now

GwynMunich, 2018-10-24:
I have 10 boards from one supplier and 10 boards from another. They are both labelled "NODEMCU ESP-32S V1.1 101010 the only discernible difference is that the good batch doesn't have text legible on the on-board USB-Serial. The batch with an issue has SiLabs CP2102 written on it.

One batch has an issue and the other one doesn't. My conclusion is that there is some low level embedded firmware in the ESP32 that needs to be updated. What Do You Think?, This is what I did:

ISSUE: some boards need the BOOT Button pressing for ~1 second while ....____......____ is being displayed (during download) and some boards don't

I get the above issue while downloading any Arduino Sketch.

If I run this from the command prompt on windows 10:

C:\Python27\python C:\Projects\ESP32\Python\esptool.py --port com9  erase_flash

I still need to hold the BOOT and it erases the flash.
BUT the next time I run the command I don't need to hold the BOOT, neither do I need to hold the BOOT to download a sketch.
HOWEVER the next time I try to download a Sketch I have the same issue. Even it the Sketch I download is a basic simple- do nothing Sketch.

projectgus, 2018-10-26:
Hi @GwynMunich ,

> My conclusion is that there is some low level embedded firmware in the ESP32 that needs to be updated. What Do You Think?

There's nothing inside the ESP32 that can be updated in this way, for compatibility with esptool.py (esptool.py only talks to the hardware and the ROM code, and these are both immutable). There are some small differences between revision 0 and revision 1 silicon versions (recent versions of esptool will print which silicon version is present when they connect), but I don't think that's the root cause here.

If I had to guess, i would say these boards have a faulty or incorrect version of the [automatic reset circuit](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#automatic-bootloader) on them. One common error is not placing a capacitor on the EN pin, try placing a capacitor (in the 100nF-2uF range) between the EN and GND pins and see if automatic reset becomes more reliable.

GwynMunich, 2018-10-26:
This is the message from the Arduino firmware flasher, just after connection is made:

Chip is ESP32D0WDQ6 (revision (unknown 0xa)) ------ FAULTY UNIT

Chip is ESP32D0WDQ6 (revision 1) ------ WORKING UNIT

It would appear that I have been shipped some older ESP-32 modules with the ESP-WROOM-32 modules that contain older silicone.

The ODD thing is the fact that they work after flash is erased.

I'm noticing that if I touch some of the GPIO pins that are connected to the internal Flash it will occasionally boot without the BOOT button being pushed. Also, if I probe these pins they appear to be floating during the boot process.

I'm wondering if the issue is do do with floating pins needing to float to ground and holding the BOOT button simply extends the period the pins are given to reach ground. I'll look at this and let you know.

in another twist I also have a small batch Mini ESP32 from WeMos; no Boot button and always downloads. and displays:
Chip is ESP32D0WDQ6 (revision (unknown 0xa)) ----- WORKING UNIT (WeMos MINI)

GwynMunich, 2018-10-26:
I'm new to github and this sort of environment in general I seem to be caught up in two separate threads on the same subject.

projectgus, 2018-10-29:
@GwynMunich These are actually all revision 1 chips. The `revision 0xa` is because the version of esptool shipped in Arduino is older than the chip, if you install a newer version from this repo (or pip) then it'll print the correct revision. The message is harmless in Arduino, though.

> I'm wondering if the issue is do do with floating pins needing to float to ground and holding the BOOT button simply extends the period the pins are given to reach ground. I'll look at this and let you know.

Quite likely something like this, with the board design or an assembly problem.

GwynMunich, 2018-10-29:
@projectgus,
Thanks for your comments. I'm duplicating what I have written on two threads, its my fault. I'll learn.
I have made some progress in that area.

I'm happy to hear the comment about the 0xa in the chip ID, the whole situation would make less sense if one chip was older and the other newer. 

@cimba007 sent me this picture from the board he was working on, the cap is a 0.1uF
https://user-images.githubusercontent.com/3453121/47588736-730eaa80-d967-11e8-9ef3-e005cf8062bd.jpg

I made a test on my faulty units and had a similar result with a 0.1uF cap as shown in this Pic:
[image]

GwynMunich, 2018-10-29:
As an interesting side note:
The reason deleting the flash using a python command works is that (as I noted earlier):
It puts the unit into a looped reset sequence. Which can be seen with Realterm at 115200.
Effectively modifying the timing of the whole reset / boot mode sequence thus allowing the unit into boot mode without the need for the BOOT key

hicham-vv, 2022-04-29:
i am facing the same issue here 
I'm noticing that if I touch boot button it will boot then flashing memory will start.
sometimes the flashing done and the sketch work perfectly, but in the most of case uploading code stops at the midle of the operation, as you see below,
I'm wondering if the issue is do with floating pins needing to float to ground and holding the BOOT button simply extends the period the pins are given to reach ground. or the the esp32 communicate badly with the flash memory.

Connecting.....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 98:f4:ab:6c:61:80
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00048fff...
Compressed 17216 bytes to 11867...
Writing at 0x00001000... (100 %)
Wrote 17216 bytes (11867 compressed) at 0x00001000 in 0.6 seconds (effective 244.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 402.8 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 556.5 kbit/s)...
Hash of data verified.
Compressed 232192 bytes to 126659...
Writing at 0x00010000... (12 %)
Writing at 0x0001e4e8... (25 %)
Writing at 0x000242d9... (37 %)

A fatal error occurred: Invalid head of packet (0x46)
*** [upload] Error 2

life-Nd, 2023-02-16:
@GwynMunich solution worked with just a jumper wire!!