espressif/esptool #199: Add option to display flash size - or what the tool thinks it is

magore, 2017-05-03:
I created a patch with a added function called get_flash_size
It displays what the esptool.py believes is the flash size for the device - matching what it would use internally for the default of "detect" when writing, etc.
I understand that the flash_id could also be used, for example in Makefile , but would require duplicate code in every Makefile to map to size.
Do you think this is useful  (when used knowing its limits of course.)?

[esptool.patch.txt](https://github.com/espressif/esptool/files/972249/esptool.patch.txt)

projectgus, 2017-05-03:
Thanks @magore . I agree it would be useful to have a command that prints the detected flash size. How about outputting this information as part of the existing flash_id command?

> I understand that the flash_id could also be used, for example in Makefile , but would require duplicate code in every Makefile to map to size.

Are you specifically interested in scraping the output of esptool.py into a Makefile, or is this just an example for discussion purposes?

Would you consider sending your change as a Github Pull Request? https://help.github.com/articles/creating-a-pull-request/

projectgus, 2017-05-03:
(The link I supplied about creating Github Pull Requests may not be the best resource. There's some more here https://help.github.com/articles/about-pull-requests/ - or lots of other blog posts/etc online.)

commit e2b2ce65 (2017-06-14): flash_id: Display detected flash size after flash_id output

Alternative implementation of patch in #199 (Closes #199).
closed 2017-06-14 by projectgus