espressif/esptool #200: Esptool install error via pip/ansible

pacm93, 2017-05-18:
Hi! I tried to install the esptool via ansible using:
```
- name: install esptool
  pip: name=git+https://github.com/espressif/esptool.git
```
or even appending #egg=esptool to the name but it doesn't properly work. It gets installed in /tmp/src/esptool, which gets erased after reboot.

Could you fix this or make a 2.0 beta release available (but non-default) on PyPI, so it is easier to automate the esptool installation?

projectgus, 2017-05-22:
Hi @pacm93,

Thanks for pointing this out. Looking at the ansible code, it looks like it [parses 'pip list' output](https://github.com/ansible/ansible/blob/devel/lib/ansible/modules/packaging/language/pip.py#L445) and possibly this is where things go wrong, as the esptool package has a version but it doesn't match anything in PyPI. I'm a little confused given how many examples in the docs show non-PyPI URLs, though.

Out of curiosity, does specifying the name as "https://github.com/espressif/esptool/archive/515adee.zip" change anything (this is the current master branch commit SHA).

> Could you fix this or make a 2.0 beta release available (but non-default) on PyPI, so it is easier to automate the esptool installation?

As soon as I have some time I plan to finalise the last small issues holding back a 2.0 final release, at which point it will appear on PyPI by default. Hopefully very soon!

projectgus, 2017-06-14:
Thanks for your patience while this release was finalised. esptool.py v2.0 is now available on pypi.

closed 2017-06-14 by projectgus