espressif/esptool #201: Timed out waiting for packet content

thundR, 2017-05-20:
Board: NodeMCU amica
Power: 1A wall wart to 3.3v breadboard breakout
Log: 

> Connecting........_
Detecting chip type... ESP8266
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 226272 bytes to 164300...
A fatal error occurred: Timed out waiting for packet content

The board used to work with the Arduino IDE esp integration, but once that stopped working, I tried manually executing the tool:
`esptool.py --port /dev/ttyUSB0 write_flash 0x00000 ledtestesp_ESP8266_NODEMCU_esp8266.bin `

Things tried: 
Timing the reset to just before the flash to catch it booting
Powering with 5v into Vin, powering with just usb, etc.
Every baud rate possible
Other bins
Other offsets

Not sure if this is a problem with the tool or the board.

projectgus, 2017-05-22:
This sounds like a hardware problem.

> FlashID=0xffffff, SizeID=0xff

This output suggests that the ESP8266 tried to talk to the NodeMCU flash chip on the board, but the flash chip didn't reply. Given you've tried different ways of powering it, I'm guessing you have a faulty flash chip (or bad solder joint on either the ESP8266 end or the flash chip end.)

projectgus, 2017-05-22:
I'm going to close this, but feel free to reopen if you think there's something else going on here.

closed 2017-05-22 by projectgus