espressif/esptool #202: Error flashing Wemos D1 mini pro with -fs 16MB (macosx)

g-sam, 2017-05-21:
After installing with `v2.0-beta3`:

```
esptool.py --port /dev/tty.SLAB_USBtoUART --baud 115200 write_flash -fm dio -fs 16MB -ff 40m 0x00000 esp8266-20170108-v1.8.7.bin
```
I get a fast reset loop. The logs are: 

```ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 32028, room 16
tail 12
chksum 0x40
ho 0 tail 12 room 4
load 0x3ffe8000, len 1092, room 12
tail 8
chksum 0x17
load 0x3ffe8450, len 3000, room 0
tail 8
chksum 0x56
csum 0x56
system param error, use last saved param!
#0 ets_task(40209c10, 31, 3ffe9020, 4)
rf_cal[0] !=0x05,is 0x00

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

//etc. over and over
```

Changing to `-fs 4MB` works fine.
```
esptool.py --port /dev/tty.SLAB_USBtoUART --baud 115200 write_flash -fm dio -fs 4MB -ff 40m 0x00000 esp8266-20170108-v1.8.7.bin
```

projectgus, 2017-05-21:
You haven't specified, but it looks like you're flashing a MicroPython. Is that right?

Support for 8MB and 16MB flash on ESP8266 depends on support inside the firmware which is being flashed. The official ESP8266 SDK only added support for these flash sizes in the recent V2.1.0 release.

This is noted in the README ([under the table here](https://github.com/espressif/esptool#esp8266-and-flash-size)) but it's pretty buried, maybe esptool should print a warning when flashing 8MB & 16MB flash sizes to ESP8266.

g-sam, 2017-05-21:
That's right.

Thanks for the info...Got my eyes on this: https://github.com/micropython/micropython/issues/2335

projectgus, 2017-06-13:
Closing this as it's not an esptool.py issue at this point. If you haven't already, clicking "Subscribe" on the linked micropython issue will let you follow progress over there.

closed 2017-06-13 by projectgus