espressif/esptool #203: A fatal error occurred: Timed out waiting for packet header after 20% firmware uploading

fwndz, 2017-05-23:
hi,

I am really new with esp8266. I tried update the AT V2.0 firmware how after a certain time the updating process stop. I was getting this kind of error after about 20% of uploading.
```
esptool.py -p /dev/ttyACM0 -b 9600 write_flash -fm qio 0x01000 user1.1024.new.2.bin
esptool.py v1.3
Connecting...
Auto-detected Flash size: 8m
Running Cesanta flasher stub...
Writing 421888 @ 0x1000... 88064 (20 %)
A fatal error occurred: Timed out waiting for packet header
```
For the hardware configuration I already used external power source connected to esp. Is there any other steps that I missed? Can someone help me on this

cheers

projectgus, 2017-06-08:
Hi @fwndz ,

Sorry for the slow reply. Does flashing always fail at the same offset (88064 / 20%)?

What hardware setup do you have?

Angus

coderfox, 2017-07-23:
Yes, I am running into the same issue.

Flashing always fail at the same offset when using a same firmware.

Firmware A:

```
Writing at 0x00008000... (27 %)
A fatal error occurred: Timed out waiting for packet header
```

Firmware B:

```
Writing at 0x00008000... (17 %)
A fatal error occurred: Timed out waiting for packet header
```

projectgus, 2019-10-18:
Closing as these issues are pretty old now, the flasher stub has been reimplemented from what v1.x was using.

Probably the root cause here is hardware or power, or a faulty serial connection, USB cable, etc.

closed 2019-10-18 by projectgus