espressif/esptool #205: ESP32-D2WD support

ESP32DE, 2017-06-01:
can you please provide the support for eMbeddedFlash Versions of the SoC
example: ESP32-D2WD in the esptool, thanks

best wishes
rudi ;-)

does we can flash the single core version ESP32-S0WD same way like the dual core version ESP32-D0WD?
if not, can u support this in the esptool then too, thank you.

projectgus, 2017-06-01:
Hi @ESP32DE,

With the current esptool.py master branch (and also the version in IDF master branch), this should be supported. esptool reads the efuse information about the flash configuration, and accesses flash accordingly.

Are you getting an error when trying to flash a ESP32-D2WD?

ESP32DE, 2017-06-02:
@projectgus 
just in time this is tested - i get on weekend the revised revision of D2 and D0 with pSRAM for solder.

the practical comes from forum:
the windows pc tool newest version 3.4.8 ( 2017-May-19 ) that use the last base of esptool.py
can not successful read eMbeddedFlash infos
[can not flash firmware](https://esp32.com/viewtopic.php?f=12&p=9535#p9517)

i saw your PR / merge for support D2 - but have you tested it with an ESP32-D2WDQ5?
there is 1.8V VDD_SDIO need ( bootstrap GPIO12 high ) 
have you successful test the Firmware Flash Download?

ESP32DE, 2017-06-02:
@costaud
same in your last version.
btw: can you [please provide the src for windows pc version](http://bbs.espressif.com/viewtopic.php?f=7&t=25&start=10#p13621) ( 3.4.8 2017-May-19)
txs
best wishes
rudi ;-)

projectgus, 2017-06-02:
Hi @ESP32DE,

> the practical comes from forum:
> the windows pc tool newest version 3.4.8 ( 2017-May-19 ) that use the last base of esptool.py
> can not successful read eMbeddedFlash infos
> can not flash firmware

Thanks for clarifying. The Windows flashing tool is maintained separately from esptool.py. @costaud is probably the best person to answer this question.

> i saw your PR / merge for support D2 - but have you tested it with an ESP32-D2WDQ5?

Yes, I've been using one most days as a development device for over two months now. If using esptool.py on the command line or via IDF makefiles, ESP32-D2WD should be fully supported. Please raise an issue on this repo if anything doesn't work as expected when using these environments.

ESP32DE, 2017-06-02:
@projectgus 
txs - i can test earliest this on weekend after soldering itself - will inform you asap.
i will use command line and eclipse for this.

@costaud what version of  "magic stub" you use for D2 ? ;-) 

txs!

ESP32DE, 2017-06-03:
@projectgus 

> Yes, I've been using one most days as a development device for over a month now. 
> If using esptool.py on the command line or via IDF makefiles, ESP32-D2WD should be fully supported. 

nice - 
can you share the info, does ESP32-D2WD be rev0 or rev1 silicon
and can ESP32-D2WD support [pSRAM on this way](https://github.com/espressif/esp-idf/blob/feature/psram_malloc/docs/api-guides/external-ram.rst#support-for-external-ram)

so we can save our time to make the PCB here final if ESP32-D2WD is not rev1 and can not support pSRAM,  and put back the chips to packages ..  thank you.

@costaud
ping

projectgus, 2017-06-05:
> can you share the info, does ESP32-D2WD be rev0 or rev1 silicon

My ESP32-D2WD has revision 0 silicon. I don't know what chips you've received, I am guessing also revision 0 but you may wish to double-check with your contact at Espressif sales or to run the IDF "hello world" example which prints the chip's silicon revision.

ESP32DE, 2017-06-05:
@projectgus 
txs angus! - please can you say, have you use special bootstrap pins for flash the embeddedflash and then later for boot from it?
we can just in time not go on here 
https://www.esp32.com/viewtopic.php?f=12&t=1682&p=9712#p9709  

thank you angus! 
best wishes 
rudi ;-)

ESP32DE, 2017-06-05:
@projectgus 
well done! flash download was ok now.
https://esp32.com/viewtopic.php?f=12&t=1682&p=9720#p9719
think can be closed -

closed 2017-06-05 by ESP32DE