espressif/esptool #207: Command to display PSRAM information

projectgus, 2017-06-07:
@ESP32DE requests as part of #206:
> i know, we need to use extra pins for clk and cs
> but i ask 👍 is it possible over esptool to check, there is psram installed and..
> can we read perhabs the ID from the ESP-PSRAM32 ic?

Marking as an enhancement request for now. Agree it would be potentially useful to support, but will need some consideration.

vshymanskyy, 2018-07-05:
This can be so useful for auto-selecting the firmware that needs to be uploaded to a particular board. Voting for this!

chrismerck, 2019-04-09:
There is a static function `psram_read_id` in `components/esp32/spiram_psram.c` which does the reading. You can modify that file to cache the `dev_id` in a global and add a getter for reading from your application later.

radimkarnis, 2022-02-18:
Hi @josesimoes, I am sorry but no progress has been made. There wasn't a huge demand for this over the years, so the priority is quite low. 

Will try to reconsider this. Or if anyone wants to send a PR, it would be highly appreciated!

josesimoes, 2022-02-18:
@radimkarnis understood! and thanks for the prompt reply.
I'm not a python person, but I can give it a try... 😅  any pointers would be highly appreciated!

radimkarnis, 2024-04-18:
After a long consideration, we have decided not to implement this command.

The reason is that there is no easy way to include the PSRAM driver without significantly increasing the stub size and codebase. Also, the driver differs between octal and quad PSRAMs, and between chips as well - maintaining this is not feasible. 

A solution can be building a basic ESP-IDF app (can be any example) with PSRAM support turned on. The second stage bootloader prints all necessary information about the PSRAM chip. The driver is also well-maintained in ESP-IDF.

Thank you for understanding.

josesimoes, 2024-04-18:
We are already doing that on our CLI that deploys apps to ESP32. Not ideal though...
Can't you code a stub with a similar approach? I mean, just provinding information on PSRAM would be helpful. Ideally some details, but I understand that could be hard.