espressif/esptool #213: A fatal error occurred: Timed out waiting for packet content

matthewbryancurtis, 2017-06-22:
Hey there, I'm having a little trouble getting esptool to erase the flash on my NodeMCU Amica development board: 

```
esptool.py --port /dev/tty.SLAB_USBtoUART erase_flash
esptool.py v2.0
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...

A fatal error occurred: Timed out waiting for packet content
```

```
esptool.py --port /dev/tty.SLAB_USBtoUART --baud 115200 write_flash --flash_size=detect 0 esp8266-20170612-v1.9.1.bin 
esptool.py v2.0
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0x0, SizeID=0x0), defaulting to 4MB
Flash params set to 0x0040
Compressed 598432 bytes to 390604...

A fatal error occurred: Timed out waiting for packet content
```

```
esptool.py --port /dev/tty.SLAB_USBtoUART --baud 9600 write_flash --flash_size=detect 0 esp8266-20170612-v1.9.1.bin 
esptool.py v2.0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...

A fatal error occurred: Timed out waiting for packet header
```

Any tips that might help me get through this issue? It also might be related to the code I put on the NodeMCU (I think I may have bricked it):

```
# boot.py
import machine

if machine.reset_cause() == machine.DEEPSLEEP_RESET:
    print('woke from a deep sleep')
else:
    print('power on or hard reset')

# configure RTC.ALARM0 to be able to wake the device
rtc = machine.RTC()
rtc.irq(trigger=rtc.ALARM0, wake=machine.DEEPSLEEP)

# set RTC.ALARM0 to fire after 10 seconds (waking the device)
rtc.alarm(rtc.ALARM0, 10000)

# put the device to sleep
machine.deepsleep()

```

Thanks for looking!

projectgus, 2017-06-22:
It does seem likely that the RTC alarm is starting and is not being disabled when esptool.py resets the board, so when it fires 10 seconds into flashing the whole thing blows up.

This is a bug we should fix, but I think you can un-brick your module right now as well:
* Disconnect the USB cable, depowering the NodeMCU board.
* Hold down the FLASH button while keeping the board unpowered.
* Plug in USB power while continuing to hold down the FLASH button.
* Run esptool.py while continuing to hold FLASH.
* Once esptool.py starts the operation, you can release FLASH.

Holding FLASH down from initial power-on keeps the ESP8266 in bootloader mode, so it never runs the program which sets the RTC timer.

matthewbryancurtis, 2017-06-26:
Thank you for responding so quickly. Your suggestion allowed me to flash both of the ESP8266s that I had been having trouble with.

One is back in perfect condition, the other is no longer responding to ampy/screen after flashing MicroPython. However I have no reason to believe this is an issue with esptool as I'm not running into any errors when flashing.

Thank you for your help!

rmokerone, 2017-07-02:
I also encountered the same problem.When I debug, I found the problem in the ``slip_reader`` function.
```python
def slip_reader(port):
    """Generator to read SLIP packets from a serial port.
    Yields one full SLIP packet at a time, raises exception on timeout or invalid data.

    Designed to avoid too many calls to serial.read(1), which can bog
    down on slow systems.
    """
    partial_packet = None
    in_escape = False
    while True:
        waiting = port.inWaiting()
        read_bytes = port.read(1 if waiting == 0 else waiting)
        if read_bytes == b'':
            raise FatalError("Timed out waiting for packet %s" % ("header" if partial_packet is None else "content"))
        for b in read_bytes:

            if type(b) is int:
                b = bytes([b])  # python 2/3 compat

            if partial_packet is None:  # waiting for packet header
                if b == b'\xc0':
                    partial_packet = b""
                else:
                    raise FatalError('Invalid head of packet (%r)' % b)
            elif in_escape:  # part-way through escape sequence
                in_escape = False
                if b == b'\xdc':
                    partial_packet += b'\xc0'
                elif b == b'\xdd':
                    partial_packet += b'\xdb'
                else:
                    raise FatalError('Invalid SLIP escape (%r%r)' % (b'\xdb', b))
            elif b == b'\xdb':  # start of escape sequence
                in_escape = True
            elif b == b'\xc0':  # end of packet
                yield partial_packet
                partial_packet = None
            else:  # normal byte in packet
                partial_packet += b
```
I think the reason is as follows：
``Ch43x chip can not return data immediately.``
#### When I use ``cp2102`` chip to read chip_id, log as follows:
```
esptool.py v2.0.1
set baudrate ...
Connecting...#### Enter slip_reader function ####
waiting =  0
read_bytes =  b''
have one error
FatalError log
.#### Enter slip_reader function ####
waiting =  0
read_bytes =  b'\xc0'
waiting =  31
read_bytes =  b'\x01\x08\x02\x00\x07\x07\x12 \x00\x00\xc0\xc0\x01\x08\x02\x00\x07\x07\x12 \x00\x00\xc0\xc0\x01\x08\x02\x00\x07\x07\x12'
###### Out from slip_reader function ######
###### Back to slip_reader function #####
###### Out from slip_reader function ######
###### Back to slip_reader function #####
waiting =  0
read_bytes =  b' '
waiting =  31
read_bytes =  b'\x00\x00\xc0\xc0\x01\x08\x02\x00\x07\x07\x12 \x00\x00\xc0\xc0\x01\x08\x02\x00\x07\x07\x12 \x00\x00\xc0\xc0\x01\x08\x02'
###### Out from slip_reader function ######
###### Back to slip_reader function #####
###### Out from slip_reader function ######
###### Back to slip_reader function #####
###### Out from slip_reader function ######
###### Back to slip_reader function #####
waiting =  0
read_bytes =  b'\x00'
waiting =  31
read_bytes =  b'\x07\x07\x12 \x00\x00\xc0\xc0\x01\x08\x02\x00\x07\x07\x12 \x00\x00\xc0\xc0\x01\x08\x02\x00\x07\x07\x12 \x00\x00\xc0'
###### Out from slip_reader function ######
###### Back to slip_reader function #####
###### Out from slip_reader function ######
###### Back to slip_reader function #####
###### Out from slip_reader function ######

Detecting chip type...###### Back to slip_reader function #####
waiting =  0
read_bytes =  b''
have one error
FatalError log

A fatal error occurred: Timed out waiting for packet header
```
#### But when I use cp2102 chips, log is as follows：
```bash
esptool.py v2.0.1
set baudrate ...
Connecting...#### Enter slip_reader function ####
waiting =  0
read_bytes =  b''
have one error
FatalError log
.#### Enter slip_reader function ####
waiting =  0
read_bytes =  b'\xc0'
waiting =  95
read_bytes =  b'\x01\x08\x02\x00\x07\x07\x12 \x00\x00\xc0\xc0\x01\x08\x02\x00\x07\x07\x12 \x00\x00\xc0\xc0\x01\x08\x02\x00\x07\x07\x12 \x00\x00\xc0\xc0\x01\x08\x02\x00\x07\x07\x12 \x00\x00\xc0\xc0\x01\x08\x02\x00\x07\x07\x12 \x00\x00\xc0\xc0\x01\x08\x02\x00\x07\x07\x12 \x00\x00\xc0\xc0\x01\x08\x02\x00\x07\x07\x12 \x00\x00\xc0\xc0\x01\x08\x02\x00\x07\x07\x12 \x00\x00\xc0'
###### Out from slip_reader function ######
###### Back to slip_reader function #####
###### Out from slip_reader function ######
###### Back to slip_reader function #####
###### Out from slip_reader function ######
###### Back to slip_reader function #####
###### Out from slip_reader function ######
###### Back to slip_reader function #####
###### Out from slip_reader function ######
###### Back to slip_reader function #####
###### Out from slip_reader function ######
###### Back to slip_reader function #####
###### Out from slip_reader function ######
###### Back to slip_reader function #####
###### Out from slip_reader function ######

Detecting chip type...###### Back to slip_reader function #####
waiting =  0
read_bytes =  b'\xc0'
waiting =  11
read_bytes =  b'\x01\n\x02\x00\x00 \x06\x00\x00\x00\xc0'
###### Out from slip_reader function ######
set baudrate ...
 ESP8266
Chip is ESP8266
```

You can see that after printing out ``Detecting chip type...`` line.  ``ch340``and ``cp2102``also output ``waiting =  0``. But, ``read_bytes = port.read(1 if waiting == 0 else waiting)`` still try to read 1 byte. ``CP2102`` immediately returned a, but ``CH340`` can not return anything.So, the error appeared!

#### Solution
If you want to solve this problem, you can try to add the following code above the ``waiting = port.inWaiting()`` line.
```python
time.sleep(0.1)
```
Good luck~

projectgus, 2017-07-03:
@rmokerone Thanks for sending this information. I'm not sure the root cause is the same as the issue you are commenting on, but it's still useful to look into. I have some questions:

* What host OS are you using?

> If you want to solve this problem, you can try to add the following code

Does this solve the problem for you?

> CP2102 immediately returned a, but CH340 can not return anything.So, the error appeared!

At the time that the detect_port() read is happening, the serial port timeout should be set to 3.0 seconds - so it should wait 3 seconds before returning 0 bytes. But maybe the CH340 driver on your system is returning early? Can you please add the following print statements to your test:

```
    while True:
        waiting = port.inWaiting()
        start = time.time()
        print("waiting = %d @ %.3f" % (waiting, start))
        read_bytes = port.read(1 if waiting == 0 else waiting)
        print("returned %d bytes @ %.3f (%.3fs)" % (len(read_bytes), time.time(), time.time() - start))
```

rmokerone, 2017-07-03:
@projectgus thank you for your reply！

>What host OS are you using?
macOS Sierra 10.12.5
>Does this solve the problem for you?
Very perfect solution to my problem.

###  CH340 OUTPUT
```
➜  esptool git:(reading) ✗ python3 esptool.py -p /dev/tty.wchusbserial1420 chip_id
esptool.py v2.0.1
Connecting...waiting = 0 @ 1499052157.410
returned 0 bytes @ 1499052157.515 (0.104s)
.waiting = 0 @ 1499052157.567
returned 1 bytes @ 1499052157.574 (0.007s)
waiting = 31 @ 1499052157.575
returned 31 bytes @ 1499052157.575 (0.000s)
waiting = 0 @ 1499052157.575
returned 1 bytes @ 1499052157.577 (0.002s)
waiting = 31 @ 1499052157.577
returned 31 bytes @ 1499052157.578 (0.000s)
waiting = 0 @ 1499052157.578
returned 1 bytes @ 1499052157.580 (0.002s)
waiting = 31 @ 1499052157.580
returned 31 bytes @ 1499052157.580 (0.000s)

Detecting chip type...waiting = 0 @ 1499052157.581
returned 0 bytes @ 1499052160.584 (3.003s)

A fatal error occurred: Timed out waiting for packet header
```

### CP2102 OUTPUT
```
➜  esptool git:(reading) ✗ python3 esptool.py -p /dev/tty.SLAB_USBtoUART chip_id
esptool.py v2.0.1
Connecting...waiting = 0 @ 1499052253.016
returned 0 bytes @ 1499052253.120 (0.105s)
.waiting = 0 @ 1499052253.174
returned 1 bytes @ 1499052253.188 (0.013s)
waiting = 95 @ 1499052253.188
returned 95 bytes @ 1499052253.188 (0.000s)

Detecting chip type...waiting = 0 @ 1499052253.189
returned 1 bytes @ 1499052253.191 (0.002s)
waiting = 11 @ 1499052253.191
returned 11 bytes @ 1499052253.191 (0.000s)
 ESP8266
Chip is ESP8266
Uploading stub...
waiting = 0 @ 1499052253.191
returned 1 bytes @ 1499052253.195 (0.004s)
waiting = 11 @ 1499052253.195
returned 11 bytes @ 1499052253.196 (0.000s)
waiting = 0 @ 1499052253.198
returned 1 bytes @ 1499052253.743 (0.546s)
waiting = 11 @ 1499052253.743
returned 11 bytes @ 1499052253.743 (0.000s)
waiting = 0 @ 1499052253.744
returned 1 bytes @ 1499052253.913 (0.169s)
waiting = 11 @ 1499052253.913
returned 11 bytes @ 1499052253.913 (0.000s)
waiting = 0 @ 1499052253.913
returned 1 bytes @ 1499052253.917 (0.004s)
waiting = 11 @ 1499052253.917
returned 11 bytes @ 1499052253.917 (0.000s)
waiting = 0 @ 1499052253.917
returned 1 bytes @ 1499052253.988 (0.070s)
waiting = 11 @ 1499052253.988
returned 11 bytes @ 1499052253.988 (0.000s)
Running stub...
waiting = 0 @ 1499052253.988
returned 1 bytes @ 1499052253.991 (0.003s)
waiting = 11 @ 1499052253.992
returned 11 bytes @ 1499052253.992 (0.000s)
waiting = 0 @ 1499052253.992
returned 1 bytes @ 1499052253.994 (0.002s)
waiting = 5 @ 1499052253.994
returned 5 bytes @ 1499052253.994 (0.000s)
Stub running...
waiting = 0 @ 1499052253.995
returned 1 bytes @ 1499052253.997 (0.003s)
waiting = 11 @ 1499052253.997
returned 11 bytes @ 1499052253.997 (0.000s)
waiting = 0 @ 1499052253.998
returned 1 bytes @ 1499052254.001 (0.003s)
waiting = 11 @ 1499052254.001
returned 11 bytes @ 1499052254.001 (0.000s)
Chip ID: 0x0081ab16
Hard resetting...
```

### CH340 after add 0.1s delay.
```bash
➜  esptool git:(reading) ✗ python3 esptool.py -p /dev/tty.wchusbserial1420 chip_id
esptool.py v2.0.1
Connecting...waiting = 0 @ 1499052416.299
returned 0 bytes @ 1499052416.402 (0.103s)
.waiting = 96 @ 1499052416.560
returned 96 bytes @ 1499052416.560 (0.000s)

Detecting chip type...waiting = 12 @ 1499052416.665
returned 12 bytes @ 1499052416.665 (0.000s)
 ESP8266
Chip is ESP8266
Uploading stub...
waiting = 12 @ 1499052416.769
returned 12 bytes @ 1499052416.769 (0.000s)
waiting = 12 @ 1499052417.381
returned 12 bytes @ 1499052417.381 (0.000s)
waiting = 12 @ 1499052417.632
returned 12 bytes @ 1499052417.632 (0.000s)
waiting = 12 @ 1499052417.737
returned 12 bytes @ 1499052417.737 (0.000s)
waiting = 12 @ 1499052417.899
returned 12 bytes @ 1499052417.899 (0.000s)
Running stub...
waiting = 18 @ 1499052418.000
returned 18 bytes @ 1499052418.000 (0.000s)
Stub running...
waiting = 12 @ 1499052418.105
returned 12 bytes @ 1499052418.105 (0.000s)
waiting = 12 @ 1499052418.209
returned 12 bytes @ 1499052418.209 (0.000s)
Chip ID: 0x0080dd2d
Hard resetting...
```

Very strange！！！

projectgus, 2017-07-03:
@rmokerone That is indeed strange! It looks like a bug in the CH340 macOS driver. Can you tell me what version of the CH340 driver you have? I will try to report it to WCH.

We can probably work around this by subclassing pyserial.Serial and implementing a read() which breaks up long timeouts into a sequence of shorter ones, so it polls more often. But we should also try to get the driver fixed!

rmokerone, 2017-07-03:
@projectgus 
I used the driver as follows:
>http://www.wch.cn/download/CH341SER_MAC_ZIP.html

Thank you for your hard work:-D

projectgus, 2017-07-03:
> implementing a read() which breaks up long timeouts into a sequence of shorter ones, so it polls more often

Actually, to confirm that workaround can be effective, could you please try:

```
while True:
        waiting = port.inWaiting()
        start = time.time()
        print("waiting = %d @ %.3f" % (waiting, start))
        read_bytes = port.read(1 if waiting == 0 else waiting)
        print("returned %d bytes @ %.3f (%.3fs)" % (len(read_bytes), time.time(), time.time() - start))
        if len(read_bytes) == 0:
                read_bytes = port.read(1 if waiting == 0 else waiting)
                print("returned %d bytes @ %.3f (%.3fs)" % (len(read_bytes), time.time(), time.time() - start))
```

projectgus, 2017-07-03:
> I used the driver as follows:

OK, thanks (for benefit of anyone who finds this later, this is V1.4 of the driver.)

rmokerone, 2017-07-03:
### CH340 OUTPUT
```bash
➜  esptool git:(reading) ✗ python3 esptool.py -p /dev/tty.wchusbserial1420 chip_id
esptool.py v2.0.1
Connecting...waiting = 0 @ 1499054081.272
returned 0 bytes @ 1499054081.375 (0.103s)
returned 0 bytes @ 1499054081.480 (0.208s)
.waiting = 0 @ 1499054081.534
returned 1 bytes @ 1499054081.542 (0.007s)
waiting = 31 @ 1499054081.542
returned 31 bytes @ 1499054081.542 (0.000s)
waiting = 0 @ 1499054081.542
returned 1 bytes @ 1499054081.544 (0.002s)
waiting = 31 @ 1499054081.544
returned 31 bytes @ 1499054081.544 (0.000s)
waiting = 0 @ 1499054081.545
returned 1 bytes @ 1499054081.547 (0.003s)
waiting = 31 @ 1499054081.547
returned 31 bytes @ 1499054081.547 (0.000s)

Detecting chip type...waiting = 0 @ 1499054081.548
returned 0 bytes @ 1499054084.551 (3.004s)
returned 0 bytes @ 1499054087.556 (6.009s)

A fatal error occurred: Timed out waiting for packet header
```

projectgus, 2017-07-03:
```
Detecting chip type...waiting = 0 @ 1499054081.548
returned 0 bytes @ 1499054084.551 (3.004s)
returned 0 bytes @ 1499054087.556 (6.009s)

A fatal error occurred: Timed out waiting for packet header
```

Urgh, that's annoying! So the bytes are lost forever somehow.

We can do the `delay(0.1)` type workaround, the only problem is this slows down every read operation by 100ms - and there are a lot of these.

One more thing to try (sorry, I don't have an OS X machine or I'd try some of these myself):

Near the top of the file:
```
DEFAULT_TIMEOUT = 0.5       # timeout for most flash operations
```

rmokerone, 2017-07-03:
It does not matter, I am happy to solve this problem.
### CH340 OUTPUT
``` bash
➜  esptool git:(reading) ✗ python3 esptool.py -p /dev/tty.wchusbserial1420 chip_id
esptool.py v2.0.1
Connecting...waiting = 0 @ 1499055848.235
returned 1 bytes @ 1499055848.242 (0.007s)
waiting = 31 @ 1499055848.242
returned 31 bytes @ 1499055848.242 (0.000s)
waiting = 0 @ 1499055848.242
returned 1 bytes @ 1499055848.245 (0.003s)
waiting = 31 @ 1499055848.245
returned 31 bytes @ 1499055848.245 (0.000s)
waiting = 0 @ 1499055848.245
returned 1 bytes @ 1499055848.247 (0.003s)
waiting = 31 @ 1499055848.247
returned 31 bytes @ 1499055848.248 (0.000s)

Detecting chip type...waiting = 0 @ 1499055848.248
returned 0 bytes @ 1499055848.753 (0.505s)
returned 0 bytes @ 1499055849.255 (1.007s)

A fatal error occurred: Timed out waiting for packet header
```

CP2102 modified output correctly.
### CP2102
```
➜  esptool git:(reading) ✗ python3 esptool.py -p /dev/tty.SLAB_USBtoUART chip_id
esptool.py v2.0.1
Connecting...waiting = 0 @ 1499055917.411
returned 0 bytes @ 1499055917.514 (0.103s)
returned 0 bytes @ 1499055917.617 (0.206s)
.waiting = 0 @ 1499055917.672
returned 1 bytes @ 1499055917.685 (0.013s)
waiting = 95 @ 1499055917.685
returned 95 bytes @ 1499055917.685 (0.000s)

Detecting chip type...waiting = 0 @ 1499055917.686
returned 1 bytes @ 1499055917.688 (0.003s)
waiting = 11 @ 1499055917.688
returned 11 bytes @ 1499055917.688 (0.000s)
 ESP8266
Chip is ESP8266
Uploading stub...
waiting = 0 @ 1499055917.689
returned 1 bytes @ 1499055917.693 (0.004s)
waiting = 11 @ 1499055917.693
returned 11 bytes @ 1499055917.693 (0.000s)
waiting = 0 @ 1499055917.695
returned 0 bytes @ 1499055918.199 (0.504s)
returned 1 bytes @ 1499055918.240 (0.545s)
waiting = 11 @ 1499055918.240
returned 11 bytes @ 1499055918.240 (0.000s)
waiting = 0 @ 1499055918.241
returned 1 bytes @ 1499055918.410 (0.169s)
waiting = 11 @ 1499055918.410
returned 11 bytes @ 1499055918.410 (0.000s)
waiting = 0 @ 1499055918.410
returned 1 bytes @ 1499055918.414 (0.004s)
waiting = 11 @ 1499055918.414
returned 11 bytes @ 1499055918.414 (0.000s)
waiting = 0 @ 1499055918.415
returned 1 bytes @ 1499055918.485 (0.070s)
waiting = 11 @ 1499055918.485
returned 11 bytes @ 1499055918.485 (0.000s)
Running stub...
waiting = 0 @ 1499055918.486
returned 1 bytes @ 1499055918.489 (0.003s)
waiting = 11 @ 1499055918.489
returned 11 bytes @ 1499055918.489 (0.000s)
waiting = 0 @ 1499055918.489
returned 1 bytes @ 1499055918.491 (0.002s)
waiting = 5 @ 1499055918.491
returned 5 bytes @ 1499055918.491 (0.000s)
Stub running...
waiting = 0 @ 1499055918.491
returned 1 bytes @ 1499055918.494 (0.003s)
waiting = 11 @ 1499055918.494
returned 11 bytes @ 1499055918.494 (0.000s)
waiting = 0 @ 1499055918.494
returned 1 bytes @ 1499055918.497 (0.003s)
waiting = 11 @ 1499055918.497
returned 11 bytes @ 1499055918.497 (0.000s)
Chip ID: 0x0081ab16
Hard resetting...
```

projectgus, 2017-07-03:
OK. I was thinking their might be a limit or an overflow for the timeout for the CH340 driver, but it doesn't seem like it. Thanks.

jtbasb, 2017-07-14:
I have the same problem, I can not flash the firmware. The board is new, unused. I bought it yesterday. I want to rule out this failure. please help. Thank you.

yangbo1979, 2017-10-26:
I'v solved this problem by changing my usb_to_ttl device.
fyi

yangbo1979, 2017-10-26:
I'v got the same problem after I upgraded my usb-to-ttl(cp210x) drivers.
and I solved this problem by change my usb-to-ttl device,I'v connected 5v,gnd,tx,rx to nodemcu dev board directly and I can download/upload firmware without problem.

here is my device without problem:
https://item.taobao.com/item.htm?spm=a1z09.2.0.0.37ca0ed35nLF2r&id=544540189717&_u=qk521h2f93

fyi.

projectgus, 2019-02-08:
Thanks for being patient with this issue.

I think esptool v2.6 and the fix for #351 will include a fix for this. If setting RTC.ALARM0 still doesn't work with esptool v2.6, please reopen this issue.

closed 2019-02-08 by projectgus