espressif/esptool #214: Unable to communicate with my board :(

vincegre, 2017-06-27:
Running esptool 2.0 on Ubuntu 16.04 and trying to communicate with my ESP8266 but it timeouts all the time at Connecting....

I can see the new dev created when I plug it in usb as /dev/ttyUSB0 and blue led on it blinks from time to time !

Here is what I get if I try to clean the flash for example:

`vincen@Fixe-Ubuntu:~$ esptool.py version
esptool.py v2.0
2.0
vincen@Fixe-Ubuntu:~$ esptool.py --port /dev/ttyUSB0 erase_flash
esptool.py v2.0
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
vincen@Fixe-Ubuntu:~$ 
`
The boards I have are these ones:
[image]

projectgus, 2017-06-27:
This style of board usually works without any manual intervention. It's possible the hardware is different or faulty in some way, though.

* If you open a serial terminal program and connect to the ESP8266 do you see any output? The blue light blinking should indicate serial TX. You can try pressing the "reset" button while connecting. Try baud rates 74880 and 115200.
* If you hold down the FLASH button while running esptool.py, does it successfully connect?
* If you hold down the FLASH button, then run esptool.py and press then release the "RST" button while it says "Connecting...", without releasing FLASH, does it succesfully connect?

There are some other troubleshooting steps listed here:
https://github.com/espressif/esptool/#bootloader-wont-respond

vincegre, 2017-06-28:
@projectgus Thanks for your help and yep first one I was able to flash it without using Flash and Rst buttons !!
It looks like I have a defective one :( I tried in same way with an other board and it worked directly !!
Holding Flash button with running esptool or Flash + Rst as you described, none of them worked :(
I tried also to flash it with NodeMCU tool but same it doesn't succeed to connect !

projectgus, 2017-06-28:
Thanks for the update @vincegre . Sorry to hear that it's probably defective.

I'm going to close this issue, but feel free to reopen if you think esptool is doing the wrong thing here.

closed 2017-06-28 by projectgus