espressif/esptool #216: Can not set SPI_PAD_CONFIG_CS0 to values greater than 0x1f

jaanh, 2017-07-17:
Running `python espefuse.py burn_efuse SPI_PAD_CONFIG_CS0 32` in espefuse.py v2.0.1 yields the following error: `A fatal error occurred: Value mask for efuse SPI_PAD_CONFIG_CS0 is 0x1f. Value 0x20 is too large.`

The documentation for espefuse.py states that GPIO 32 and 33 are allowed as SPI pins and from what I understand should be entered as their real value (e.g "32" or "33").

jaanh, 2017-07-19:
Worked around the problem by changing lines 315-316 to:
`if args.new_value & (efuse.mask >> efuse.shift) != args.new_value:
        if args.new_value not in [32, 33]:
            raise esptool.FatalError("Value mask for efuse %s is 0x%x. Value 0x%x is too large." % (efuse.register_name, efuse.mask >> efuse.shift, args.new_value))`

projectgus, 2017-07-19:
Hi @jaanh ,

Thanks for reporting this. It's a bug and we will add a fix. The workaround you should have sounds like a good one until then.

Angus

jaanh, 2017-10-04:
Has the issue been fixed?

closed 2021-03-22 by espressif-bot