espressif/esptool #217: Unable to flash a module among multiples !

vincegre, 2017-07-19:
Hi,

I have flashed already few modules (NodeMCU V2) with your great tool esptool.py and thanks for it ;)
I just encounter an issue with one of them where I'm unable to flash or erase flash on it ! In fact esptool is unable to connect to it unless all other modules were working fine (all modules identical from same manufacturer). At first I thought module was defective but bleu LED on it blinks at startup or if I push the little RST button on board ! I tried multiple different speeds without much success !
I then went with Minicom to check if it talks at boot and only stuff I get is this:
`�$`don't use rtc mem data
{l�
`
Is it faulty board ? or something I can fix in a way or an other ?

Thanks for help,

Vincèn

projectgus, 2017-07-19:
If you hold the FLASH button down while tapping RST, do you get a different result?

You can also try running esptool.py while holding FLASH, and/or tapping the RST button as you run esptool.py.

If either of these attempts work, the hardware may be partly faulty (the automatic reset to bootloader mode is not working coorrectly.)

vincegre, 2017-07-20:
@projectgus  If I hold flash button while tapping RST, I get a string of garbage characters (different each time) displayed in terminal !
For trying to run esptool while holding flash and/or tapping rst button, no luck at all :(
I guess I can throw away that module at recycle bin no ?

projectgus, 2017-08-14:
If you set the baud rate to 74880bps, do you get anything meaningful in the serial terminal?

In all likelihood the module hardware is faulty in some way.

vincegre, 2017-08-14:
@projectgus thanks for your answer but unhappy I have no program compatible with that terminal speed to check it out :( Minicom doesn't handle that speed and looks it's pretty tricky to get that speed on Linux system :( Any suggestions how to do that easily in Linux (Ubuntu 16.04 by the way) ? ;)

projectgus, 2017-08-14:
If you have esptool then you have miniterm (it's bundled with pyserial, which is a dependency of esptool):

Try something like `miniterm.py --raw /dev/ttyUSBx 74880`.

vincegre, 2017-08-14:
@projectgus  thanks for info ;) was not aware of that :(

Here is result I get when I push rst button on module:
```
vincen@Fixe-Ubuntu:~$ miniterm.py --raw /dev/ttyUSB0 74880
--- Miniterm on /dev/ttyUSB0  74880,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 29796, room 16 
tail 4
chksum 0xf7
load 0x3ffe8000, len 2268, room 4 
tail 8
chksum 0x5d
load 0x3ffe88dc, len 8, room 0 
tail 8
chksum 0x60
csum 0x60
��+
   խ�-+PUUP�TT
Ӏ����������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������
```

Does it make any sense ?

projectgus, 2017-08-14:
It looks like it's booting normally (garbage happens when it changes baud rate again.) If you hold down the FLASH button at the same time, is the output different?

vincegre, 2017-08-15:
@projectgus If I push and hold down flash button while pushing rst, I get only that:

` ets Jan  8 2013,rst cause:2, boot mode:(1,6)`

vincegre, 2017-09-06:
@projectgus some news after my last test ? or trash that module ?? Thanks

projectgus, 2017-09-07:
Sorry, I missed your last reply.

"boot mode:(1, x)" suggests that it's going into the UART bootloader mode. I don't know why it's not responding to esptool.py at this point (maybe the TX side of UART communication is broken so it's read-only?)

It might be repairable but trashing the module is probably the most efficient use of your time.

vincegre, 2017-09-08:
@projectgus Thanks for explanation and yep I guess it's better to trash it directly instead of trying to fix such thing ;)

closed 2017-09-08 by vincegre