espressif/esptool #218: ESP reports wrong amount of memory !

vincegre, 2017-07-24:
Hi

Just received some D1 Mini pro boards and I wanted to check them but got a bad surprise !! From what I have chocked on Internet that module exists only with 16Mb of Flash (confirmed by a print on the board itself !!).
When I check it with ESPtool it reports only 2M :(

`vincen@Fixe-Ubuntu:~$ esptool.py flash_id
esptool.py v2.0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Manufacturer: ef
Device: 4015
Detected flash size: 2MB
Hard resetting...
vincen@Fixe-Ubuntu:~$ 
`
Problem with ESP or the module ? For info my ESPTool works perfectly with some other modules I have.

Thanks

Vincèn

projectgus, 2017-08-14:
Hi @vincegre,

Sorry for the slow reply. The detected flash size is based on the Flash ID conforming to a convention where the last byte (0x15 in this case) is proportional to the flash size (in this case `2 ^ 0x15` bytes ie 2 megabytes).

This is only a convention so there's no guarantee any particular manufacturer has followed it (although most seem to). Some thoughts & questions:

* 2 megabytes (MB) is 16 megabits (Mb). So it's possible the module vendor is messing with you and has quoted you a megabit spec where you expected megabytes. Genuine Wemos D1 Mini Pros should have 16MB of flash, though (is it possible these are knockoffs?)
* Can you read the part number markings from the physical flash chip itself?
* You can manually configure a flash size in bytes by running a command line like `esptool.py write_flash --flash_size 16MB ...`. This will override the detected size, but if the flash is really only 2MB then it will probably not work (most flashes "wrap around" to the beginning once you write past the end.)

vincegre, 2017-08-28:
@projectgus sorry for late answer as I was away and had no time to check on that so you were right and I could flash it successfully with that command:

`esptool.py -p /dev/ttyUSB0 write_flash --flash_size 16m 0x0 ESPEasy_v2.0.0-dev11_test_1024.bin`

Thanks :+1:

closed 2017-08-28 by vincegre

projectgus, 2017-08-28:
Glad you got it working.

Just for the record, `--flash_size 16m` is the old size format which is *megabits* ie 2MB. You need to pass `--flash_size 16MB` if you really have 16 megabytes of flash (ie 128 megabits) and want the ESP to use it all.