espressif/esptool #219: Blinky example does not work on ESP-12

Navis-Raven, 2017-07-26:
The blinky example does not work on ESP-12 with esptool by doing the simple 
make flash command or even by doing this esptool command : 

esptool.py -p /dev/ttyUSB0 write_flash 0x00000 blinky-0x00000.bin 0x10000 blinky-0x10000.bin

This topic is the continuty of this one https://github.com/pfalcon/esp-open-sdk/issues/279#issuecomment-317261706 . But I put it here because it seems there is a problem with esptool configuring with some kind of ESP-12F or/and ESP-12E .

Navis-Raven, 2017-07-26:
PS: I'm talking about this blinky example available here, not another one: https://github.com/pfalcon/esp-open-sdk/tree/master/examples

projectgus, 2017-08-14:
Hi @Navis-Raven ,

Sorry for the slow reply. Looking at the parent issue, it looks like you've since got this working OK. Please reopen if you think there is something esptool can do differently here.

Angus

closed 2017-08-14 by projectgus