espressif/esptool #220: MD5 verify failure on ESP32

lukeweston, 2017-07-31:
I'm getting rubbish on verify after writing the flash, even though the flash write seems to be "working".
Here's the syntax I'm using.
esptool.py --port /dev/ttyUSB0 write_flash 0x1000 ./esp32-20170731-v1.9.1-391-g06168c92.bin

The binary is ESP32 MicroPython, here:
http://micropython.org/resources/firmware/esp32-20170731-v1.9.1-391-g06168c92.bin

esptool version: 2.0.1
Flash is 32Mbit.
The offset is 0x1000
Operations such as erase flash, chip id and flash ID all work fine.
Silicon: ESP32D0WDQ6, Winbond 25Q32FVSIG on a ESP-32S module on a "DOIT" breakout board.
Any suggestions or ideas? Thanks :)

```
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 0)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1085152 bytes to 603007...
Wrote 1085152 bytes (603007 compressed) at 0x00001000 in 59.8 seconds (effective 145.2 kbit/s)...
File  md5: 921d77cc041ee7c6f02d01ad029edf22
Flash md5: 84b46cd6bac076d482f18ed02ea0eef8
MD5 of 0xFF is 2f6caf0d130f51473f80fad98934e572
A fatal error occurred: MD5 of file does not match data in flash!
```

projectgus, 2017-08-14:
Hi @lukeweston ,

As discussed elsewhere, this looks like it was probably a power stability issue with the flash VCC. Can we close this issue on that basis?

Thanks,

Angus

closed 2017-08-15 by lukeweston

abhayghatpande, 2018-11-26:
@projectgus 
we are facing a similar md5 issue on esp32devkit. is this an esp32 issue or is it specific to the board? could you give me some references for the VCC problem and how to resolve it? we only had this problem recently appear after being able to flash several tens of times before.

```
$ esptool.py flash_id
esptool.py v2.5.1
Found 3 serial ports
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 0)
Features: WiFi, BT, Dual Core
MAC: 30:ae:a4:00:38:9c
Uploading stub...
Running stub...
Stub running...
Manufacturer: ef
Device: 4018
Detected flash size: 16MB
Hard resetting via RTS pin...
```

projectgus, 2018-11-28:
There have been some rare instances where the flash chip accidentally sets one or more write protection bits. This leads to some sectors not updating each time a new binary is flashed - so the incorrect flash MD5 value is always the same, but always wrong (in extreme cases it can lead to the chip staying fully erased with all bytes 0xFF - this is why esptool prints the MD5 of all-0xFFs)

If the flash MD5 changes each time, it usually indicates a signal integrity problem - most likely with the ESP32's connection to the SPI flash. If something has been connected to the ESP32 GPIOs which are shared with the SPI flash, it can cause this. Or it can sometimes be a power problem causing the flash chip to glitch out.

abhayghatpande, 2018-11-28:
Thank you for that explanation, @projectgus. Appreciate it.

The one thing we did notice was that the chip appeared not to be erasing the flash, but moved directly to writing it, even though we explicitly set it to do so.

Will definitely keep an eye out and report on the md5 value if the issue occurs again.

sv47, 2022-01-15:
Hi. I have the same problem. Solution: disconnect all sensors from GPIO, as well as power supply VCC, GND

aluxen, 2022-02-27:
Same problem here but nothing works for me, could anyone give me any hints or suggestions?
Thanks very much in advance!

radimkarnis, 2022-02-28:
Hi @aluxen,

does the flash MD5 change or stay the same between tries?
If it changes, the issue could be anything like a bad USB cable, bad flash chip connection, or power issues.
If it doesn't change, the chip might be write-protected or damaged.

There is an extensive [troubleshooting guide](https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html) which covers most known problems.

Also, you can try lowering the baud rate or flashing with the `--no-stub` option.

Radim

aluxen, 2022-02-28:
Hi @radimkarnis, md5 flash is always the same, when I try to write_flash with --no-stubs or lower baud rate, the error is the same.

Some remarks, the program in esp32 is still running and the command erase_flash without --no-stubs runs successfully in 0.0s while in some youtube tutorials it takes about 9.0s, and when I run erase_flash with --no-stubs, "Fatal error occurred: ESP32 ROM does not support erase_flash function." but I followed the steps https://techoverflow.net/2022/02/08/how-to-fix-esp32-a-fatal-error-occurred-esp32-rom-does-not-support-function-erase_flash/, but the error is the same.

Thanks a lot again for your help, but no luck here, I think my esp32 is damaged, i don't want to take up too much of your time, I bought a new esp32, but if you have any tips or suggestions, I'll be very happy if you can resuscitate my esp32, have a good day!

radimkarnis, 2022-02-28:
@aluxen,

> the command erase_flash without --no-stub runs successfully in 0.0s while in some youtube tutorials it takes about 9.0s

This is indeed weird and implies some problem with the flash_chip. If you are using the latest esptool, it is now able to detect some problems with flash chips when connecting. Running `flash_id` should reveal if the flash chips reports back normally.

There is also a possibility of enabled write-protection, as Angus explains:

> There have been some rare instances where the flash chip accidentally sets one or more write protection bits. This leads to some sectors not updating each time a new binary is flashed - so the incorrect flash MD5 value is always the same, but always wrong

[read_flash_status](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/advanced-commands.html#read-flash-chip-registers-read-flash-status) command can be used to compare the faulty chip with a working one.

I am sorry to say this, but replacing the ESP32 might sometimes be the safest and fastest thing to do.

aluxen, 2022-02-28:
@radimkarnis you resurrected my esp, I followed [read_flash_status](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/advanced-commands.html#read-flash-chip-registers-read-flash-status) and ran "esptool.py write_flash_status --bytes 2 --non-volatile 0" and everything worked again, now the command erase_flash is taking about 7s, thank you very much, i am very happy, thank you so much for you time!!!!!

radimkarnis, 2022-02-28:
@aluxen no, I just gave you a hint, YOU resurrected your esp - great job! 🎉  

I am glad this got resolved and happy I could help.

riteshsonawane1372, 2023-04-05:
> @radimkarnis you resurrected my esp, I followed [...]

[image]
Need some help I ran both the cmd

riteshsonawane1372, 2023-04-05:
[image]
@aluxen it's been 3 days I'm having the same issue