espressif/esptool #222: cannot flash wemos d1 mini in Arduino IDE

stevekramer99, 2017-08-15:
The error I get when uploading is "error: espcomm_upload_mem failed".  Can this be fixed?  The output I have when uploading is:

```
Sketch uses 222221 bytes (21%) of program storage space. Maximum is 1044464 bytes.
Global variables use 31588 bytes (38%) of dynamic memory, leaving 50332 bytes for local variables. Maximum is 81920 bytes.
C:\Users\Perigalacticon\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9/esptool.exe -vv -cd nodemcu -cb 921600 -cp COM5 -ca 0x00000 -cf C:\Users\PERIGA~1\AppData\Local\Temp\arduino_build_434248/Blink.ino.bin 
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 921600
	setting port from COM1 to COM5
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 80 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 80 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 80 instead of C0
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 80 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 80 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 80 instead of C0
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 80 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 80 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 80 instead of C0
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
```

projectgus, 2017-08-15:
Hi @stevekramer99,

It's a little confusing, the esptool.exe included in the ESP8266 Arduino IDE is a different tool to this Python-based esptool.

You can open issues regarding this tool here: https://github.com/igrr/esptool-ck/issues

projectgus, 2017-08-15:
Hi @stevekramer99,

It's a little confusing, the esptool.exe included in the ESP8266 Arduino IDE is a different tool to this Python-based esptool.

You can open issues regarding the tool you are using here: https://github.com/igrr/esptool-ck/issues

closed 2017-08-15 by projectgus