espressif/esptool #223: ESP8285 isn't correctly detected

marcelstoer, 2017-08-17:
I've had this [cute ESP8285 board from Tindie](https://www.tindie.com/products/onehorse/esp8285-development-board/) for more than a year. From time to time I use for testing stuff.

The current esptool.py fails to detect the chip type on this board properly.

> Detecting chip type... ESP8266

projectgus, 2017-08-17:
That's a good point. "Chip type" for now is ESP8266/ESP32 only, it's more the "device family" than the specific model number.

I'll see what we can do to detect ESP8285 specifically.

Does it work properly apart from mis-detecting it as an ESP8266?

marcelstoer, 2017-08-18:
Yep, has worked just fine for as long as I had this chip. ESP8285 requires `dout`, though.

closed 2017-09-12 by projectgus

marcelstoer, 2017-12-03:
I finally got around to verifying the fix - all good.

projectgus, 2017-12-03:
@marcelstoer Good to hear. Sorry I forgot to ping you about the v2.2 release.

marcelstoer, 2017-12-03:
No worries, I've kept coming back here to check every other week or so 😉

commit b5fe4d9a (2017-09-12): Detect ESP8285 as a chip type

Closes #223