espressif/esptool #226: Failed to connect to ESP32: Timed out waiting for packet header

mallareddy63, 2017-08-23:
Hie, Im using esp32 with arduino IDE for receiving sensor values from nrf24lo1, when Im trying to upload the code it just displays error for esp32 but the code work with arduino uno. code.zip files contains code that im using. can u guys tell me how to get out from this problem. 

Thanks in advance,

[Code.zip](https://github.com/espressif/esptool/files/1245281/Code.zip)
[image]

tehniq3, 2017-08-24:
I have same problem... see in my movie: from [(https://youtu.be/NZs4i81xTGg)]
ESP32 board enter in upload mode but ... same error

projectgus, 2017-08-28:
Hi @mallareddy63  & @tehniq3 ,

This problem is to do with communications between Arduino (running esptool.py) and the board, rather than anything to do with your code.

Can you please tell us:
* What hardware (ie what development board) are you using?
* What Windows version are you using?

Most likely the development board you have is faulty or has a design flaw which means it won't reset into bootloader mode automatically.

To make it work:

You can try holding down the button marked "BOOT" or "FLASH" while esptool.py is "Connecting...". If you see esptool.py connect then you can release this button.

If this doesn't work, try *also* pressing and releasing the button marked "EN" or "RST" while esptool.py is "Connecting..." (keeping the first button held down the whole time). Again, if you see it connect then you should stop pressing this button.

This may take a few tries to get right.

mallareddy63, 2017-08-28:
Thanks dude @projectgus , its working

my board: nodemcu 32s
windows: 10
Arduino: latest one 

 initially i upload the code to the board and hold down boot button  while esptool.py is "Connecting..."
my emulator was able to send parameters through rf

tehniq3, 2017-08-28:
my board: DOIT ESP32 ESPDEVKIT v1
windows 10
see https://youtu.be/eRF4IqNOw_s
also, I try it to push and hold buttons after enter un uploade mode :((( 
PS: a friend have 2 identical board and same problem..

mallareddy63, 2017-08-28:
Bro initially I click upload option in Arduino IDE to upload the code to the board ,

Make sure you select the correct board and port

Once the ide displays text "connecting ...." I just press and hold the "BOOT " button for some time until connection is finished and it displays "downloading........." At that time I release the button 

After that it displays some text and in the last it displays "Hard resetting" , then I open serial monitor and found it working 

If possible send me code bro I will try it

Electromania, 2018-01-17:
I have same problem with ESP32. I had perfectly working wemos board. i kept it running overnight (accidentally) with measurement data being stored into SPIFFS memory. In the morning, i noticed my OLED display was Frozen - i.e. no values were changing. Then i restarted the ESP32. and then......... everytime i try to upload any sketch i get following error. I tried tons of google searchers and all forums, tried all options mentioned everywhere. Tried Arduino IDE, Eclipse Oxygen. Tried to skip onboard USB-Serial ic of Wemos and use external USB-Serial adapter. Pressed Boot, Enable buttons, reduced baud rate etc. Nothing help and problem remains same.

Could it be that since i was writing to SPIFFS memory (unfortunately in infinite loop - accidentally)  it might have just filled up whole memory ? if that is so, there should be some way of Clearing this memory ???  because, if someone does this accidentally again , then each time the ESP32 board will be useless. 

Is it a hardware issue or software issue ? 
------------------------------Following is error from Arduino IDE----------------------
Sketch uses 184176 bytes (14%) of program storage space. Maximum is 1310720 bytes.
Global variables use 11040 bytes (3%) of dynamic memory, leaving 283872 bytes for local variables. Maximum is 294912 bytes.
C:\Users\xxx\Documents\Arduino\hardware\espressif\esp32/tools/esptool.exe --chip esp32 --port COM3 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 C:\Users\xxx\Documents\Arduino\hardware\espressif\esp32/tools/partitions/boot_app0.bin 0x1000 C:\Users\xxx\Documents\Arduino\hardware\espressif\esp32/tools/sdk/bin/bootloader_qio_80m.bin 0x10000 C:\Users\xxx\AppData\Local\Temp\arduino_build_147733/SPIFFS_Read_Write_sample.ino.bin 0x8000 C:\Users\xxx\AppData\Local\Temp\arduino_build_147733/SPIFFS_Read_Write_sample.ino.partitions.bin 
esptool.py v2.1
Connecting........_
Chip is unknown ESP32 (revision (unknown 0xe))
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 8192 bytes to 47...

A fatal error occurred: Timed out waiting for packet content
A fatal error occurred: Timed out waiting for packet content
*************************************************************************************

-----------------------following error is from Eclipse-----------------
Flashing binaries to serial port COM3 (app at offset 0x10000)...
python /home/xxxx/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port "COM3" --baud 115200 --before "default_reset" --after "hard_reset" write_flash -z --flash_mode "dio" --flash_freq "40m" --flash_size detect 0x1000 /home/xxxx/esp/blink/build/bootloader/bootloader.bin 0x10000 /home/xxx/esp/blink/build/blink.bin 0x8000 /home/Yusuf/esp/blink/build/partitions_singleapp.bin
esptool.py v2.1-beta1
Connecting....
Chip is unknown ESP32 (revision (unknown 0xe))
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Flash params set to 0x0220
Compressed 17552 bytes to 10304...

make: *** [/home/xxxx/esp/esp-idf/components/esptool_py/Makefile.projbuild:53: flash] Error 2
A fatal error occurred: Timed out waiting for packet content

14:00:28 Build Finished (took 1m:19s.935ms)

MikeA1s, 2018-02-22:
For me it was a power issue, I am using a EzSBC esp32 dev board. Interesting it happens when I upload a sketch that requires a lot of pin usage. The way I "fixed" the situation was to apply 3v  2amp external supply to the 3v pin on esp32, plus and external ftdi to rx/tx, with common ground. 

***Applying 5v 2amps to Vin on module did not solve the problem.

ben34, 2018-03-03:
I got very often the same kind of issue when uploading code with ESP32 left on breadboard. Upload works fine when removing the ESP32 from breadboad during the upload operation

KargJonas, 2018-04-16:
For me spamming the flash and boot buttons worked out. Not at first but you gotta try it a few times ...

projectgus, 2018-04-16:
Thanks everyone for the updates.

Going to close this as it seems like most people find solutions using the flash and boot buttons. If you have a development board which doesn't automatically reset reliably, make sure to ask the manufacturer to improve their design! :)

closed 2018-04-16 by projectgus

nageshu, 2018-05-18:
hey @projectgus 
Actually I am facing same problem. sorry to open this again i found your guidance useful.
I am using ESP32 loRa board which have inbuilt loRa module sx1278.
Before showing the error i have interfaced accelerometer and rtc ds1307. but then after that i am getting i have tried all method you discribed but it is not working.
can you please tell what is exact reason behind this error?
or any other method by which can get rid of this error.

Please help !!

thanks in advance!!!!!!

MikeA1s, 2018-05-18:
When I had the error and after investigating and researching I came to conclusion that it was a power supply issue. When there are multiple events taking place on esp32 plus WiFi on and maybe Bluetooth, even if you have disconnected all the peripheries, the sketch is still running on esp32, so when you connect esp32 to your computer and try to upload new sketch there is not enough power from computer to place esp32 into flash mode. Plus, I feel the 5v step down is weak on some esp32 jigs, so I went straight for 3.3v to esp32, plus RX/TX through ftdi (of course 3.3v on ftdi).  That was it for me.

projectgus, 2018-05-21:
@nageshu Testing power, as @MikeA1s suggests, and also check any connections to bootstrapping GPIOs (GPIO numbers 0, 2, 12, 15) on the board. See these two headings on the wiki: https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#select-bootloader-mode 

A full list of troubleshooting steps can be found here:
https://github.com/espressif/esptool/#troubleshooting

FrancescoSaverioZuppichini, 2018-05-28:
I have the same error, somebody fix it?

MikeA1s, 2018-05-30:
You say fix it, but what have you tried and what are you trying to do we don't know.

marisancans, 2018-07-05:
In my case I had somewhere seen that you need to connect GNT to 3V3, so I did, and that caused me problems. Spent hours of my stupidity. Sometimes, look for different angles of attack or even go do something else for a change

tehniq3, 2018-07-06:
GNT or GND ???/ GND to 3V3 blow up the chip...

..............................................................................................................
http://www.arduinotehniq.com/
&
http://nicuflorica.blogspot.ro/
&
http://arduinotehniq.blogspot.com/

yasheena, 2018-08-04:
I solved this problem by attaching a 100nF capacitor  from EN to GND.
(I use a pure ESP-WROOM-32 with an FTDI board and the 2 transistors between DTR/RTS/EN/GPIO0).

xyybob, 2018-08-31:
The cap trick also worked for me! Keeping EN low while GPIO0 is toggled seems to do the trick.

I'm probably misguided here but it appears the timing of esptool.py tries to accomodate bugs in earlier versions of Esp32 (I'm using v1.1 here) - I would attempt a codse change but I'm not clever enough

mrfaptastic, 2018-09-28:
The only way I have managed to get a LOLIN D32 to flash in Arduino using windows, is by jumpering D0 to GND (Ground) just before I click 'upload'.

I then need to remove this jumpering in order to test the sketch. Very annoying.

https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#select-bootloader-mode

aledaluz, 2018-10-06:
Hold Boot for a seconds, when appear Connecting.....____.....,
It works!

andro78, 2018-10-27:
Thanks a lot.

Hold Boot button!!

briangerez, 2018-11-16:
@MikeA1s Excellent contribution!
I disconnected some sensors that were consuming energy. Then the code was loaded without problems!

RussBarr, 2018-11-27:
I did not have this issue until I downloaded the latest version. I an using the ESP32THING from sparkfun. Sometime it connects without holding down the BOOT LOAD button (GPIO0). Other time I must press the BOOT button during the connect. This happens on three different computers running win10 and win7. I am able to build and load BLINK but when I try to run my old code that uses RTOS, I2S, DMA, RMT, and interupts it builds and loads but does not run. The old version from a year ago works fine.
[ws2812_1024FFT_real_imag_OK.zip](https://github.com/espressif/esptool/files/2621229/ws2812_1024FFT_real_imag_OK.zip)

projectgus, 2018-11-28:
Hi @RussBar,

If you add a capacitor between the EN pin and GND (in the 1uF-10uF range) then automatic reset into flashing mode will probably become more reliable. Not all third party development boards have enough capacitance on this pin.

If esptool is flashing the binary successfully but it doesn't run then this is probably an issue to open with ESP-IDF or the arduino-esp32 project. If you're not sure of the specific problem then it might also be useful to post on the https://esp32.com discussion forum. When posting in any of those places, please include as many details as possible (for example, what exactly is the log output when the sketch fails to run).

richard-ubi, 2019-02-14:
> 嘿@projectgus
> 其实我面临同样的问题。很抱歉再次打开此处，我发现您的指导很有用。
> 我正在使用内置loRa模块sx1278的ESP32 loRa板。
> 在显示错误之前，我已经连接了加速度计和rtc ds1307。但之后我得到错误，无法连接esp32和超时等待数据包错误。我已经尝试了你描述的所有方法，但它没有用。
> 你能告诉我这个错误背后的确切原因吗？
> 或任何其他方法可以摆脱这个错误。
> 
> 请帮忙 ！！
> 
> 提前致谢！！！！！！

In fact, I would like to ask blind where can live ESP32 loRa board which have inbuilt loRa module sx1278.

Alvin1Zhang, 2019-02-14:
您现在用哪个 GPIO 管脚连接加速度计和 RTC 呢？ESP32 有一些 Strapping 管脚配置，详情请查看 《ESP32 技术规格书》 中的 2.4 章节《Strapping 管脚》，您也可参考如下链接 https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#select-bootloader-mode 。谢谢。

dialvessilva, 2019-03-08:
Solução do problemas de todos... ou melhor para mim funcionou e muito
[https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/?unapproved=357864&moderation-hash=65b6814463675f72f8fe34a0a4da3cec#comment-357864](url)

Abraço a todos

sravanth005, 2019-03-12:
Hi all,

**the Os i use UBUNTU 16.04 in which,i work on a project esp32-cam (https://robotzero.one/esp32-camera-module/#comment-3073) refer the link.
i followed the link above though it scripted foe windows OS for some commands i went through (https://docs.espressif.com/projects/esp-idf/en/latest/get-started/linux-setup.html#) link and i installed.but after flashing at the end i get error like this

my board:ESP32-cam (AI thinker board)
OS: UBUNTU 16.04
connector(board to laptop): CP2102
for more information about the board refer the first link**

Generating libtcpip_adapter.a.sections_info
Generating libulp.a.sections_info
Generating libunity.a.sections_info
Generating libvfs.a.sections_info
Generating libwear_levelling.a.sections_info
Generating libwifi_provisioning.a.sections_info
Generating libwpa_supplicant.a.sections_info
Generating libxtensa-debug-module.a.sections_info
Generating esp32.common.ld
LD build/camera_web_server.elf
esptool.py v2.6
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting......................................_____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
/home/admin-unizen/esp/esp-idf/components/esptool_py/Makefile.projbuild:64: recipe for target 'flash' failed
make: *** [flash] Error 2

****This is my path which i set:
export PATH=$PATH:/home/admin-unizen/esp32/xtensa-esp32-elf/bin
export IDF_PATH=/home/admin-unizen/esp/esp-idf

i dont know how to solve.can you explain me fully why this happening and will you say the commands from first to last to solve this issue****

Thanks in advance

projectgus, 2019-03-13:
@sravanth005 The ESP32 doesn't seem to be responding to serial commands. Usually this means it is not in the "bootloader" mode. You'll need to connect pin IO0 to GND as shown in the "Connections" diagram at the [link you posted](https://robotzero.one/esp32-camera-module/). After IO0 is connected to GND, you will need to reset the ESP32 somehow - either by pressing a reset button if there is one, or by cycling the power to the ESP32 (unplug & replug USB, maybe.)

If it's still not working, double-check all the other wiring matches what is shown in that diagram.

If it's still not working after that, you'll need to ask the manufacturers of that development board for help as this problem is somewhere in the hardware.

wanglimingGit, 2019-03-19:
我用的ESP32-S裸板外接USB TTL（PL2303），尝试很多方法仍然是“connecting ....”。后更换USB TTL（PL2303）后烧写正常。我是外接的3.3电源可能是某种情况将USB TTL（PL2303）烧毁了 导致无法连接。我觉得如果开发版遇到此情况可尝试直接使用USB TTL 进行烧录。

mallareddy63, 2019-03-19:
press BOOT button or EN button on esp32s when it shows connecting on
Arduino compiler

krisvalencio, 2019-04-04:
can someone help me, i'm using esp-idf on msys32 that i followed from https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#get-started-start-project
but got this at flashing

Christoper@Christoper MINGW32 /c/msys32/home/christoper/esp/esp-idf/examples/get-started/Helloworld
$ make flash
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/msys32/home/Christoper/esp/esp-idf/requirements.txt are satisfied.

App "hello-world" version: v4.0-dev-207-g76da10a66-dirty
Flashing binaries to serial port COM10 (app at offset 0x10000)...
esptool.py v2.7-dev
Serial port COM10
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
make: *** [/home/Christoper/esp/esp-idf/components/esptool_py/Makefile.projbuild:67: flash] Error 2

mallareddy63, 2019-04-05:
u can overcome this problem by long pressing the "EN" button on esp32 when
Arduino IDE displays  Connecting........*.....*.....*.....*.....*.....*
....._____

jpack2015, 2019-06-28:
The boot didn't work for me and I'm not doing all the capacitor stuff, what worked for be was turning the baud rate down to 115200.

m-i-k-e-e, 2019-08-04:
I have the same issue with my ESP8266. After disconnectiong the TX from the board, everything went fine.

koson, 2019-09-09:
Hi, try  this diagram

EN pin ---- R 2.2k -------- (-) C10uF (+) --------- 3.3V pin

I've tried, it works on my DOIT ESP32 Devkit.

strfn, 2019-09-26:
> I got very often the same kind of issue when uploading code with ESP32 left on breadboard. Upload works fine when removing the ESP32 from breadboad during the upload operation

Just spent an hour banging my head ... THANKS !

joeservo, 2019-12-26:
does anyone know what is happening during the Connecting...___.....? is a byte being transmitted to the esp32? 

in a console window or using cat /dev/ttyXXXN i see
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))

waiting for download

seems like its good all is good.

UART0 is the correct port to be used correct? same one the message above is printed on.

danirebollo, 2020-01-08:
Hi, 
I've been able to fix this with 1uF capacitor from IO0 to GND. I have another on EN but the one that solves this bug on some silicon revisions are the 1uF capacitor on IO0.

NetForces, 2020-02-20:
@danirebollo 3.3uF electrolytic between GPIO0 and GND worked on my NodeMCU ESP-32s here. Thanks !

Did not need the `monitor_rts` or `monitor_dtr` here with platformio v4.2.1 on macOS

jordanadania, 2020-05-31:
i needed more power

nicechocolate, 2020-06-23:
This problem seems to vary from board to board, sketch to sketch.  For me, it isn't a short-of-power issue, because I have one board that ALWAYS downloads perfectly, every time, even with an LED display, a tft display and a chain of twenty 2812 LEDs attached to it.  I have another, identical board that seldom loads and resets properly, the only differences can be in the manufacturing.

To summarise:
** First fix: hold the EN button when you see "Connecting......______" and keep it held until the download starts.
** Second fix is a capacitor from the EN connection to ground. This worked for me, with a 0.01uF (10nF) polyester capacitor.  1nF took several seconds (plenty of ......____.....) before it worked, so that seems to be the threshold for that board.  33pF was too little and a ceramic 0.01uF capacitor didn't work, either.  Other brands and constructions of polyester capacitors also worked.
** Maybe third fix, 3.3uF capacitor between GPIO0 and ground when loading. I can't test this, my 30-pin devkit does not have GPIO0 exposed.
** Taking it out of the breadboard made no difference to mine.
** Adding 22uF across the power rail made no difference, not across 5V nor 3.3V

Not exhaustive testing, but empirical results.  I hope I've helped someone, the tip about the EN really helped me.
I think these are noisy little boards, if I have mine plugged-in to program them, that slows-down my Wifi, which is a dongle in the same USB adaptor.

mrdc, 2020-07-08:
@nicechocolate 
@projectgus 

Hello,

Unfortunately, adding capacitors haven’t helped me with this ESP32 dev board: https://github.com/LilyGO/TTGO-T7-Demo

I’ve soldered 1uF capacitor:
1. between EN and GND - auto upload is not working
2. between IO0 and GND - auto upload Is not working
3. between EN and GND + between IO0 and GND - auto upload is still not working :(

Any ideas what else can be done to fix the issue?

projectgus, 2020-07-08:
@mrdc I was going to suggest reporting this to the board designer, but [I see you've done this already](https://github.com/LilyGO/TTGO-T7-Demo/issues/7). It looks like they have implemented an auto-reset circuit using the dual transistor package shown at U2. They also already have the often-discussed capacitor between EN and GND where it should be (C30). 

It's possible this is a QC issue with this exact individual board (something is not soldered correctly), or it's possible the design has a flaw. Unfortunately not a lot that can be done from the esptool side, hopefully the designer is able to help you out.

mrdc, 2020-07-09:
> It's possible this is a QC issue with this exact individual board

I have 2 boards - the same behavior :/

nicechocolate, 2020-07-09:
Sorry, I can't offer much help here, I have the WROOM DOIT DEVKIT V1 board, as shown in this (different) Git post: [https://github.com/espressif/arduino-esp32/issues/544] and I just buy them on EBay.  I have the advantage that I am not trying to design for a job, only as a hobby and I am free to use plenty of space inside my cases and design ("design", that's a laugh, it's just bodging and cutting until it fits) the device around the board.  

I wonder whether it's a power issue on your board, since my board has its own regulator.  Guessing, really.

I appreciate that you may be trying to fit your board into a limited space, but with the amount of time ("=money") you have already spent on solving the problem, I'm tempted to say "cut your losses and use a different board".  If you plan for a larger board, it may then be possible/easier to fit the smaller board if a fix comes to light.  Or use just the ESP SMT module itself and design your own board and have them made, as do one or two regular "diarists" on YouTube, like Ralph Bacon.

mrdc, 2020-07-09:
> have the WROOM DOIT DEVKIT V1 board

In my case only the dev board with WroverB was acceptable, so it’s not a limited space issue :)

joeservo, 2020-07-09:
> @nicechocolate
> @projectgus
> 
> Hello,
> 
> Unfortunately, adding capacitors haven’t helped me with this ESP32 dev board: https://github.com/LilyGO/TTGO-T7-Demo (Board schematics: https://github.com/LilyGO/TTGO-T7-Demo/blob/master/t7_v1.4.pdf)
> 
> I’ve soldered 1uF capacitor:
> 
> 1. between EN and GND - auto upload is not working
> 2. between IO0 and GND - auto upload Is not working
> 3. between EN and GND + between IO0 and GND - auto upload is still not working :(
> 
> Any ideas what else can be done to fix the issue?

what are using to press GPIO0 to enter flash mode? you must press it before releasing reset to enter flash mode.

mrdc, 2020-07-09:
> what are using to press GPIO0 to enter flash mode? you must press it before releasing reset to enter flash mode.

In my case the board doesn’t have any buttons - the only way to enter flash mode is to connect IO0 and GND. But by design it should enter automatically.

joeservo, 2020-07-09:
> In my case the board doesn’t have any buttons - the only way to enter flash mode is to connect IO0 and GND. But by design it should enter automatically.

by design it should work but in practice it never does, thats why we're debugging.  i made my own flashing board but i had good luck with a MM232R eval board as well (soldered wires to my vroom).   

what are you using to flash and what is exactly the issue?

mrdc, 2020-07-09:
> by design it should work but in practice it never does, thats why we're debugging. i made my own flashing board but i had good luck with a MM232R eval board as well (soldered wires to my vroom). [...]

I’m using ESPTool, the issue is that my board is not entering flash mode automatically as my other boards, so I need to unplug it, short IO0 and GND, plug it in, then flash, then remove IO0 and GND and restart, otherwise the board will stay in waiting for firmware mode. Other boards don’t need all these steps - everything is done via software.

NetForces, 2020-07-11:
> @nicechocolate
> @projectgus
>
> Hello,
>
> Unfortunately, adding capacitors haven’t helped me with this ESP32 dev board: https://github.com/LilyGO/TTGO-T7-Demo (Board schematics: [...]

1uF is too small, I had excellent, almost perfect result with a 3.3uF **electrolytic** capacitor between GPIO0 and GND. 

The only case it would not work is if the loop() was too tight (running too fast). Seems to somewhat lock up the controller. In these cases pressing the EN button when the upload starts worked.

kozoke, 2020-09-27:
My Nodemcu esp-32s - you need to push and hold the IOO (or flash on the pinout) diagram.  I forgot about this.  Uploaded after push/held the button.

perrstep, 2021-09-27:
Hi, I faced the same issue working with a NodeMCU ESP32s.  I would advise that you check for any (short?) circuit errors in your sketch.  

It turned out for me that I was using a 3 pin DHT temperature sensor ( + , data, - ) and that I inadvertently connected its + pin to the negative rail on my breadboard while checking for loose connections. At that moment, the sensor started giving wrong readings, and the ESP32 starting having the timeout issue.  I verified that the ESP32 was still good by disconnecting it from all circuitry and uploading a blink program to it. Only to reconnect it to my full circuit and face the timeout issue again.  At that point I could convince myself that the problem was my circuit and indeed realized that I misconnected that DHT sensor pin.  Fixing the circuit fixed the timeout.  And the DHT was still functional.  I Hope this can help in the future.

nicechocolate, 2021-09-28:
Thank you persstep, that must help somebody. Depending on which I/O port you were using, it will have confused one of the internal devices; who knows which, though, there are plenty of them in there!

Most people are encountering difficulty in getting sketches to flash "automatically" even without any external devices, which seems to be a design or construction problem with certain development boards. As I said in an earlier post, I have one board that flashes faultlessly, and another that always needs a capacitor. Same boards, same manufacturer, as far as I can tell. Some need the button pressed, some don't, and it's a bit frustrating when it happens. You start the compile and upload, then go to make a cup of tea or something and then when you get back, expecting your sketch to be executing, your compiler is sitting there saying "...___...___...___...___...___... device timeout".