espressif/esptool #227: esptools various USB>TTY

josep112, 2017-08-23:
I would like to record several esp8266 at one time, would it be possible?

EX: esptool.py --port /dev/ttyUSB0 /dev/ttyUSB1 write_flash 0x00000 Blink.ino.bin

projectgus, 2017-08-28:
Hi @josep112 ,

We don't have this feature at the moment, however we're considering how to expose this kind of functionality.

Because esptool.py is a command line tool, it's easy to combine it with other tools in order to get this kind of parallel execution, though.

For example, as you seem to be on Linux, you can use [GNU Parallel](https://www.gnu.org/software/parallel/):

```
parallel -d ' ' --group esptool.py --port '{}' write_flash 0x0 blink.bin ::: /dev/ttyUSB0 /dev/ttyUSB1 /dev/ttyUSB2
```

This will run esptool.py in parallel on the three serial ports named after the `:::`. The output is grouped per-command, and no output is printed until each esptool.py finishes (otherwise output is interleaved and can be very messy and hard to understand.)

Add the `--dry-run` argument to parallel for easier debugging. Consult the Parallel docs for more details.

You can also store the list of serial ports in a file, or provide them after the parallel argument

josep112, 2017-08-28:
Hi, projectgus
Thanks for the help, it did not work.
parallel -d ' ' --group esptool.py --port '{}' write_flash 0x00000 /usr/src/serial/Blink.ino.bin ::: /dev/ttyUSB0 /dev/ttyUSB1
parallel: Input is read from the terminal. Only experts do this on purpose. Press CTRL-D to exit.

projectgus, 2017-08-28:
Hmm, you may have an older version of parallel to me.

```
$ parallel --version
GNU parallel 20170322
Copyright (C) 2007,2008,2009,2010,2011,2012,2013,2014,2015,2016,2017
```

What version do you have?

josep112, 2017-08-29:
parallel --version
WARNING: YOU ARE USING --tollef. USE --gnu FOR GNU PARALLEL

GNU parallel 20120422
Copyright (C) 2007,2008,2009,2010,2011,2012 Ole Tange and Free Software Foundation, Inc.

My version seems to be old

projectgus, 2017-08-29:
Try passing `--gnu` to parallel in addition to the other arguments. My version no longer has the "tollef mode", that I think Debian/Ubuntu turns on by default.

dobairoland, 2021-09-15:
I'm closing this issue given the fact that parallelization is best to be done on level above esptool as it was discussed.

closed 2021-09-15 by dobairoland

radimkarnis, 2024-05-24:
Hello, yes. Each instance of esptool can only talk to one open port at a time. It can be scripted, if needed, to service multiple ports or deal with batches of chips, but we don't plan to add this functionality directly into the tool to prevent bloating.